ExpressLRS/ExpressLRS #1319: Upgrade to 2.1.0 via WiFi/web fails

sfjuocekr, 2022-01-23:
## Current Behavior
I try to update my receivers over WiFi, the update is pushed but never written.

## Steps to Reproduce
I'm using the ES900TX, all receivers were at 2.0.1 and tried to push the web update to them. Updating over serial works as intended.

- TX hardware: ES900TX
- RX hardware: ES900RX
- Handset model: TX18s
- EdgeTX version (including nightly number): 
- ExpressLRS version (TX & RX MUST MATCH): 2.1.0 on the TX, 2.0.1 on the RX (because it failed to update).
- Packet Rate: 200
- Telemetry Ratio: 1/64
- user_defines: 

-DRegulatory_Domain_FCC_915
-DMY_BINDING_PHRASE="xxx"
-DAUTO_WIFI_ON_INTERVAL="60"
-DHOME_WIFI_SSID="xxx"
-DHOME_WIFI_PASSWORD="xxx"
-DFAST_SYNC

CapnBry, 2022-01-23:
What happens when it fails? It says it is done but nothing happens, or it just can't find them to upload to? Are you using the configurator, the web interface upload, or platformio to update?

sfjuocekr, 2022-01-23:
I have tried the configurator and webif. Did not try platformio, but I can do that.

It does not seem to fail, it completes the upload and the RX then resumes to blink rapidly.

```
Uploading .pio/build/HappyModel_RX_ES900RX_via_WIFI/firmware.bin
 ** UPLOADING TO: http://10.0.2.100/update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
 28  375k    0     0   28  108k      0   115k  0:00:03 --:--:--  0:00:03  115k
 40  375k    0     0   40  150k      0  90585  0:00:04  0:00:01  0:00:03 90532
 57  375k    0     0   57  214k      0  75169  0:00:05  0:00:02  0:00:03 75143
 68  375k    0     0   68  256k      0  69331  0:00:05  0:00:03  0:00:02 69313
 85  375k    0     0   85  320k      0  65601  0:00:05  0:00:04  0:00:01 65588
 96  375k    0     0   96  364k      0  63989  0:00:06  0:00:05  0:00:01 53512
100  375k    0     0  100  375k      0  53372  0:00:07  0:00:07 --:--:-- 41885
100  376k  100   110  100  375k     15  53012  0:00:07  0:00:07 --:--:-- 38134
{"status": "ok", "msg": "Update complete. Please wait for LED to resume blinking before disconnecting power."}
** UPLOAD SUCCESS. Flashing in progress.
** Please wait for LED to resume blinking before disconnecting power
========================= [SUCCESS] Took 16.71 seconds =========================
```

Through the webif; it uploads, flashes and says completed. Continues to flash rapidly, after a power cycle it is still on 2.0.1.

```
Firmware Rev. 2.0.1 (beff02)
```

Updating over serial works no problem, I can go back and forth between the two.

With platform.io:

```
Uploading .pio/build/HappyModel_RX_ES900RX_via_WIFI/firmware.bin
 ** UPLOADING TO: http://10.0.2.100/update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
 12  375k    0     0   12 48776      0   272k  0:00:01 --:--:--  0:00:01  270k
 33  375k    0     0   33  127k      0  94451  0:00:04  0:00:01  0:00:03 94382
 45  375k    0     0   45  170k      0  78610  0:00:04  0:00:02  0:00:02 78574
 62  375k    0     0   62  233k      0  69770  0:00:05  0:00:03  0:00:02 69750
 68  375k    0     0   68  256k      0  61435  0:00:06  0:00:04  0:00:02 61435
 90  375k    0     0   90  341k      0  63788  0:00:06  0:00:05  0:00:01 56718
100  375k    0     0  100  375k      0  61489  0:00:06  0:00:06 --:--:-- 52166
100  375k    0     0  100  375k      0  53012  0:00:07  0:00:07 --:--:-- 41777
100  376k  100   110  100  375k     15  52707  0:00:07  0:00:07 --:--:-- 37591
{"status": "ok", "msg": "Update complete. Please wait for LED to resume blinking before disconnecting power."}
** UPLOAD SUCCESS. Flashing in progress.
** Please wait for LED to resume blinking before disconnecting power
========================================================================== [SUCCESS] Took 9.19 seconds ==========================================================================
```

kasatka60, 2022-01-24:
I also have a ES900TX + ES900RX 868MHz, but everything was flashed without problems via Wi-Fi.

sfjuocekr, 2022-01-24:
Did you put the receiver in WiFi update mode and check the version on it?

sfjuocekr, 2022-01-24:
Uploading an updated build on 2.1.0 works.
Uploading any build on 2.0.1 seems to fail with the rapidly blinking.

```
Uploading .pio/build/HappyModel_RX_ES900RX_via_WIFI/firmware.bin
 ** UPLOADING TO: http://10.0.2.100/update
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
 28  375k    0     0   28  106k      0   104k  0:00:03  0:00:01  0:00:02  104k
 45  375k    0     0   45  170k      0  78681  0:00:04  0:00:02  0:00:02 78681
 51  375k    0     0   51  192k      0  65975  0:00:05  0:00:02  0:00:03 65953
 68  375k    0     0   68  256k      0  61709  0:00:06  0:00:04  0:00:02 61709
 85  375k    0     0   85  320k      0  65379  0:00:05  0:00:05 --:--:-- 65379
100  375k    0     0  100  375k      0  61812  0:00:06  0:00:06 --:--:-- 52951
100  375k    0     0  100  375k      0  53250  0:00:07  0:00:07 --:--:-- 42027
100  375k  100   110  100  375k     15  53001  0:00:07  0:00:07 --:--:-- 43995
{"status": "ok", "msg": "Update complete. Please wait for LED to resume blinking before disconnecting power."}
** UPLOAD SUCCESS. Flashing in progress.
** Please wait for LED to resume blinking before disconnecting power
========================= [SUCCESS] Took 16.68 seconds =========================
```

This was after manually flashing it with esptool:

```
esptool.py v2.8
Serial port /dev/ttyACM0
Connecting...
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 40MHz
MAC: e0:98:06:e3:79:4c
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 2MB
Erasing flash...
Flash params set to 0x0330
Took 2.06s to erase flash block
Wrote 385024 bytes at 0x00000000 in 38.9 seconds (79.3 kbit/s)...

Leaving...
Soft resetting...
```

Putting it in WiFi mode:

```
Firmware Rev. 2.1.0 (96c62a) FCC915
```

CapnBry, 2022-01-24:
Huh. I do not know what's happening then. The update succeeds but doesn't actually do the flash? I can't even imagine how that would happen.

The weird thing I am seeing is `Auto-detected Flash size: 2MB`? I thought all ESP8285 were 1MB. Is our update process trying to write to a place where flash doesn't exist and is therefore failing? Have you tried manually setting it to 1MB and then flashing over wifi?

pkendall64, 2022-01-24:
The H16 version of the chip does indeed have 2MB flash.

sfjuocekr, 2022-01-24:
Not to cause any confusion, the manual flash with esptool over serial works.

If I go back to 2.0.1 and try to do a WiFi upgrade it fails again.

It doesn't really matter, as the serial flash works as intended and flashing from the webif works on 2.1.0. Maybe if someone else runs into this problem, just use serial instead.

blockfeed, 2022-01-26:
@sfjuocekr 

I noticed something similar recently (on Matek 24-D), and I believe it may be related to an ExpressLRS Configurator local corrupt build cache. I was able to build binaries for UART, passthrough and WiFi using the "build binary" button, but only the binary for UART worked when flashed via FTDI. When using the passthrough or WiFi options, the binaries that were built had a different checksum than the valid UART binary and were garbage/invalid (caused the receiver to bootloop regardless of flash method).

Once I cleared the build cache and deleted the local PlatformIO temp directories manually, the binaries generated from WiFi build and Passthrough build all had the checksum of the working built-for-UART binary. Flashing then worked reliably from any method.

I'm happy to submit a bug report specifically for Configurator, if that would help.

sfjuocekr, 2022-01-27:
I never really used to configurator to flash, always used Platform.IO from within VSCode!

Was just testing :)

closed 2022-04-05 by github-actions[bot]