ExpressLRS/ExpressLRS #1346: trying to flash a new DIY mini 900mhz fails

mikiha81, 2022-01-24:
Hi when trying to flash a new board of diy 900mhz using betflight ( and pressing the boot button ) does not work from betaflight 
but when i connected to an external uart ftdi board , flashing works and updating from betaflight starts to work too . 

attaching logs of the flash from the configurator 
i also tried flashing from vscode . 

  context: {
    data: 'Checking size .pio\\build\\DIY_900_TX_ESP32_SX127x_RFM95_via_UART\\firmware.elf\r\n'
  },
  level: 'info',
  message: 'logs stream output'
}
{
  context: {
    data: 'Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"\r\n' +
      'RAM:   [==        ]  16.8% (used 55180 bytes from 327680 bytes)\r\n' +
      'Flash: [======    ]  56.3% (used 1106445 bytes from 1966080 bytes)\r\n'
  },
  level: 'info',
  message: 'logs stream output'
}
{
  context: { data: 'Configuring upload protocol...\r\n' },
  level: 'info',
  message: 'logs stream output'
}
{
  context: {
    data: 'AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa\r\n' +
      'CURRENT: upload_protocol = esptool\r\n'
  },
  level: 'info',
  message: 'logs stream output'
}
{
  context: {
    data: 'Looking for upload port...\r\n' +
      'Use manually specified: COM11\r\n' +
      'Uploading .pio\\build\\DIY_900_TX_ESP32_SX127x_RFM95_via_UART\\firmware.bin\r\n'
  },
  level: 'info',
  message: 'logs stream output'
}
{
  context: { data: 'esptool.py v3.1\r\nSerial port COM11\r\n' },
  level: 'info',
  message: 'logs stream output'
}
{
  context: {
    data: 'Connecting........_____....._____....._____....._____....._____....._____....._____\r\n' +
      '\r\n' +
      'A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header\r\n'

commanderguy3001, 2022-01-25:
are you sure you pressed the button all the time, especially during powering on the receiver?
and are you sure you connected the receiver uart rx/tx to the FC uart tx/rx in the right way (not connecting rx to rx and vice versa)?
also: was there enough power supplied to the receiver while you tried flashing it via passthrough?

mikiha81, 2022-01-25:
yes  the same procedure worked before . and after i flashed the rx with the UART tried it again with the fc and it worked . 
it seems only the first flash fails . and  i did the same flash with the uart flasher (pressed the button the same way ) and it worked .

pkendall64, 2022-01-25:
If you got this working, can you close the ticket please. Also there is a new PR #1351 that may have helped with this first flash problem.

mikiha81, 2022-01-25:
looks like PR #1351 is the culprit .. closing.

closed 2022-01-25 by mikiha81