ExpressLRS/ExpressLRS #1349: Baud rate for passthrough/UART should be configurable

sfjuocekr, 2022-01-25:
```
======== PASSTHROUGH INIT ========
  Trying to initialize /dev/ttyACM0 @ 420000
```

Baud rate is hardcoded to 420000, but some people like myself have receivers on slower UART's.

There could be a selection for this, taking the RCVR_UART_BAUD is not wise as a receiver might not be running at a lower baud rate.

commanderguy3001, 2022-01-25:
Why can't you run on a higher uart baud rate is an important question I think, or: why do you run it at a slower rate in the first place?

sfjuocekr, 2022-01-25:
Because there are devices that do not support the default, higher, baud rate.

For example, I use a Teensy board as a joystick with my receiver attached to it. It can't run 420000, so I have set RCVR_UART_BAUD to 115200.

The configurator allows changing RCVR_UART_BAUD but not the baud at which it is going to flash. I think this should be configurable as well.

I can flash it with esptool manually at any baud rate no problem.

CapnBry, 2022-01-26:
Ugh is all I have to say about this. The upload baud rate is defined by the PIO target definition, we have a handful of different speed it can run at. Just update your one-off joystick RX through wifi? You're making work for us! 😛

Also: This is strictly a Configurator thing I think.

sfjuocekr, 2022-01-27:
Yes I think so as well, I might look at it when I find the time. Never done any UI work :)

closed 2022-06-25 by github-actions[bot]