ExpressLRS/ExpressLRS #1350: Added configurator option to disable telemetry.

inouekou, 2022-01-25:
In Japan, receivers are not radio-certified, so they cannot transmit WiFi or telemetry.
Also, in Japan, many people use FUTABA transmitters, and telemetry cannot be turned off by lua script.

Please add an option to the ExplessLRS Configurator to turn off telemetry.

*Instead of commenting "I want this too", consider leaving a :+1: instead, to reduce spam and clutter*

inouekou, 2022-01-25:
I should have posted it to Configurator.

closed 2022-01-25 by inouekou

jurgelenas, 2022-01-25:
> I should have posted it to Configurator.

I have transferred your issue to the firmware repository. It is firmware issue first, not the configurator

pkendall64, 2022-01-25:
If you have a module with an OLED and joystick, you can disable telemetry from there.
This is not a configurator or firmware issue. The enhancement we made was to allow the user to set telemetry ratio in LUA or from OLED/TFT interface per model.

inouekou, 2022-01-25:
@jurgelenas Thank you so match.

@pkendall64 The only product that can be used in Japan is BetaFPV nano TX.
There is no certified product that can use OLED and joystick.
This means that neither Lua script nor OLED/TFT can be used with FUTABA transmitters.
This is due to the fact that Japan's radio wave certification system is unique compared to the rest of the world.