ExpressLRS/ExpressLRS #1358: Thor TX fan never stops once enabled

blockfeed, 2022-01-26:
## Current Behavior

The Thor TX unit has a thermal sensor which enables the fan at a specific threshold. The fan turns on, but never turns off regardless of time or setting change(s).

## Steps to Reproduce

1. Turn on Thor TX, use fly as usual. In my environment, enabling Dynamic Power and setting it to 500mw will usually kick the fan on within about 5 minutes.

2. The fan is enabled automatically at around 49C, approximately.

3. The fan never turns off. It is rather loud and unpleasant.

## Possible Solution (Not obligatory)

Power cycle the radio. After reboot, the fan is not enabled and does not start automatically (unless the proper temperature threshold has already been hit).

## Details

The thermal sensor appears to detect the high temperature, but the fans are never disabled even after returning to a normal temperature.

## Your Environment

- TX hardware: Thor TX (2.1.0)
- RX hardware: Matek 24-D (2.1.0, diversity enabled)
- Handset model: TX16S 
- OpenTX version (including nightly number) : EdgeTX 2.6
- ExpressLRS version (TX & RX MUST MATCH): 2.1.0
- Packet Rate: 500
- Telemetry Ratio: 1:64
- user_defines:

500mw, Dynamic Power enabled, 5M max bauds. Lock on first connection and binding phrase are enabled.

JyeSmith, 2022-01-27:
Probably not a bug, but because the temp never drops below the low temp threshold of 35C.

https://github.com/ExpressLRS/ExpressLRS/blob/02f73ac88b9e8a091b6cb4301969cfbb80654c14/src/lib/THERMAL/thermal.h#L25

blockfeed, 2022-01-27:
Thanks for the clarification, I'll close out this report. Appreciate your help!

closed 2022-01-27 by blockfeed