ExpressLRS/ExpressLRS #1382: After the elrs 2.2 version is out of control, it cannot be reconnected.

wsnfly, 2022-01-31:
## Current Behavior

## Steps to Reproduce

1.After the elrs 2.2 version is out of control, it cannot be reconnected.  Use the zorro remote and lightning receiver.  When the flight distance reaches 6 kilometers, it loses control due to obstruction, and it cannot be reconnected after losing control.  In the end, it was forced to land by digital transmission.  It can be connected normally after powering on again.

https://m.youtube.com/watch?v=YsR14p3QuWM

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:zorro
- RX hardware:thunder2.4
- Handset model:2.2.0
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

CapnBry, 2022-01-31:
Do you have `NO_SYNC_ON_ARM` enabled? I can't reproduce this and lose connection all the time.

wsnfly, 2022-02-01:
With this tag enabled, this phenomenon does not occur every time.  Can't reproduce at close range.  I'll test it at a distance with the full recording later.  I remember that the early beta version 915 also had this phenomenon

SunjunKim, 2022-02-01:
> With this tag enabled, this phenomenon does not occur every time

You HAVE TO DISABLE `NO_SYNC_ON_ARM` for any long-range flight. FYI.

closed 2022-03-12 by github-actions[bot]