ExpressLRS/ExpressLRS #1394: BetaFPV Micro TX 2.4 1000mW Unable to change setting

DimitarKrastev, 2022-02-02:
## Current Behavior

When the TX is connected to the RX, no settings modification is possible using the 5D joystick on the TX. Any change in settings causes the TX to halt and reboot causing disconnect from the model and the settings are not saved. Changing the settings with no RX connected seems to be working fine

## Steps to Reproduce

1. Power on BetaFPV 1000mW TX.

2. Power on RX and wait for connection.

3. Change any setting on the TX screen using the 5D joystick

## Your Environment

- TX hardware: BetaFPV 2.4 Micro TX 1000mW
- RX hardware: AxisFlying Thor 2.4
- Handset model: Taranis X9D Plus
- EdgeTX 2.6.0
- ExpressLRS version (TX & RX MUST MATCH): 2.2.0 (both)
- Packet Rate: N/A
- Telemetry Ratio: 1:2 (doesn't seem to make any difference)
- user_defines:
BINDING_PHRASE
FEATURE_OPENTX_SYNC
UART_INVERTED
UNLOCK_HIGHER_POWER

schugabe, 2022-02-02:
Your arming channel (=channel 5) needs to be LOW when disarmed and HIGH when armed. When the quad is armed the joystick is locked and you can't use the menu. Could you check if your arming channel is setup correctly and if arming/disarming does change anything of your described behaviour?

DimitarKrastev, 2022-02-02:
I know that. When the model is armed the joystick does nothing. When disarmed I can browse through the menu and change settings but if I change anything the TX reboots when I click the joystick to save. This doesn't happen if the model (RX) is not powered on.

I just checked that it kind of sometimes works and sometimes doesnt.

One thing to note. I am using ArduPilot with extended CRSF telemetry for Yapoo telemetry LUA widget (if that makes any difference).

My Telemetry ratio is 1:2 to provide smoother horizon on the widget. I am still testing, but if I have set a lower telemetry ratio (1:16) it seems less likely that a settings change crashes the TX. I am still verifying if thats a coincidence.

CapnBry, 2022-02-02:
```
tack:
0:
  0x401139a0: SPIClass::transfer(unsigned char*, unsigned int) at C:/Users/bmayland/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:269       
0:
  0x40081af5: SX1280Driver::GetIrqStatus() at C:\Users\bmayland\Documents\GitHub Projects\ExpressLRS\src/lib/SX1280Driver/SX1280.cpp:273
0:
  0x40081c10: SX1280Driver::IsrCallback() at C:\Users\bmayland\Documents\GitHub Projects\ExpressLRS\src/lib/SX1280Driver/SX1280.cpp:273
0:
  0x40081c45: SX1280Hal::dioISR() at C:\Users\bmayland\Documents\GitHub Projects\ExpressLRS\src/lib/SX1280Driver/SX1280_hal.cpp:343
0:
  0x40082e21: __onPinInterrupt at C:\Users\bmayland\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-gpio.c:274
0:
  0x4008f955: _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
0:
  [DATA (0x0x400621b6)]
0:
  0x40098ae7: esp_rom_spiflash_read_status at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:662
0:
  0x40098b22: esp_rom_spiflash_wait_idle at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:662
0:
  0x40098eca: esp_rom_spiflash_write_status at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:662
0:
  0x40098f37: esp_rom_spiflash_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:662
0:
  0x40090388: spi_flash_unlock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:155
0:
  0x40090543: spi_flash_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:381
0:
  0x4010ffa1: nvs::nvs_flash_write(unsigned int, void const*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_ops.cpp:72
0:
  0x4010e3fa: nvs::Page::writeEntry(nvs::Item const&) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:871
0:
  0x4010e732: nvs::Page::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int, unsigned char) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:871
0:
  0x4010cfd6: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp:579
0:
  0x4010babf: int nvs::Storage::writeItem<unsigned int>(unsigned char, char const*, unsigned int const&) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp:512
  \-> inlined by: nvs_set<unsigned int> at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp:385
0:
  0x4010c23d: nvs_set_u32 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp:512
0:
  0x400d34f1: TxConfig::Commit() at C:\Users\bmayland\Documents\GitHub Projects\ExpressLRS\src/lib/CONFIG/config.cpp:120
0:
  0x400d21e6: loop() at C:\Users\bmayland\Documents\GitHub Projects\ExpressLRS\src/src/tx_main.cpp:569
  \-> inlined by: CheckConfigChangePending at C:\Users\bmayland\Documents\GitHub Projects\ExpressLRS\src/src/tx_main.cpp:615
  \-> inlined by: loop() at C:\Users\bmayland\Documents\GitHub Projects\ExpressLRS\src/src/tx_main.cpp:1029
0:
  0x400da781: loopTask(void*) at C:\Users\bmayland\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
0:
  0x4009356e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

CapnBry, 2022-02-02:
@DimitarKrastev I'm so happy you included the extra information about your telemetry ratio because that is actually where the problem was. When we need to save the config changes, we have to stop all RF to not cause a crash. Our code hadn't considered what would happen if the TX's radio was already in receive mode, it just makes sure no more TX goes out and prevents _switching to_ receive mode. I've updated the code to also take the RF out of receive mode.

Can you test https://github.com/ExpressLRS/ExpressLRS/pull/1395 and verify this fixes the issue for you? To flash it, use the configurator and select the "Git Pull Request" tab. You can flash back to 2.2.0 when your testing is done if you like.

DimitarKrastev, 2022-02-03:
Oh wow. That was so freaking fast. Much appreciated. I will flash it right away and let you know.

Thanks a lot!

DimitarKrastev, 2022-02-03:
Unfortunately that kind of fixed it, but introduced a whole bunch of problems.

I flashed and power cycled the radio and the TX. I haven't updated the RX module (from what I saw in the commit this change only applies to the TX module if I am not mistaken).

The first time I tried to change a setting it rebooted just as before (I am sure I was already running the version from the pull request as it said Head I in the version). Every next attempt seemed to work just fine with 1 exception. Switching packat rate resulted in complete loss of link until I switched back to the original packet rate. I guess this could be due to the fact that I have LOCK_ON_FIRST_CONNECTION on the RX although I would expect if I change the packet rate on the TX while the link is active it would instruct the RX to do the same. My understanding is that this feature is only meant for quicker connection restore after a failsafe, not because a packet rate was changed. I flashed back to the official 2.2.0 and it behaves the same, so I suppose thats another bug entirely.

One big problem though is that one of the times I made a change in the settings the TX crashed completely and didnt reboot. I had to power down the radio. I think it was one of my attempts to change the packet rate, but I am not sure. I started changing random settings just to see if it is stable.

CapnBry, 2022-02-03:
Argh, you're right it is still happening it seems. I'll keep looking.

CapnBry, 2022-02-03:
@DimitarKrastev I've pushed a new change that fixes a problem with the code I added not firing all the time it needed to. The original code had a 40% chance to work, the first PR had a 60% chance of working, but theoretically now it should be 100%.

Can you try flashing again with this PR? Be sure to restart the Configurator if you still have it open from last time, because it seems it won't pick up the changes I've made unless it is restarted. The version in the Lua should read `HEAD ISM2G4 e38978` (be sure to exit the Lua and come back after updating as the version number does not update after the first load).

DimitarKrastev, 2022-02-03:
Confirmed on e38978. Changing TX power no longer reboots the TX but sometimes freezes it. It doesn't auto reboot so I need to power cycle the radio.

Also changing the packet rate doesn't work if the RX is connected. I still loose the link.

The occurance is less frequent though.

I have only flashed the TX. Is that correct or should I flash the RX as well?

CapnBry, 2022-02-04:
Yeah you only need to flash the TX, and you're correct about the `LOCK_ON_FIRST_CONNECTION`-- the RX should switch if the TX tells it to (but not if the TX reboots itself or locks up before the change). It does lose the link (it has to because the packets are not compatible across rates) but it either immediately syncs up <100ms or disconnects then reconnects at the new rate in a 1-2s.

I can't reproduce the behavior you're experiencing now though. I've not gotten any lockups at all in any of my testing, and I did literally 100 power level changes using the joystick just now and had no reboots or lockups at all with that PR. Do you have any other information or setting or unusual setup that might allow me to reproduce it?

DimitarKrastev, 2022-02-07:
Sorry, I just sold my Taranis (going to buy a TX16S) so I have no way to reproduce and provide any further feedback for now.

In any case it was much more stable, so you can close this issue and I will open a new one if I keep experiencing this problem.

Thanks a lot!

closed 2022-02-07 by CapnBry