ExpressLRS/ExpressLRS #1402: NamimnoRC FLASH 2400 OLED TX on version 2.2 no longer connect to RX modules

piwi3910, 2022-02-05:
## Current Behavior

after upgrading to 2.2 no connection is made between tx and rx (naminmo 2400 esp)
reverting back to 2.1 on the tx made the connection work again.

## Steps to Reproduce

upgrade tx module to 2.2

## Details

RX modules on 2.2 and TX back on 2.1 still works.
disabling model match made no difference

## Your Environment

- TX hardware: NamimnoRC FLASH 2400 OLED TX
- RX hardware: NamimnoRC FLASH 2400 ESP RX
- Handset model: Radiomaster TX16S
- OpenTX version (including nightly number) edgetx 2.6.0 fac0eed6
- ExpressLRS version (TX & RX MUST MATCH): 2.2
- Packet Rate: 500
- Telemetry Ratio: 1:32
- user_defines:

SunjunKim, 2022-02-05:
I have no problem with Namimno FLASH OLED TX & ESP RX on 2.2 (and 2.1 as well).
Could you elaborate on your situation in more detail?

piwi3910, 2022-02-05:
sure, i'm honestly not sure whats happening.
perfectly functioning solution on 2.1
megabaud 400k
model match
500hz
1:32 tel

updated the TX module to 2.2 -> RX still on 2.1 -> no more connection
TX module is still powered, oled working, and lua script sees the TX module, bit no longer the RX
no connection, no telemetry.

updated the RX module to 2.2 -> no connection
downgraded the TX back to 2.1  RX still on 2.2 -> connection asap
same config between both, except for the new option: Regulatory_Domain_ISM_2400
set the same on both RX and TX

RX config:
Regulatory_Domain_ISM_2400
binding phrase
lock on first connection
auto wifi on internal 60
home wifi ssid and password

TX config
Regulatory_Domain_ISM_2400
Binding phrase
Uart inverted
Feature opentx sync
auto wifi on internal 20
home wifi ssid and password

RX config

piwi3910, 2022-02-05:
flashed TX via the radio and via wifi
RX via betaflight passthrough and wifi

as far as i can see, the flash is working correctly.

cruwaller, 2022-02-08:
NamimnoRC OLED TX is based on ESP32 MCU so flash via radio is not possible. Do you mean flashed via USB or do you actually have an older STM32 version?

piwi3910, 2022-02-08:
correct, i flashed it via the usb plug, and via enabling wifi on the radio with the lua script.
in both cases the flash was successful, and the lua script saw the module with the 2.2 version on it.

closed 2022-06-23 by github-actions[bot]