ExpressLRS/ExpressLRS #1430: R9MM receiver turns off immediately

lowki, 2022-02-13:
## Current Behavior

Hye there, I'm trying to flash my R9MM receiver but after flashing the bootloader, the receiver turns itself off (all leds turn off) just after powering it up.
Therefore when I try to flash firmware via Betaflight passthrough, it fails as the receiver is turned off.

I flashed the receiver back to FrSky's firmware, and the receiver stays on even if it's not bound.

Did I miss something during the flashing process, is it a normal behaviour?

AlessandroAU, 2022-02-14:
The bootloader should only be active for a second after power-on, but it should actually detect that the main code is missing and stay in bootloader when you first flash. Try holding down the button as you apply power to persist the bootloader state.

lowki, 2022-02-19:
Thank you for the answer, I don't know what did the trick, but after another bootloader flash, my receiver stayed on.

Your advice should definitely be in documentation if it is a thing :)

closed 2022-02-19 by lowki