ExpressLRS/ExpressLRS #1440: Lua Settings not saved to EEPROM on R9M-Lite-Pro

namirda, 2022-02-18:
## Current Behavior
Any changes made to the default settings in the LUA script are lost when Tx is powered down

## Steps to Reproduce
Make changes to default settings in the lua script, power off and restart the TX and then check lua parameters.

## Details

This problem was first raised in [Issue 326](https://github.com/ExpressLRS/ExpressLRS/issues/326#issue-792754776) where it was  [stated](https://github.com/ExpressLRS/ExpressLRS/issues/326#issuecomment-793073905) that the settings were not being saved in that early version because the EEPROM is an SPI chip which is not supported in elrs. 

Subsequent discussion in that same thread  [here](https://github.com/ExpressLRS/ExpressLRS/issues/326#issuecomment-788787055) corrects that statement and says that the problem is solved and the PR merged.

However as of V2.2 the problem still exists

## Your Environment

- TX hardware: R9M-Lite-Pro
- RX hardware: R9MM
- Handset model: Taranis X-Lite
- OpenTX version: 2.3.14
- ExpressLRS version : 2.2.0  442a9d9
- Packet Rate: 200
- Telemetry Ratio: 32

pkendall64, 2022-02-18:
Does it save any of the settings? For example, the VTX administrator settings are global.

namirda, 2022-02-18:
No - VTX settings are not saved either.

pkendall64, 2022-03-26:
After looking at the datasheet for the EEPROM chip it seems that the address we are using (0x51) is incorrect and should be 0x50. If you are up for building and flashing the code from platformio, you could check out the `2.3.x-maintenance` and change the address on line 6 in `src/include/target/Frsky_TX_R9M_LITE_PRO.h` from 0x51 to 0x50 and see if it fixes the problem for you.

namirda, 2022-03-27:
I have cloned the ExpressLRS repo, checked out 2.3.x-maintenance, made the change you suggested to the EEPROM address and recompiled/flashed using the configurator 1.3.9 in local mode - unfortunately it did NOT fix the problem.

pkendall64, 2022-03-27:
Hmm, thanks for checking. Can you provide a closeup photo of the board where the EEPROM chip is?
The other thing you could try is values from 0x50 to 0x57. The final 3 bits have to match the wiring on the circuit board for pins 1,2 & 3 of the EEPROM chip. Or you could measure with a multimeter and see which pins are connected to GND (pin 4) and which to VCC (pin 8).

namirda, 2022-03-28:
I built several .elrs files with values of TARGET_EEPROM_ADDR ranging from 0 to 7 using the configurator but then discovered that they are all identical - ie they have the same SHA256 hash. Seems my changes to Frsky_TX_R9M_LITE_PRO.h are not being picked up in the build process. Any ideas??

[image]

pkendall64, 2022-03-28:
You will probably need to clean out the firmware files between each build.

namirda, 2022-03-28:
I have cleared the firmware files using the option in the configurator but have the same problem - the .elrs file produced by the build seems to be same whatever change I make to the include file. 

Sorry - I'm new to this stuff and don't understand what I'm doing wrong.

My exact steps were as follows: 1) I cloned the xlrs repo to a local folder on my machine 2) I checked out  the 2.3.x-maintenance branch 3) I made the changes to the .h file as you suggested and then 4) I built the firmware using the local option as shown in the screenshot in my previous post. The build runs successfully but the produced .elrs file has the same SHA256 hash in all cases.

c3r63ru5, 2022-06-09:
I'm having the same issue with saving to EEPROM on the FrSky R9 Lite Pro TX. I rebuild with TARGET_EEPROM_ADDR 0x50 (changed from 0x51) and the issue remains.

I just realised values are not being saved after using ELRS for months (started with v1) and after flashing 2.x. At first I thought the EEPROM might be defective. It could very well be that values were never saved, I would have to flash a second TX to be able to say for sure its not working. Is anybody successfully saving values to the TX EEPROM?

Picture of EEPROM attached for reference.

[image]

namirda, 2022-06-10:
@c3r63ru5 

Good that I am not the only one with the problem! Can you please compare the sha hash values of the two elrs file which you built with eeprom address 50 and 51. I found them to be the same!!

c3r63ru5, 2022-06-10:
> @c3r63ru5
> 
> Good that I am not the only one with the problem! Can you please compare the sha hash values of the two elrs file which you built with eeprom address 50 and 51. I found them to be the same!!

I'm using Visual Studio Code with the PlatformIO Extension to build. Without running the Clean task, the binary is not changed on Build (after changing the header file). When running Clean before Build, the firmware binaries will differ. I checked resulting binaries to make sure ;)

c3r63ru5, 2022-06-10:
> Hmm, thanks for checking. Can you provide a closeup photo of the board where the EEPROM chip is?
> The other thing you could try is values from 0x50 to 0x57. The final 3 bits have to match the wiring [...]

The EEPROM is labeled 402WT, hard to find a data sheet, could it be this one?
[M24C01/02-W M24C01/02-R M24C02-F](https://www.st.com/resource/en/datasheet/m24c02-w.pdf)

Pin 1 E0 Chip Enable 0
Pin 2 E1 Chip Enable 1
Pin 3 E2 Chip Enable 2
Pin 4 VSS Ground
Pin 5 SDA Serial Data
Pin 6 SCL Serial clock
Pin 7 WC Write control
Pin 8 VCC Supply Voltage

c3r63ru5, 2022-06-26:
@pkendall64 Anything I can do to help get this addressed? Is this a confirmed bug and EEPROM storage does not work for this TX hardware?

namirda, 2022-06-26:
I got my build problems sorted out using platformio and subsequently built 8 elrs files using a range of eeprom addresses from 0x50 to 0x57. None of them worked

N

pkendall64, 2022-06-27:
Would you be willing to try the `master` branch? I've been doing some tests and have found that the 2.x configuration structure is 272 bytes in size on the STM32, but the size of the EEPROM is 256 bytes, so who knows what will happen!
In `master` (3.x) we have re-structured and optimised the size of the config object and now it is under 256 bytes and this may help fix the problem.
I'm not 100% sure as I don't have a Lite PRO module, I only have an R9M module.

namirda, 2022-06-27:
I have done the same with the master branch - 7 builds using eeprom address 0x50 to 0x57. 
The elrs settings are still not saved after power off

c3r63ru5, 2022-06-27:
> Would you be willing to try the `master` branch? I've been doing some tests and have found that the 2.x configuration structure is 272 bytes in size on the STM32, but the size of the EEPROM is 256 [...]

I built and flashed master, also does not save settings properly. You wrote 2.x config struct is 272 bytes, and I remembered it worked in 1.x so I reflashed 1.2.1 to confirm. Saving to EEPROM worked with 1.2.1 but does not work with 2.5 and also not working with master (3.x).

c3r63ru5, 2022-07-09:
> master has definitely been broken saving things to external EEPROM for several months, I noticed it too when writing the config migration. With PR #1675 merged that should be resolved, but if it didn't work on 2.5 then it probably won't work now, as my fix only addresses the regression from 2.5 where it worked on the R9MM test I have. I don't have an R9M-Lite-Pro to test with though so I can't offer any more guidance. You might want to try latest master though to see if it has magic in it.

Gave the latest master a try to see if the merged #1675 fixes the issue with the R9MLP but unfortunately that does not resolve the issue.

CapnBry, 2022-07-09:
> Gave the latest master a try to see if the merged #1675 fixes the issue with the R9MLP but unfortunately that does not resolve the issue.

#1675 was merged to 3.x.x-maintenance, not master.

t0ffi, 2022-07-29:
> > Gave the latest master a try to see if the merged #1675 fixes the issue with the R9MLP but unfortunately that does not resolve the issue.
> 
> #1675 was merged to 3.x.x-maintenance, not master.

Maintenance is similar to RC? It is safe to use? Or wait for Official Release?

pkendall64, 2022-07-29:
The maintenance brach is where we are putting any bug fixes for the release, and it's looking stable so far.

closed 2022-11-13 by pkendall64