ExpressLRS/ExpressLRS #1445: PWM ELRS Receiver will not bind if 2 aileron servos connected using y leads.

anujchandkapoor, 2022-02-19:
Hi, I am using the Matek 2.4 ELRS PWM receiver (Native PWM, not the converter one).
Now as per the ELRS code channels 1-4 are outputting PWM with full res.
So to connect a simple plane I need the following:
1. Channel 1: 2 Alerons Servos using a Y lead
2. Channel 2: Elevator Servo
3. Channel 3: Throttle
4. Channel 4: Rudder
Now when I connect this the receiver does not bind, when I remove the 2 aileron sevos and plug it in again it works.
Please can someone look into this?

CapnBry, 2022-02-19:
I can not reproduce, using one servo or two with a Y lead it still connects just fine for me with SG90 servos.

It could be a problem with the servos pulling down the signal line, because if CH1 is held low on boot it enters the bootloader and will not connect to the TX and I suspect that is what is happening to you. Have you tried swapping the E and A servos to CH1 and CH2 respectively and see if that makes a difference? I know it no longer is AETR at that point but it might be one servo is ok but two is too much for the internal pullup on the boot pin.

anujchandkapoor, 2022-02-20:
Hi Bryan,Thanks for replying back. You may be right it looks like the case. What i did was set two output on the receiver as channel 1 from the wifi update page so now I have channel 1 being output from actual channel 1 and another channel one being output from channel 5 of the receiver, then i got rid of the Y lead and connected the individual servos on CH1 and CH5 of the receiver and it is working. Just was curious if this will mess up with the resolution on hardware CH5 not sure. I have another receiver on the eachien flying fish which has 4.3g digital servos connected using the Y lead and it works fine just that sometimes it connects once then after few seconds again.I am quite sure you are absolutely correct thats exactly what must be happening, also another point i noticed is that if I don't connect these 2 servos with Y leads during boot up and I connect it after few seconds it does work.Really appreciate you and your teams hard work and connection with the community. Thanks and do let me know if you want me to do some more experiments with this setup, just to help others who may have similar issues.Thanks and RegardsAnuj

CapnBry, 2022-03-10:
I've been able to reproduce this and also spoke with Matek about it, who is also able to reproduce it. This has to be fixed in hardware side though, since the receiver is not booting to our code at all in this condition.

anujchandkapoor, 2022-03-11:
I am glad I was able to bring this to your notice.For the time being the solution is to make the 5th channel as a duplicate 1st channel while flashing firmware. By doing this the receiver outputs the same on hardware channel 1 and 5 thus by allowing me to connect 2 aleron servo's individual to physical Ch1 and Ch5.But by doing this my 5 channel receiver becomes a 4 channel receiver.Unfortunately I bought 6 of these receivers because I really wanted to convert all my fixed wings to ELRS.Too bad 😞 but thanks for writing back to me, I am blown away by all the amazing work that you guys are doing.Thank You.RegardsAnuj.

CapnBry, 2022-03-11:
> I am glad I was able to bring this to your notice.For the time being the solution is to make the 5th channel as a duplicate 1st channel while flashing firmware. By doing this the receiver outputs [...]

Are you changing the actual firmware to do this? You can just duplicate the channel in the receiver's config (in the web interface). You also can just plug the wires in as RETAx and work around it, giving you back 5 channels. CH1 is affected but CH2 can also be pulled down by strong pulldowns in the servos, so just use CH3 or CH4 for servos that are so bossy about the signal line.

anujchandkapoor, 2022-03-11:
Oh that's a good idea. I will try that and let you know how it goes.And no i am not using custom firmware like you said I am using the web interface to duplicate channels but now as you suggested ill put aleron on ch2 or ch3.Thanks and RegardsAnuj.

CapnBry, 2022-03-13:
@anujchandkapoor can you unplug the servos from the receiver and measure the resistance between GND and the signal pin with your servos (on the y-lead, not just 1 servo)? I'm able to reproduce the problem with 2x servos with 12k pulldowns on each, bringing their combined impedance down to 6k on a y-lead. I just want to verify that my test setup that triggers the issue is in the same ballpark as yours to evaluate how to fix the hardware.

gretel, 2022-03-18:
> Matek 2.4 ELRS PWM receiver

ELRS-R24-P?

mgrennan, 2022-03-22:
Great work.  Do we know of the hardware fix?  I too bought a boatload of these for all my planes.  Can I create a transistor buffer and fix this?

CapnBry, 2022-03-23:
In my testing on the bench I was able to get it to boot by decreasing the value of the pullup resistor on the receiver I have (which is the crteensy diy design). I put a 68k pullup in parallel with the 12k on CH1 (bringing it down to 10.2k) and it would boot with the two servos connected. CH2 also needed a 12k pullup to boot with one of those servos, and 10k to boot with two of them. I believe the Matek has 12k on CH2 as already so it might also need to be brought down to 10.2k if you put a Y-lead on that one too.

So you could try putting 68k between 3.3V and CH1 output to see if that works for you, and possibly lowering that value until it works, but I wouldn't go below 10k in parallel with the existing 12k. If you hit that then I think a different solution would be needed.

MATEKSYS, 2022-03-23:
> In my testing on the bench I was able to get it to boot by decreasing the value of the pullup resistor on the receiver I have (which is the crteensy diy design). I put a 68k pullup in parallel with [...]

12K on CH1, 1K on CH2(TX)

MATEKSYS, 2022-03-24:
> Thanks, @MATEKSYS! However, I found that 12K on CH1 was not sufficient when using 2x servos which each had 12K resistance to ground when measuring their (disconnected) signal pin. On CH2 no problem: 1mohm, 490k. problem: 12k. If I put a 12k pullup before the series resistor going to output ch 2 (TX), the receiver boots with that 12k servo attached. If I add a second servo with a Y-lead to it, it doesn't boot. Lowering the resistor to 10k, it boots again. So CH2 should have at most a 10k pullup, although 1K seems pretty strong?
> 
> I put a 68k pullup in parallel with the 12k existing on the boot pad (brings CH1 down to 10.2k) and the two servos together work on CH1. So maybe lowering to 10k there may be enough or maybe 8.2k since it was closer to not working with 12k.
> 
> Again, this is all on the crteensy design and doesn't have the additional components the E24-P has already.

12K is not sufficient for 2x EMAX ES08MA servos, but 1K pullup on CH2 of R24-P is more than sufficient for 2x servos, you can hook up 5x ES08MA in parallel to CH2 without boot issues. :-)

diankurus, 2022-04-06:
> 12K is not sufficient for 2x EMAX ES08MA servos, but 1K pullup on CH2 of R24-P is more than sufficient for 2x servos, you can hook up 5x ES08MA in parallel to CH2 without boot issues. :-)

hi, Mateksys

any info about channel 5. in 2,4 pwm version ?

Im using matek 2.4ghz pwm one.
I flashed with matek 2400 pwm 2.3.0
Im using happy model es24tz
Channel 1: Alerons
Channel 2: Elevator Servo
Channel 3: Throttle
Channel 4: Rudder
Channel 5 : Servo to tailwheel.
Why in channel 5 not run in full resolution

channel 5 on my servo like jitters. but it wont happen when i use other receiver so it not servo issues. slight movement on stick will do full movement on servo

CapnBry, 2022-04-06:
@diankurus to get full resolution on CH5, go into the webui on the receiver and change the Output 5 to use something other than AUX1, and use `Wide` switch mode on the TX. Set up the handset's mixer to output to that new channel instead of CH1 (AUX1).

KevinFairgrieve, 2022-05-07:
I have a similar issue.
The TX and RX bind no problem.
Radio TX16S running EdgeTX 2.7
BETAFPV Micro RF TX module 1W
BETAFPV Micro receiver PWM
Both module and Rx updated to the latest Express LRS 2.4.0

I want to run a 4ch model using 2 servo`s in the wing, 1 servo for elevator, 1 servo for rudder and 1 for the ESC.

With the BETAFPV Rx only using 4 conventional channels an the fifth as a low resolution one.

Maybe this video will make it a bit clearer.

[https://www.youtube.com/watch?v=mrNGyFmWW0U](url)

CapnBry, 2022-05-08:
@KevinFairgrieve this is exactly what I've described above, the pullup on the output pin is not strong enough to counter the low resistance of the two servos in parallel. Moving the parallel servos to Output 4 should allow them to work, or if you want to mod the hardware with another resistor you can probably find a value that lets the RX boot in the channel order you prefer.

CapnBry, 2022-05-08:
> Why in channel 5 not run in full resolution

As Matek stated, CH5 coming from the TX only has two positions, just use CH6-CH12 instead and update the mapping in the receiver's webui to map your new input channel to Output 5. Make sure your TX is in Wide switch mode and 500Hz for the smoothest motion.

StanfRo, 2022-06-10:
Just for the record, I've got the same issues with the Happymodel ExpressLRS ELRS EPW5 2.4GHz 5CH PWM RC Receiver. It works with a "Y" cable and one servo connected but not both. I am happy to change my channel order to as a workaround (not tested yet).

wingunder, 2022-12-22:
Hi,

I know that this issue is relatively old, but nevertheless still open. I also know that most people that read these issues are advanced users. This post is aimed at the few that are not that advanced.
  
My proposal/workaround would be _not_ to use a Y-cable if you have a 2-servo aileron setup.
It seems like a lot of ELRS PWM modules only support 5 output channels. As channel 5 is a switch, we basically end up with only 4 full range PWM outputs for sensors. In this case a Y-calble on the ailerons would be very tempting, in which case the __receiver__ mapping would look as follows:

Elevator: input 1, output 1
Rudder: input 2, output 2
Throttle: input 3, output 3
Left Aileron: input 4, output 4 (via Y-cable)
Right Aileron: input 4, output 4 (via Y-cable)
Arm/disarm switch: input 5, output 5

If you're an absolute purist, who cannot imagine flying without channel 5 (AUX1), for arm/disarm purposes, then please stop reading and ignore the rest of this proposal. 

Like most of us know, ELRS uses the channel 5 input for arming, ALWAYS! So, ELRS treats channel 5 as a switch. This is a crucial point. Now, if you can manage to arm/disarm your model successfully from your radio, _and_ you can properly set up the fail-safe on the receiver, you might as well omit channel 5 (AUX1) altogether. This will free up channel 5, but there's a slight trick for using this channel for other purposes.  

I propose doing something like this on your __receiver__ mapping: 

Elevator: input 1, output 1
Rudder: input 2, output 2
Throttle: input 3, output 3
Left Aileron: input 4, output 4
Right Aileron: input __6__ (!!!), output 5  <----- Don't miss this.
NOTE: We don't have an arm/disarm output channel any more, as we're now ignoring the channel 5 input.

So, you'll need to send your ailerons to the __transmitter__ output channels 4 and __6__. These will arrive at the __receiver's__ input channels 4 and __6__, and will the __receiver__ will output them on ports 4 and 5.
The __receiver__ will not receive any channel 5 inputs from your __transmitter__.
Like this, you'll end up having 5 full range channels and no need for a Y-cable.

closed 2023-04-15 by github-actions[bot]

StanfRo, 2023-04-15:
Genius