ExpressLRS/ExpressLRS #1449: LOCK_ON_FIRST_CONNECTION bug (2.2.0)

SagittariusA33, 2022-02-21:
Few days ago updated from 1.2.0 to latest 2.2.0 and when switching the packet rate in script I get in situation where I cannot connect to receiver. It should work after powercycling the RX but it doesn't work even after powercycling both RX and TX. 

Flashed fw with LOCK_ON_FIRST_CONNECTION disabled and works just fine as expected. 

Am I missing something? Because it should reconnect again after powercycle when changing packet rate with this option enabled.

R9mm and R9 MX
R9M (pre-2019) 
Frsky QX7 accst
OpenTX 2.3.14
ELRS 2.2.0

StonedDawg, 2022-03-04:
do you have modelmatch on? can you upload please write all the parameter value you have from the lua script

SagittariusA33, 2022-03-14:
Model match is off, 100 Hz, 1:16 telemetry, hybrid switches.

R9M:
[image]

R9mm:
[image]

But like I said, disabling "lock on first connection" fixed it. I don't exactly need it, but maybe I'm not the only one facing this problem.

closed 2022-04-05 by github-actions[bot]