ExpressLRS/ExpressLRS #1469: Make dependency on AUX1 optional at Compile Time

MartinHugh, 2022-03-02:
### Discussed in https://github.com/ExpressLRS/ExpressLRS/discussions/1468

<div type='discussions-op-text'>

<sup>Originally posted by **MartinHugh** March  2, 2022</sup>
The Arm state influences the way the firmware works by affecting things like dynamic range and ground features. The Arm state is tied to AUX1.

Given the need to expedite channel information for some use cases, the use of AUX1 as the unequivocal Arming channel makes some sense.

However, this constraint makes two assumptions :

1) that FC Arming is needed in the first place. Many fixed/flying wing aircraft fly without Flight Controllers and do not need Arming state represented, and even if they do need it representing, it is often handled within the TX only, and not even transmitted.  Also some aircraft dont even have motors, like gliders.

2) that AUX1 already does not already have a dependency on something else required to fly.  ZOHD Kopilot is a RTH/stabiliser component whose flight mode channel (when passed by SBUS) is AUX1. This is not configurable, and use cases where FC arming is not needed, and a Kopilot would be used means that Express LRS cannot be used.

Allow features which are dependant on the state of AUX1 at runtime, to be overridden by compile-time options, so that AUX1 can be freed up for other use when not needed.

</div>

pkendall64, 2022-03-02:
Another of those features you talk about is the disabling of the 5-way joystick on OLED/TFT modules so you don't accidentally change some setting and cause a failsafe!

Also, you mention in the ZOHD Kopilot case "when passed by SBUS", which ELRS doesn't do so thats kind of a moot point then, right?

MartinHugh, 2022-03-02:
> Another of those features you talk about is the disabling of the 5-way joystick on OLED/TFT modules so you don't accidentally change some setting and cause a failsafe!

It would be good to get a full list of what is affected by the Arm switch.  In this particular case it smacks of sticking plaster after the fact over a badly thought out Joystick feature though, although I think disabling with Arm is a reasonable solution.

> 
> Also, you mention in the ZOHD Kopilot case "when passed by SBUS", which ELRS doesn't do so thats kind of a moot point then, right?
You are dead right.  Thanks for the heads up, completely moot point.

pkendall64, 2022-03-02:
IsArmed is used as follows

1. gSensor code to set minimum power if the unit is not moving (idle) and not armed (only Axis module currently)
2. Is sent to the LUA script to stop some settings from being modified when armed
3. OLED/TFT displays armed and stops the interaction with the joystick
4. Some thermal update code when not armed (only Axis module)
5. Don't allow VTX channel changes on combined RX/VTX
6. Used by dynamic power, if disconnected and armed - set power to max configured power
7. "no sync on arm" feature for racers
8. disables sending command to RX to start wifi if armed

MartinHugh, 2022-03-02:
> IsArmed is used as follows
>
> 1. gSensor code to set minimum power if the unit is not moving (idle) and not armed (only Axis module currently)
> 2. Is sent to the LUA script to stop some settings from [...]

Many thanks @pkendall64 very useful info.

MartinHugh, 2022-03-02:
For those not using Arming, I think most of those could be ignored.

pkendall64, 2022-03-02:
It depends on what you mean by "ignored"!

If you mean the code considers it always armed, then some of feature in LUA will not work and you won't be able to start wifi on a connected RX.

If you mean not armed, then dynamic power will not work as expected and the joystick could easily cause problems as BetaFPV found out!

MartinHugh, 2022-03-02:
> It depends on what you mean by "ignored"!
>
> If you mean the code considers it always armed, then some of feature in LUA will not work and you won't be able to start wifi on a connected RX.
>
> If you [...]

Good points.  I guess that is why enabling/disabling them at compile time would be ideal if you really did not want to set up arming.  But based on your two proposals I think not armed would work well for me.  Perhaps that could be a simplified option at compile time, to assume always armed or always unarmed.

I do understand that this sort of use case was not part of the original remit.  Just saying that ELRS has become such a force that it may want to consider it, because there are a lot of adopters who are off the originally considered user base.   That has to be a good thing :-)

StonedDawg, 2022-03-03:
@MartinHugh if you need "no arming required system " you can just assign a dummy switch for ch5/aux1

MartinHugh, 2022-03-03:
> @MartinHugh if you need "no arming required system " you can just assign a dummy switch for ch5/aux1

Of course, and that is what I am doing now, though that eats a precious PWM channel.

MartinHugh, 2022-03-04:
Also note that a number of PWM solutions only offer 6 outputs.  The first four are taken.
This leaves just two, and if one has be given away to indicate the armed status, that does not leave a lot for flaps/undercarriage etc.

There is actually one particular Matek PWM solution which only gives 5 channels.

What I am asking is that if the armed state can be set in firmware at compile time, it frees up the need to pointlessly sacrifice a PWM channel.

avsaase, 2022-03-04:
You can map another channel to that 5th PWM output. There is no need to put that 1 bit channel on a PWM pin.

MartinHugh, 2022-03-05:
> You can map another channel to that 5th PWM output. There is no need to put the1 bit channel on a PWM pin.

True if using integrated PWM boards, but mapping is not possible if using a conversion board like the Matek CRSF-PWM-6 or CRSF-PWM-C.  The firmware for these is the standard CRSF firmware and does not permit mapping.  I have a boat load of these and find it the restriction to throw away a good PWM channel unfortunate,

StonedDawg, 2022-03-05:
@MartinHugh yes you can. this is the parameter you look for (from mateksys' website) 
[image]

MartinHugh, 2022-03-05:
Thanks @StonedDawg for the heads up on this.   I have seen it and assumed it was only available via the iNav/BF CLI. as that is the section under which this is mentioned.  I dont use flight controllers - so that would not be an option.

But I note that there is a direct tool (Part A) which allows CLI.  It does not mention this command in its help list so I discounted it long ago.  Perhaps it is supported.  I will try it and find out.   This is not ideal though anyway because the RXs are already soldered to the boards, with the UART committed to the conversion function.  So I doubt this would be feasible without  disordering the receivers.

Thanks anyway.  One to consider when I buy more receivers at the very least.

StonedDawg, 2022-03-05:
@MartinHugh you have to use the UART2 which matek intended it for gps port.
I am not against having the special flag AUX1 to be configurable to other channel, but currently this is not critical issue, and who knows, maybe one day it will be configurable 😊.

MartinHugh, 2022-03-05:
> @MartinHugh you have to use the UART2 which matek intended it for gps port. this means you don't have to desolder the rx. I am not against having the special flag AUX1 to be configurable to other channel, but currently this is not critical issue, and who knows, maybe one day it will be configurable 😊.

@StonedDawg , thanks for your constructive input.  I recognise this is not a critical issue, and that mine may well be a marginal use case.  Nevertheless, I wanted to make the point known so that it may at least be considered in future.

CapnBry, 2022-03-11:
I just can't see us undoing the IsArmed behaviors since so many of our systems rely on them. Every system needs its own solution for how to know when to do or not do its thing. There's all sorts of ways to fake an arm switch (such as using any switch, or if throttle > 988) that I think are better solutions compared to every one of the things in [pkendall's awesome list](https://github.com/ExpressLRS/ExpressLRS/issues/1469#issuecomment-1057375713) just being broken if you compile out the IsArmed check.

I feel like allowing IsArmed to always return false breaks so many things that users wouldn't understand that when they turn off ARM checks then dynamic power doesn't work, etc etc, it would cause a lot of bad experiences when things don't work like they should.

MartinHugh, 2022-03-11:
> I just can't see us undoing the IsArmed behaviors since so many of our systems rely on them. Every system needs its own solution for how to know when to do or not do its thing. There's all sorts of [...]

Reasonable points I have to say.  If it were a clean sheet of paper, then the design might be able to approach from a different angle.  But such is the nature of design and development, once you have started down the road, there are always consequences to any modification.

Thanks for your comments.

closed 2022-06-23 by github-actions[bot]