ExpressLRS/ExpressLRS #1472: Unable to view RX settings in LUA

MartinHugh, 2022-03-03:
Express LRS version : 2.2.0
LUA Script                : latest 2.2.0 one.
Radio                        : Radiomaster TX16S
OS                             : EdgeTX 2.6.0
ELRS TX                     : HappyModel ES24TX 2400 TX
ELRS RXs                   : ES24TX EP1
Packet Rate               : 500
Telemetry Ratio         : 1:64

When I use the LUA script to drill into the RX settings, I select  "Other Devices" from the first menu, it then displays the RX name. 
When I select the RX name in that menu to display the RX settings, it shows a listing of some settings for less than half a second and then goes back to the menu showing the RX name.   If I try to select the RX name a second time it goes back to the first menu.

This used to work OK.

What am I missing here ?

Thanks for reading

MartinHugh, 2022-03-03:
A further note.   Neither can I access the settings of the HappyModel EP1s from the Crossfire LUA or the Agent Lite LUA.

Under those two I get a list showing the TX and the RX names together.  The TX one works OK, but when I select the RX I just get a blank screen.  Like the LUA cant get the data.

StonedDawg, 2022-03-03:
@MartinHugh there is no parameter for RX yet in 2.2, however you can try master branch for latest features which already has rx parameter like telemetry power (only rx that has PA), wifi, antenna select (only diversity rxes).

MartinHugh, 2022-03-03:
> @MartinHugh there is no parameter for RX yet in 2.2, however you can try master branch for latest features which already has rx parameter like telemetry power (only rx that has PA), wifi, antenna select (only diversity rxes).

Many thanks @StonedDawg.  If this is working normally that is good news, and this issue can be closed.