ExpressLRS/ExpressLRS #1475: Receiver won't connect when packet rate is set to 500Hz

dawr68, 2022-03-09:
## Current Behavior

Receiver won't connect when packet rate is set to 500Hz. It's working on 250Hz packet rate and below. 

## Steps to Reproduce

1. Turn on transmitter
2. Set 500Hz Packet Rate
3. Turn on receiver
4. There is no connection between them. Receiver LED blinking slowly (500ms on/off).
5. Change Packet Rate to 250Hz - receiver connects, everything is working.
6. Change Packet Rate back to 500Hz - receiver losses connection.

## Details

Whole setup is brand new. I tried different different antennas but no change in behavior.

## Your Environment

- TX hardware: Happymodel ES24TX PRO
- RX hardware: Matek ELRS-R24-D
- Handset model: Radiomaster TX16S
- OpenTX version (including nightly number): Edge 2.6
- ExpressLRS version (TX & RX MUST MATCH): 2.3 on TX and RX
- Packet Rate: 500Hz
- Telemetry Ratio: tested all settings - no change

SunjunKim, 2022-03-09:
It seems like a similar case with issue #1247.

Could you try removing those two caps around the XO?
https://github.com/ExpressLRS/ExpressLRS/issues/1247#issuecomment-1006528318

dawr68, 2022-03-11:
Removing caps solved my issue, so it was totally hardware related. Thanks

closed 2022-03-11 by dawr68