ExpressLRS/ExpressLRS #1482: After flashed the firmware, the TX Module automatically enters the WIFI state

YI-BOYANG, 2022-03-12:
I have a DIY 915 TX Module, and after brushing DIY 900 TX ESP32 SX127x E19 2.3.0 firmware, it will inevitably enter WIFI state after rebooting, and Lua cannot be used normally! Repeated repeated brushing firmware can not be solved!

YI-BOYANG, 2022-03-24:
> 

I can assure you that it was not a firmware burn error, and now I have found the cause: it was my computer! The same firmware target, burned with my friend's computer-generated firmware, returned to normal. But even if MY computer-generated firmware succeeded, the firmware wouldn't make it work, either 2.x or 1.x.

pkendall64, 2022-03-24:
So have you tried clearing the platformio cache and firmware files and rebuilding?

YI-BOYANG, 2022-03-24:
> 清除 platformio 缓存和固件文件

No, I will clear the platformio cache and firmware file first, then try to rebuild the firmware, thank you!

pkendall64, 2022-03-25:
> > 清除 platformio 缓存和固件文件
> 
> No, I will clear the platformio cache and firmware file first, then try to rebuild the firmware, thank you!

How did you get on? Can we close this issue as it's not a bug in the firmware?

closed 2022-04-07 by YI-BOYANG