ExpressLRS/ExpressLRS #1512: uploadforce does not work when flashing WI-FI targets

jurgelenas, 2022-04-06:
It looks like `--target uploadforce` does not work when flashing WI-FI targets:

```
{
  context: {
    args: [
      'run',
      '--project-dir',
      '"C:\\Users\\d3ad1ock\\AppData\\Roaming\\ExpressLRS Configurator Nightly\\firmwares\\github\\ExpressLRS\\src"',
      '--environment',
      'BETAFPV_Nano_2400_RX_via_WIFI',
      '--target',
      'uploadforce',
      '--upload-port',
      '192.168.0.150'
    ]
  },
  level: 'info',
  message: 'pio cmd'
}
```

Resulted in:

```
{
  context: {
    data: '\r\n' +
      'TARGET MISMATCH\r\n' +
      'Current target: BETAFPV_2400_RX.\r\n' +
      'Uploaded image: BETAFPV_NANO_2400_RX.\r\n'
  },
  level: 'info',
  message: 'logs stream output'
}
```

jurgelenas, 2022-04-06:
Pinging @CapnBry

CapnBry, 2022-04-06:
Huh. It should work, I'll take a look today.

CapnBry, 2022-04-06:
I've tested this with master and switching back and forth between targets which works fine using:
```
pio run --environment DIY_2400_RX_ESP8285_SX1280_via_WIFI --target uploadforce --upload-port elrs_rx.lan
pio run --environment HappyModel_EP_2400_RX_via_WIFI --target uploadforce --upload-port elrs_rx.lan
```
The issue is that `forceupload` relies on the devWifi.cpp supporting the force option directly (in one call), which is not in 2.3.0. There are few ways we can address this:

1. Use the `--target uploadconfirm` after the TARGET MISMATCH error in a two step process on the Configurator. This would require adding a button to the Configurator at the end of the mismatch upload to continue the process if a mismatch is detected.
2. Backport `uploadforce` to 2.x-maint. This would fix it going forward, but the target name change is already merged into 2.x-maint so it won't allow users to upgrade 2.3.0 to 2.3.1. We could solve this by reverting the target name change, backport uploadforce, release 2.3.1, then re-merge the target name change and release 2.3.2 at the same time. That allows an upgrade path, but is pretty convoluted.
3. Change `uploadforce` to a two-step process on the script side. This would first `curl http://elrs.local/?force=1` before attempting the firmware upload. This will work around the issue that blocks the single-step force upload, but adds a step that will be unnecessary for any other version other than this.
4. Just force the user to do via_BetaflightPassthrough or via_UART for this one scenario, since the former doesn't rely on any code on the target (just the upload script) and the latter always overwrites anyway. This is not a smooth user experience either though.

I'd say Option 1 was the way I envisioned this working originally, which puts the option to force change the target when it occurs instead of being an option you select before building. However, I understand that it is more complicated to implement.

Option 3 is probably the easiest to implement but leaves cruft moving forward. We could combine this with Option 2, so that only 2.3.x has this extra curl call and users could always go 2.3.0 to 2.3.x to 3.0 but not 2.3.0 to 3.0 directly if the target name changes?

I'm afraid I have more options than answers here. Since the target name changes are in 2.x-maint already, maybe implement 3 on master, then backport both to 2.x-maint and just ignore the fact that we're having to do something dumb (two curl calls). I'm open to better suggestions though.

deadbytefpv, 2022-04-06:
additional option on no.4: Build firmware, then Upload via Web Update page, and it will detect the mismatch, but "Update Anyway"..

closed 2022-11-27 by jurgelenas