ExpressLRS/ExpressLRS #1523: BLE Joystick under windows 10

adriancodre, 2022-04-12:
I have a Jumper T-Pro with internal ELRS module. When i enable the joystick capability of elrs, i can not find it in widnows 10 using aTP-LINK UB400 dongle, but on my android phone i see the joystick and can connect to it. What am i missing ?

mavenius, 2022-05-04:
I came for the opposite problem; what are you able to use the joystick for on Android? I can bind and connect to it, but it doesn't seem to be detected by any apps. Thanks!

adriancodre, 2022-05-05:
i did not try in any app, but it connected to it. i bought a cheap dongle and that worked under windows, but the distance was not great at all and the connection was not usable so cable is king :)

mavenius, 2022-05-05:
Cool, thanks for responding. I used BT with Velocidrone yesterday (using my laptop's interested BT.) It works, but it's a little laggier than when wired or using the XSR-SIM dongle.

Dr-Gandalf, 2022-06-01:
I have similar issue, it works ok on internal Bluetooth adaptor of my dell laptop and a Bluetooth enabled azus motherboard, but it is not detected by a generic usb Bluetooth dongle (V3+EDR)
[image]

dougshell, 2022-07-04:
I have found that the BT range on my Zorro is incredibly short (inches). Have you guys tested the offending hardware at obnoxiously close ranges?

taraskornuta, 2022-07-25:
I have an external BT module TP-LINK UB500 and HappyModel 2.4RX (Win10, 11). It successfully paired and I even can see the joystick in Windows device manager. 
But none of the three simulators can't work with this joystick. The sim crashes or hangs. 
Ardupilot Mission Planner also can't see the joystick

socalengineer, 2022-10-28:
I also have the same issue: Under windows using two different USB bluetooth adapters no bluetooth joystick was found (ASUS and TP-Link adapters).

 I have resorted to using a wire.

Under Mac OSX my laptop sees the Joystick just fine, I can even see the adapter with my iPhone 13 Pro.

Using a Happy Model ES24TX Pro.

Dr-Gandalf, 2022-11-01:
> I have similar issue, it works ok on internal Bluetooth adaptor of my dell laptop and a Bluetooth enabled azus motherboard, but it is not detected by a generic usb Bluetooth dongle (V3+EDR) [image]

I finally found the problem I had. My Bluetooth adaptor was only version 2.0, and this Bluetooth version is not compatible with Bluetooth LE. Solution, update the adaptor to Bluetooth version >= 4.0

jurgelenas, 2022-11-01:
> I finally found the problem I had. My Bluetooth adaptor was only version 2.0, and this Bluetooth version is not compatible with Bluetooth LE. Solution, update the adaptor to Bluetooth version >= 4.0

Bluetooth

jurgelenas, 2022-11-27:
> > I also have the same issue: Under windows using two different USB bluetooth adapters no bluetooth joystick was found (ASUS and TP-Link adapters).
> > I have resorted to using a wire.
> > Under Mac OSX my laptop sees the Joystick just fine, I can even see the adapter with my iPhone 13 Pro.
> > Using a Happy Model ES24TX Pro.
> 
> Thought I would post my update here that I had to get it to work--I tried what others on this thread mentioned to bring the Joystick closer to the PC, that worked. However, I'm only getting 5 channels (4ch for controls, and 1 switch). I need more switches to work any advice?

You need to setup that in your radio model settings.

jurgelenas, 2022-11-27:
Closing as all of this looks like it is related to the user environment / old bluetooth dongles.

closed 2022-11-27 by jurgelenas