ExpressLRS/ExpressLRS #1525: BETAFPV 2400 TX Micro 1000mW does/can not support EU_CE_2400

SolkGit, 2022-04-13:
## Current Behavior
Using ExpressLRS Configurator v1.3.11 there is a option to choose REGULATORY_DOMAIN_EU_CE_2400 (max. 10 mW output power) for BETAFPV 2400 TX Micro 1000mW device. Also in LUA script 10 mW are shown as maximum and actual possible power. But this tx module doesn't support 10 mW output power. It is a configurator UI bug.

## Steps to Reproduce
Using ExpressLRS Configurator v1.3.11
1. Choose under releases: 2.4.0
2. Choose under device category: BETAFPV 2.4 GHz
3. Choose under device: BETAFPV 2400 TX Micro 1000mW
4. Under regulatory domains you can also select EU_CE_2400 (max 10 mW)

## Possible Solution (Not obligatory)

Delete the option EU_CE_2400 for the named device in configurator UI.

## Details
- no details
- 
## Your Environment
- TX hardware: BETAFPV 2400 TX Micro 1000mW
- ExpressLRS version (TX & RX MUST MATCH): 2.4.0

CapnBry, 2022-04-14:
Just a note: this is a Configurator issue, but in master (3.0) the LBT implementation allows up to 100mW so effectively this will be resolved when that comes out. We may want to have the build error out `if MinPower > 10mW ` in 2.x-maint.

pkendall64, 2022-10-05:
Closing as 3.0 has LBT

closed 2022-10-05 by pkendall64