ExpressLRS/ExpressLRS #1544: How to confirm the ELRS RX module is fine

Ncerzzk, 2022-04-23:
Hi, I have a HappyModel_EP_2400_RX module , and I use it with  PX4, while after binding with my betafpv controller, the info send by HappyModel_EP_2400_RX seem to  be not recognised by PX4.

the PX4 can indeed recv some bytes on the uart port( for me is /dev/ttyS2), 
while it seems that PX4 could not find the sync bytes, like :

(uint8_t)crsf_payload_size_t::rc_channels + 2    (0x18)
(uint8_t)crsf_frame_type_t::rc_channels_packed  (0x16) 

I dump 1000 bytes from uart, and could not find these too bytes too. 

While because I'm not very familiar with the crsf protocol, there may be something I did wrong. Thanks very much if someone could help me.

Ncerzzk, 2022-04-26:
here is my test python script:
```python
import serial
import time
def serial_opera():
    com = "/dev/ttyS2"
    baudrate = 420000

    ser = serial.Serial(com, baudrate, 8)
    if(ser.isOpen()):
        print("open!")
    result=""
    f=open("crsf.data","w+")
    arr=ser.read(1000)
    map_result=map(hex,arr)
    for i in map_result:
        result+=i+" "
    f.write(result)
    f.close()

serial_opera()
```

and the data:
```
0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x5e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x5e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x84 0x8a 0x8d 0x0 0x32 0x47 0x40 0x42 0x40 0x40 0x40 0x40 0xfe 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x5e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x5e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x5e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x5e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x84 0x8a 0x8c 0x0 0x72 0x23 0x20 0x21 0x20 0x20 0x20 0x60 0xf9 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x5e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x1 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0x22 0x5e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0xd0 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x8c 0x8a 0xf0 0x81 0x5c 0x2 0x87 0xb8 0xf8 0xfe 0x81 0x70 0x70 0xd0 0xc4 0x90 0x84 0x80 0x80 0xa2 0x1e 0x71 0xe2 0xe0 0x84 0x8a 0x8d 0x80 0x32 0x47 0x40 0x42 0x40 0x40 0x40
```

Ncerzzk, 2022-04-26:
> 
Hi, I just directly connect the RX module to the uart port of Orangepi(http://www.orangepi.org/) which is similar with raspberrypi.

SunjunKim, 2022-04-26:
I have what have been wrong on your setup, but here's a demo of a normal packet looks like from a RX under a connection:

```
C8 18 16 E0 03 1F 2B C0 F7 8B 5F FC E2 17 E5 2B 27 5F F9 CA 07 00 00 44 3C E2 B8 
C8 dest = CRSF_ADDRESS_FLIGHT_CONTROLLER 
   18 len = 24 bytes (22 byte payload + 2 byte CRC)
      16 type = CRSF_FRAMETYPE_RC_CHANNELS_PACKED 
      	 ## ## ## ## ## ## ## ## ## ## ## ## ## ## ## ## ## ## ## ## ## ## payload: 22 bytes
      	                                                                   ## ## CRC
```

Ncerzzk, 2022-04-27:
Thank you for providing the demo packet. I will use another rx module to test.

StonedDawg, 2022-04-27:
@Ncerzzk im not sure about PX4, but it is linux, so maybe also check the endianess of bytes?

Ncerzzk, 2022-04-27:
@StonedDawg  Thank you, the endianess of my CPU is little-endian.  While because I dumped the bytes one by one and still couldn't find the sync bytes, so it may be not the cause.

Ncerzzk, 2022-05-02:
Hi @SunjunKim @StonedDawg , I have find the root cause.
For default uart clock setting in linux(like raspberryPi or other Pi), the default baudrate of ELRS(420000) could not be produced. Because the default uart clock source is not high enough.

And there are 2 ways to resolve this problem:
- higher the uart clock source
- edit the baudrate of ELRS rx module to a baudrate that your board could produce. For me, I changed it to 375000 or 500000, and it work normally.

Ncerzzk, 2022-05-02:
issue closed , thank you .

closed 2022-05-02 by Ncerzzk

StonedDawg, 2022-05-02:
@Ncerzzk thanks for reporting back! this will help us in case someone else has similar problem. 😁 good thing we've had a custom rx baudrate define.