ExpressLRS/ExpressLRS #1546: Es900 rx  Servo output issue on non critical outputs. Over channel 4

AngelisFpv, 2022-04-25:
Hello, I’m having an issue where all my channels that aren’t being used for my ailerons ,elevator, rudder and throttle are not allowing for precise control for my servos for my camera and flaps. They do not center properly and I cannot add precise offsets via my radio or any slow down or up. Any mixes on my radio act differently when seen on receiver tab on my flight controller. HELP!! They work fine for switch outputs however…. I’m using the r9m with elrs and the happymodel es900 rx

pkendall64, 2022-04-25:
You should be asking on discord in the help-and-support channel. There will be a lot more people there that can help you.
But the first question would be, are you using wide switch mode? because that will give you more resolution on the AUX channels. See https://www.expresslrs.org/2.0/software/switch-config/

AngelisFpv, 2022-04-26:
> You should be asking on discord in the help-and-support channel. There will be a lot more people there that can help you. But the first question would be, are you using wide switch mode? because [...]

Im using hybrid however it does not allow me to change yo wide. It just switches itself back. Will do thanks!

commanderguy3001, 2022-04-26:
> Im using hybrid however it does not allow me to change yo wide. It just switches itself back. Will do thanks!

You won't be able to change the switch mode as long as your receiver is connected, so that might be your issue, so try to turn every receiver off and change it then.

If that still doesn't work I too would recommend you to hop on the discord and ask in the #help-and-support channel over there.
Link: [ExpressLRS](https://discord.gg/expresslrs)

closed 2022-06-04 by github-actions[bot]