ExpressLRS/ExpressLRS #1548: Script Syntax error on lua with x9d+

spyder0069, 2022-04-26:
## Current Behavior
Trying to run the latest elrsV2  lua script on a x9d+ running 2.7.0 gives:

Script syntax error
Tools/elrsv2.lua:7
unexpected symbol near '<'

pkendall64, 2022-04-26:
That would mean that you have not downloaded the LUA correctly. Probably downloaded from github in html form hence the `<` character.
Please ask for help on discord #help-and-support channel.

spyder0069, 2022-04-26:
I am not 100% sure which source I got that lua from so to confirm I just downloaded the lua through the expresslrs configurator.  I deleted the original and copied the new one to the SD.  It gives a different error on the x9d+:

Script syntax error
not enough memory

spyder0069, 2022-04-26:
This x9+ is running edgetx 2.7.0.  I have not tried on opentx.

StonedDawg, 2022-04-27:
@spyder0069 you can try turn off the x9d and run it again, it might be out of memory on 1st launch only. if you still can't solve this, please jump to discord for further guidance.

spyder0069, 2022-04-27:
This morning I powered up the radio and the lua worked so turning off the radio and back on must have corrected it.  Thanks for the help.

closed 2022-05-15 by github-actions[bot]