ExpressLRS/ExpressLRS #1562: Betafpv PWM receiver not arming standard plane ESC

Mookie-Man, 2022-05-06:
Im new to the ELRS world and i fly fixed wing. I've just got the betafpv 500mw tx module and betafpv 2 x pwm receivers today. I was proper excited until finding that the ESC will not arm so get no throttle once the battery is connected. It seems that the receiver takes to long to connect to the module and by that time the esc thinks there is no signal so the throttle don't start or work there on in. Reconnecting the battery does the same thing. I have found a workaround which involves removing the red cable from the esc and connecting a separate BEC to another battery to power the receiver making sure that the receiver is connected to the module then plugging in another battery to the esc to power the motor. This is the only way it will work. This however is not ideal. I've also noticed that the servo movement is not very smooth with slow stick movements. The ESC is quite an old hobbywing guard and the servos are HXT900`s but all is working fine through a FRsky receiver...TBH im a bit bummed about this cause as soon as i see there where pwm receivers i bought them straight away as this project is so dam cool...Ive updated both module and receivers to 2.4 Also using a TX16s with edge 2.7. Ive played with all settings in the lua to know avail...

commanderguy3001, 2022-05-06:
> servo movement is not very smooth with slow stick movements
what packet rate are you running?
(you can check that in the expressLRS lua script)
also, what switch mode are you running?
and one last question: do you have the arm switch on channel1, and what channels do you use in general?

Some more info for you to look at:
https://www.expresslrs.org/2.0/software/switch-config

Mookie-Man, 2022-05-06:
Thanks for your reply. Tried all packet rates. 50-500hz no difference. Also tried both wide and hybrid switch mode. This is just a plain old Hobbywing guard 30amp esc for planes. don't have an arm switch just plug the battery in and the built in BEC powers the receiver. I get the 3 beeps to state the the number of cells then there should be another beep for the esc to initialize, but like i said the receiver does not connect in time. once it does all servos work but no throttle. 
Channels used are standard AETR  ch1 aileron (only one servo connected for testing) ch2 elevator  ch3 Throttle ch4 rudder . Set up on the receiver standard 1,2,3,4,5(Aux) 

I have also plugged and bound the receiver to a delta wing model which has a new Hobbywing Skywalker 50 amp esc and 2 x ES08MAII servos. On testing, of which I connected and disconnected the battery 10 times this esc would started up 8 times out of 10 and I'm pretty sure that the closer the TX was to the receiver the quicker it connected. Also with these servos the slow stick movements did seem a lot smoother. 

This ESC and servos on the original test are pretty old, around 10-11 years old, Maybe that's the problem with this old bird.???

commanderguy3001, 2022-05-06:
Sounds like the age of that gear might indeed be your problem, especially since the receiver seemed to work with different hardware.
But I also do remember that some people had somewhat similar issues recently regarding pwm, and @CapnBry reworked the whole PWM code to fix it.
I don't know if that is included in the current release though, or if you would have to flash master for that.

Mookie-Man, 2022-05-06:
I really don't want to change all my servos. The HXT900 is such a great little servo and i have loads installed on various planes.. They work great with all other receivers. Suppose its just the way it is. 

Shame, because its getting harder to find a cheap RF solution with the range and telemetry options of this project. All d16 stuff is pretty much gone now. Seems like there is no point in the 4 IN 1 multi now. Don't leave much for a fixed winger apart from the big boys now. I`d rather give my cash to these open source projects any day as its not about the money. Be nice for a branch for us fixed wingers with 6-8 full res channels.

Whoever is involved great work.

commanderguy3001, 2022-05-06:
you could try and flash master and see if it works then, as I said, that has a completely new and better pwm implementation

CapnBry, 2022-05-06:
The updated PWM code is in PR, not in master:
https://github.com/ExpressLRS/ExpressLRS/pull/1558

Some ESCs are really picky about their startup signal and maybe some of that is addressed in the code above, however I tested every ESC I own and none of them have this issue so I have nothing to test against. That said, there may not be much we can do about it if the ESC doesn't like our startup sequence after the PR. You can also try different output pins on the RX as some of them are driven differently in hardware.

Servos generally all work unless their input impedance is too low, or is made too low by running them in parallel. This can be fixed with adding resistance between the signal and 3.3V line to "pull them up". This has been recommended to all manufacturers but I'm not sure any of them have done this yet. The HXT900 should work, I've tested with SG90s and other random 9g servos and they all work. If you're having issues with all servos and ESCs on every pin you might want to check that your RX is even working because that's highly unlikely that nothing works with it.

Mookie-Man, 2022-05-06:
Thanks for you reply guys. CapnBry on your branch, what HZ rate should I chose for the HXT servos.? and should i use the same rate for the ESC channel.?

Mookie-Man, 2022-05-06:
Ill try it this evening (UK time) and let you know how I get on. You wouldn't get this service from horizon. lol. Thanks again.

CapnBry, 2022-05-06:
> One other thing. Do you own any Hobbywing ESC`s and was your testing done on any.?

Nope

Mookie-Man, 2022-05-06:
CapnBry Do i have to compile and flash the tx with your branch as well as the rx..I have flashed the receiver and it will not bind.?

Mookie-Man, 2022-05-06:
After some testing I've found that the packet rate needs to be on 500hz. The receiver seems to connect a lot faster and the esc arms every time. Like you said in a previous post though having 2 aileron servos connected on a y cable or having pin out 5 set to channel 1 the servos do not work. but if you connect the y cable after the receiver is powered on they all work. Havent tried your branch yet as waiting for the reply as to whether I need to flash the tx.

closed 2022-06-23 by github-actions[bot]

0crap, 2022-06-25:
Same issue here.

Old foamy with a Turnigy Plush 18A ESC.
Definitely related to the time it needs for the RX to connect to the radio.
When I use 150Hz the ESC will never work.
At 250Hz you have a 40% chance the ESC will work.
At 500Hz it always works. (Fastest connection time to the radio at power on the RX.)

The RX used is a Matek ELRS-R24-P with ExpressLRS v2.5.0