ExpressLRS/ExpressLRS #1563: BLE Joystick doesn't work under Android

CurryKitten, 2022-05-09:
## Current Behavior

I've tried using the BLE joystick on an Android device.  I find that I can connect it to my Android tablet, but I can't get it to appear as a joystick when testing with a sim.  I've noticed in the Dsicord group that a few people attempted to use it on Android, and also couldn't seems to get it to work using different sims and a bunch of gamepad testers.  In these cases of it failing on Android, it seems absolutely fine on Windows/Mac with the same radio.

## Steps to Reproduce

Put your radio/module into BLE joystick mode, pair it with your Android device - attempt to use a SIm - the Sim doesn't see any joystick connected.  I was using (and a few of my users) my own CurryKitten FPV Simulator, I saw another using Freerider.  Both these sims use Unity as their Game engine.  unity doesn't have any special support for bluetooth devices, it simply lists the available joysticks on the system.  If a bluetooth device has been paired as a joystick, then it should appear as a joystick in the sim.  

1.

2.

3.

4.

## Possible Solution (Not obligatory)

I don't know enough about the BlueTooth stack to suggest anything here, but given that the code appears to work fine on Windows/Mac - I assume there's something quite specific about Android that perhaps needs a more specific definition

## Details

## Your Environment

- TX hardware: RadioMaster Zorro
- RX hardware: N/A
- Handset model: RadioMaster Zorro
- OpenTX version (including nightly number) Edge TX 2.6.0
- ExpressLRS version (TX & RX MUST MATCH): 2.3
- Packet Rate: N/A
- Telemetry Ratio: N/A
- user_defines: N/A

commanderguy3001, 2022-05-10:
unfortunately I'm not qualified enough to talk about the bluetooth stack, but just from general knowledge/intuition:
it might be android that hates ANY controller connected over bluetooth, to test that: did you try any other (non elrs) bluetooth controller with the sims?

otherwise it might also be unity that can't work with joysticks on android in particular (for whatever reason, but nothing can surprise me anymore when it comes to unity...)

CurryKitten, 2022-05-10:
Yes - so if I pair an Xbox One controller (for example) that is detected and works fine in the sims.  I say fine, obviously, it's horrible compared to the sticks on a radio, but it works.

CurryKitten, 2022-05-11:
The only bluetooth controller I found which isn't a big brand console controller is made by Sminiker - but seems to act a bit differently.  Whist both the Xbox One controller and ExpressLRS Joystick have a little gamepad icon next to them in the Bluetooth setting in Android, the controller shows up as a "S600 Gamepad" with a keyboard symbol and lets me move around icons on the desktop and start up applications.  It does get picked up as a joystick in the sim, but I can't pick up any movement from the axis... but probably because it's acting as a keyboard for some weird reason.  At this point I've run out of Bluetooth controllers to test though

pkendall64, 2022-05-17:
> I've made a BLE gamepad using an ESP32 with the BLEGamepad library and it works on my android phone (with Curry Kittens sim and FPV Freerider). I've had a quick glance through the code and they're pretty similar, but I'm happy to post my code if someone with an ELRS controller wants to look. I'm using ESP32-BLE-Gamepad-0.3.3 Tested on Android 11 (Miui 12.1.4) and Android 10 (Cyanogeod)

Yes please, we're happy to take a look at the differences

peteallenm, 2022-05-18:
Apologies, this is rather clunky as I removed a load of other functionality for this test.
I can confirm this still works on my phone however.

The big functional difference I can see with your code is that I had to add some code to only send if a value had changed. Without that (even at 100hz) the controller got more and more laggy until the axis stopped responding. However I just tested without this at 200hz, and for the 20 seconds of my test it worked perfectly.
[FPVGamePadNoWifi.zip](https://github.com/ExpressLRS/ExpressLRS/files/8713648/FPVGamePadNoWifi.zip)

arthurson, 2022-11-22:
[image]

i have gamesir g4s for bluetooth joystick, it works for freerider and flyskydive, while elrs wont as usual

rickard-von-essen, 2022-12-27:
I had a similar issue with connecting a BetaFPV LiteRadio 2 (old version) to an Android device over USB.

After some research it turns out that Android is very picky about what USB HID game pads it allows, if the HID descriptor doesn't match those criterias it will not show up on Android.

For the LiteRadio 2 the problem was that it defines Sim Ctrls/throttle and rudder instead of Generic Desktop Control/X, Y, Z which Android doesn't recognize as a proper gamepad. 

Maybe something similar applies to Bluetooth too.

PR #2050 (2023-03-07): Update BLE Joystick
This PR aims to fix #1563 and does further improvements to the BLE Joystick.

Android does not understand simulation controls like rudder or throttle.
To achieve a higher compatibility a **generic gamepad with 8 axis and 16 buttons** is used instead of simulation controls. 

**CRSF Channel mapping**:
- first 8 CRSF channels are mapped to axis 1-8 in this order: x, y, rx, ry, z, rz, slider1, slider2
- remaining 8 channels are mapped to buttons 1-8
- the user can modify the model's outputs to remap channels
- 3 way switches can be mapped to an axis to receive a meaningful out [... cut]
closed 2023-03-07 by pkendall64