ExpressLRS/ExpressLRS #1591: wrong fimware

shijianzhong, 2022-05-26:
## Current Behavior

wrong firmware
[image]

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details

## Your Environment

- TX hardware:
- RX hardware:
- Handset model:
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH):
- Packet Rate:
- Telemetry Ratio:
- user_defines:

shijianzhong, 2022-05-26:
[image]
[image]
[image]
[image]

pkendall64, 2022-05-26:
This is not a bug. Please ask on discord in the #help-and-support channel.
This target is actually an alias of the DIY target and it looks like you are running a very old version of the firmware on the RX.
I suggest you update using an FTDI adapter.

closed 2022-05-26 by pkendall64

shijianzhong, 2022-05-26:
use wifi， not ftdi