ExpressLRS/ExpressLRS #1635: VRX Backpack weird behaviour when high RF noise

jiridoko, 2022-06-15:
I was taking part in a drone race last weekend (in Laxton FPV club in the UK, doubt that's relevant but here you go) and a lot of people were running 2.4GHz radios (ghost, elrs, wifi, bluetooth, frsky etc.). My LQ between the tx and my quad was not great but totally flyable. My ELRS VRX backpack on the other hand - that was causing a lot of trouble. It kept changing the channel on my Rapidfire module all the time. I was worried it would change the channel while I am in the air. It didn't actually change the channel and after my first race I cut off the backpack. The rapidfire module kept beeping all the time - the sound it makes when changing channels. My goggles were on my face and the transmitter hung on my neck strap. I can confirm that the backpack works just fine in any other scenarios, when I was at the race (arguably when I needed this functionality the most), it did not work.

What solution is there for this? Are there vrx backpack modules (or is it a problem with the tx module's backpack?) with external antennas instead of the tiny PCB antenna? Is there a way to disable the backpack temporarily?

ELRS tx 2.4.0
ELRS rx 2.4.0
ELRS tx backpack 1.1.0
ELRS vrx backpack 1.1.0
TX module: Namimno Flash 2.4GHz
Backpack module: Happymodel VRX-Backpack EP82
RX module: Happymodel EP1 (ceramic antenna)

JyeSmith, 2022-06-15:
There was an issue in the BP V1.0 release which was corrected in V1.1 https://github.com/ExpressLRS/Backpack/pull/65 . So please double check youve updated.

RF also likes to beep every time is gain/looses video sync.  Was the beeping different to this?

ELRS V2.x also has a guard that when armed it will not send the VTx, or backpack, change channel command.  One possibility is your Arm switch is not high on arm.  Something to check just in case.

YI-BOYANG, 2022-06-16:
> There was an issue in the BP V1.0 release which was corrected in V1.1 [ExpressLRS/Backpack#65](https://github.com/ExpressLRS/Backpack/pull/65) . So please double check youve updated.
> 
> RF also likes to beep every time it gains/looses video sync. Was the beeping different to this?
> 
> ELRS V2.x has a guard that when armed it will not send the VTx, or backpack, change channel command. One possibility is your Arm switch is not high on arm. Something to check just in case.

  
Yes, I prefer the ARM range to be low, because I also had problems changing VTX parameters in flight, when I set the ARM range to high, it never happened again.

closed 2022-06-29 by github-actions[bot]