ExpressLRS/ExpressLRS #1667: 3.0 RC1 Wifi TCP BF connection stop working when activate GPS and ESC protocol

RipperGOT, 2022-07-03:
If you leave the FC in default settings, the connection works fine. But when you activate the GPS + ESC protocol, it stops working

## Steps to Reproduce

Set defaults in the FC. Config RX provider CRSF, connect via TCP, set GPS, set ESC-------- Link Fail 
Set defaults in the FC. Config RX provider CRSF, connect via TCP, set ESC, set GPS-------- Link Fail 

- RX hardware: HM EP1 x3
- FC : Matek F722 STD, Matek F722 PX, Iflight Succex E F7

CapnBry, 2022-07-09:
I don't know if even those items are needed to reproduce this, as I can't get it the MSP over WiFi to work at all on release Betaflight 4.3 with my test FC.

CapnBry, 2022-07-25:
Turns out this is because Betaflight is overflowing the size of its buffer (320 bytes) for the BOXNAMES MSP command
[image]

GPS enabled and ESP is PWM: BOXNAMES size is 317 bytes
```
crsf2msp e=0 n=1 s=7
C8 3E 7B EC C8 37 FF 74 3D 01 41 52 4D 3B 41 4E .>{..7.t=.ARM;AN
47 4C 45 3B 48 4F 52 49 5A 4F 4E 3B 48 45 41 44 GLE;HORIZON;HEAD
46 52 45 45 3B 46 41 49 4C 53 41 46 45 3B 47 50 FREE;FAILSAFE;GP
53 20 52 45 53 43 55 45 3B 48 45 41 44 41 44 95 S RESCUE;HEADAD.
crsf2msp e=0 n=0 s=8
C8 3E 7B EC C8 28 4A 3B 42 45 45 50 45 52 3B 4F .>{..(J;BEEPER;O
53 44 20 44 49 53 41 42 4C 45 3B 54 45 4C 45 4D SD DISABLE;TELEM
45 54 52 59 3B 42 4C 41 43 4B 42 4F 58 3B 46 50 ETRY;BLACKBOX;FP
56 20 41 4E 47 4C 45 20 4D 49 58 3B 42 4C 41 16 V ANGLE MIX;BLA.
crsf2msp e=0 n=0 s=9
C8 3E 7B EC C8 29 43 4B 42 4F 58 20 45 52 41 53 .>{..)CKBOX ERAS
45 20 28 3E 33 30 73 29 3B 43 41 4D 45 52 41 20 E (>30s);CAMERA
43 4F 4E 54 52 4F 4C 20 31 3B 43 41 4D 45 52 41 CONTROL 1;CAMERA
20 43 4F 4E 54 52 4F 4C 20 32 3B 43 41 4D 45 3D  CONTROL 2;CAME=
crsf2msp e=0 n=0 s=10
C8 3E 7B EC C8 2A 52 41 20 43 4F 4E 54 52 4F 4C .>{..*RA CONTROL
20 33 3B 50 52 45 41 52 4D 3B 47 50 53 20 42 45  3;PREARM;GPS BE
45 50 20 53 41 54 45 4C 4C 49 54 45 20 43 4F 55 EP SATELLITE COU
4E 54 3B 56 54 58 20 50 49 54 20 4D 4F 44 45 62 NT;VTX PIT MODEb
crsf2msp e=0 n=0 s=11
C8 3E 7B EC C8 2B 3B 50 41 52 41 4C 59 5A 45 3B .>{..+;PARALYZE;
41 43 52 4F 20 54 52 41 49 4E 45 52 3B 56 54 58 ACRO TRAINER;VTX
20 43 4F 4E 54 52 4F 4C 20 44 49 53 41 42 4C 45  CONTROL DISABLE
3B 4C 41 55 4E 43 48 20 43 4F 4E 54 52 4F 4C A9 ;LAUNCH CONTROL.
crsf2msp e=0 n=0 s=12
C8 29 7B EC C8 2C 3B 53 54 49 43 4B 20 43 4F 4D .){..,;STICK COM
4D 41 4E 44 53 20 44 49 53 41 42 4C 45 3B 42 45 MANDS DISABLE;BE
45 50 45 52 20 4D 55 54 45 3B 0D                EPER MUTE;.
```

GPS enabled and DSHOT enabled (adds "Flip over after crash" to BOXNAMES) now 339 bytes and note that the reported MSP length in this has changed from 0xff to 0x00
```
crsf2msp e=0 n=1 s=9
C8 3E 7B EC C8 79 00 74 00 53 01 41 52 4D 3B 41 .>{..y.t.S.ARM;A
4E 47 4C 45 3B 48 4F 52 49 5A 4F 4E 3B 48 45 41 NGLE;HORIZON;HEA
44 46 52 45 45 3B 46 41 49 4C 53 41 46 45 3B 47 DFREE;FAILSAFE;G
50 53 20 52 45 53 43 55 45 3B 48 45 41 44 41 97 PS RESCUE;HEADA.
crsf2msp e=0 n=0 s=10
C8 3E 7B EC C8 6A 44 4A 3B 42 45 45 50 45 52 3B .>{..jDJ;BEEPER;
4F 53 44 20 44 49 53 41 42 4C 45 3B 54 45 4C 45 OSD DISABLE;TELE
4D 45 54 52 59 3B 42 4C 41 43 4B 42 4F 58 3B 46 METRY;BLACKBOX;F
50 56 20 41 4E 47 4C 45 20 4D 49 58 3B 42 4C 6E PV ANGLE MIX;BLn
crsf2msp e=0 n=0 s=11
C8 3E 7B EC C8 6B 41 43 4B 42 4F 58 20 45 52 41 .>{..kACKBOX ERA
53 45 20 28 3E 33 30 73 29 3B 43 41 4D 45 52 41 SE (>30s);CAMERA
20 43 4F 4E 54 52 4F 4C 20 31 3B 43 41 4D 45 52  CONTROL 1;CAMER
41 20 43 4F 4E 54 52 4F 4C 20 32 3B 43 41 4D 88 A CONTROL 2;CAM.
crsf2msp e=0 n=0 s=12
C8 3E 7B EC C8 6C 45 52 41 20 43 4F 4E 54 52 4F .>{..lERA CONTRO
4C 20 33 3B 46 4C 49 50 20 4F 56 45 52 20 41 46 L 3;FLIP OVER AF
54 45 52 20 43 52 41 53 48 3B 50 52 45 41 52 4D TER CRASH;PREARM
3B 47 50 53 20 42 45 45 50 20 53 41 54 45 4C A6 ;GPS BEEP SATEL.
crsf2msp e=0 n=0 s=13
C8 3E 7B EC C8 6D 4C 49 54 45 20 43 4F 55 4E 54 .>{..mLITE COUNT
3B 56 54 58 20 50 49 54 20 4D 4F 44 45 3B 50 41 ;VTX PIT MODE;PA
52 41 4C 59 5A 45 3B 41 43 52 4F 20 54 52 41 49 RALYZE;ACRO TRAI
4E 45 52 3B 56 54 58 20 43 4F 4E 54 52 4F 4C EB NER;VTX CONTROL.
crsf2msp e=0 n=0 s=14
C8 3E 7B EC C8 6E 20 44 49 53 41 42 4C 45 3B 4C .>{..n DISABLE;L
41 55 4E 43 48 20 43 4F 4E 54 52 4F 4C 3B 53 54 AUNCH CONTROL;ST
49 43 4B 20 43 4F 4D 4D 41 4E 44 53 20 44 4F 53 ICK COMMANDS DOS
41 42 4C 45 3B 42 45 45 50 45 52 20 4D 55 54 08 ABLE;BEEPER MUT.
```

We've got some stuff broken on our side as well that flips out when this goes wrong and ends up leaving our side broken after that, which we can fix, but this needs an upstream fix.

nightflasher, 2022-08-23:
Having jhemcuf7dual (target name) and no issues with betaflight over wifi, I would say it's uart and flight controller dependent.
On MatekF411 (target name) I can't get into DFU mode, when ExpressLRS connection is established...

CapnBry, 2022-08-23:
> Having jhemcuf7dual (target name) and no issues with betaflight over wifi, I would say it's uart and flight controller dependent.
> On MatekF411 (target name) I can't get into DFU mode, when ExpressLRS connection is established...

They why the F are you posting here? If you're saying you can't flash firmware through the ELRS wifi, that's not possible. It is an MSP configuration connection only.

nightflasher, 2022-08-24:
Wow! That's totally misunderstood!
So read carefully: when a ExpressLRS CONTROL LINK connection is established => you can't get the MatekF411 into DFU mode. So: turn of the transmitter and you can get the MatekF411 into DFU mode.

Don't ever use "why the f", as a developer in github! Be kind and ask back! 

Sad moment in ExpressLRS history 😢

CapnBry, 2022-08-24:
> Don't ever use "why the f", as a developer in github

Well don't jump on any random issue and bring up an entirely different topic then :)

All STM32 will not go into DFU mode if there is an external device pumping data into a UART (usually UART1, but varies by MCU). This includes a GPS or an active RC link connection from any system. Here's a bonus tip for you too: don't try to upgrade your firmware on a device that is currently in active use.

sercona, 2023-03-05:
there is a lesson in this.  when I update my elrs rx's, I take them offline and only give them power and gnd, then connect via wifi, upgrade and config and then wire back in-circuit.  same for fc, I remove most things from it, flash it, then connect things to it once I can usb-connect to it again.

its not obvious to everyone that this is a good way to do things, as you have to unsolder or remove some tx, rx lines (etc) but I find it just simplifies my life so I do it by rote.

closed 2024-07-30 by github-actions[bot]