ExpressLRS/ExpressLRS #1676: 'Not enough space' error when flashing receiver with new firmware

QUADMAN7, 2022-07-05:
i have been trying to flash a BetaFPV lite 2.4GHz receiver with the new expresslrs firmware but when trying to flash it the first time, it said that the target ('BETAFPV Lite 2400 RX') was wrong and that i should be using the 'DIY 2400 RX ESP8285 SX1280' target instead. after this error, i built new firmware using the configurator for this new target but upon flashing it, it came up with the error 'Not Enough Space'.
Does anyone know how to fix this???
Thanks for any help.
Also, i was just wondering if the page that comes up when you search 10.0.0.1 to flash your stuff works on windows 11 because my computer is running windows 11 and for some reason the page doesn't load.
thanks again for the help :)

commanderguy3001, 2022-07-05:
first of all: the easiest solution to the "not enough space" problem is to [flash the receiver using betaflight passthrough](https://www.expresslrs.org/2.0/quick-start/receivers/betafpv2400/#flashing-via-passthrough) instead of wifi.
second: the "wrong target" message is expected if you flash a receiver for the first time, as all the receivers use some other software from the factory as, at the point they were flashed, there was no official target for a specific receiver yet.
- > in that case you can just press the "flash anyway" button in the web ui.
third: yes, the 10.0.0.1 page also works with windows 11, but remember that you need to be connected to the wifi created by the receiver for that to work. If you specified your home wifi name + password, it will connect to that, and the way to access the web UI in that case is to go to https://elrs_rx.local, or check the IP from your routers interface.

closed 2022-07-06 by pkendall64

Zlyden3D, 2022-10-25:
I met the same issue. 
The reason was the firmware version 1.0 on the receiver, and I couldn't flash firmware 3.0.1
After several tries with the same result (Not Enough Space) I flashed RX to ERLS 2.0 version, after that successfully flashed ERLS 3.0.1 version.

andyp1per, 2022-11-03:
Even though my firmware claimed 2.0 I had first to update to 2.5.1 and then use the repartition hack and then was finally able to flash 3.0.1. Even then the uploader claimed failure but seems ok

QUADMAN7, 2024-05-16:
You are correct. Repartitioner firmware is only for transmitters. I found out the hard way and bricked a receiver trying to do that but managed to unbrick it with a USB-TTL adapter.

MortyFleem, 2024-05-16:
I seem to have spontaneously successfully updated my receiver now.
I was making moves to update with an much older version of the firmware when I changed the settings that became available on the 10.0.0.1 screen. I then clicked the 'reboot' button near the bottom of that screen and now I seem to have a steady green light on the RX and the TX screen is saying full signal. Happy days. Palava though.

LeonardoCCamara, 2024-11-09:
BINGOOOOO
Baixar arquivo(s) de firmware
O(s) arquivo(s) de firmware foram configurados para seu (EXEMPLO: RadioMaster RP1 2.4GHz RX) com as opções especificadas.
Para atualizar o arquivo de firmware para seu dispositivo, coloque-o no modo WiFi e conecte-se a ele pelo navegador, depois carregue o arquivo firmware.bin.gz na aba Update .

IMPORTANTE
O arquivo de firmware firmware.bin.gz deve ser instalado como está. NÃO descompacte ou descompacte o arquivo ou você receberá um erro.

livingperson, 2025-05-10:
Had "ERROR[4]: Not Enough Space" error when attempting update receiver from 3.3.2 to 3.5.4 via WiFi.  After seeing @MortyFleem's comment, I tried again, and it worked 🤷🏼‍♂️

ramiss, 2025-08-14:
ELRS upload function will not accept a .bin.gz file, it must be .bin

ramiss, 2025-08-14:
OK, stand down.  renaming the .bin.gz to .bin and then uploading seemed to work.  Closed.