ExpressLRS/ExpressLRS #1677: BETAFPV 2400 TX Micro 1000mW Telemetry issue on 50 Hz

ctausendfreund, 2022-07-05:
## Current Behavior

With the BETAFPV 2400 TX Micro 1000mW the telemetry on 50 Hz does not work reliable. After about 3 to 10 minutes I get multiple alternating "telemetry lost" and "telemetry recovered" messages. After a few seconds the telemetry stays lost. In the lua script, the C disappears when the telemetry is lost. The control link does work all the time, though. When I switch to any other packet rate, the telemetry works again. When I then switch back to 50 Hz the telemetry is lost again. When I restart my taranis I have telemetry again on 50 Hz for 3 to 10 minutes until the telemetry is lost again. I tried multiple hardware combinations (see below) all with the same issue. Only when I change my TX hardware, I don't have this problem. With a DIY 2.4GHz TX I have no telemetry issues on 50 Hz.

## Your Environment

- TX hardware: BETAFPV 2400 TX Micro 1000mW (I tried 3 different models)
- RX hardware:
  - EP1
  - EP2
  - BETAFPV ELRS 2,4 G Nano
  - Matek ELRS-R24-D
- Handset model:
  - FrSky Taranis Q X7 with Crossfire inverter mod on 400000 baud and 115200 baud
  - FrSky Taranis Q X7 without mod on 115200 baud
- OpenTX version (including nightly number)
  - 2.3.14
  - 2.3.15
- ExpressLRS version (TX & RX MUST MATCH): 
  - 2.3.0
  - 2.4.0
  - 2.5.0
- Packet Rate: 50 Hz
- Switch Mode: I tried both
- Telemetry Ratio: I tried all

pkendall64, 2022-07-05:
You have not said what power output you have the module set at. But, the problem could well be that at 1W power output the unit gets extremely hot and causes the xtal oscillator to get got as well, which causes frequency shift and you start to get lost packets.

ctausendfreund, 2022-07-05:
You are right, I forgot to mention it (added it now to my environment description): The power level does not seem to matter. Even on 25mW with a permanently enabled fan I have this issue. When I quickly power cycle the taranis I get another 3 to 10 minutes with working telemetry. So I'm not sure if it really can be a temperature issue. Btw. power cycling the RX does not recover the telemetry.

Is there a way to enable some debug logging on the serial port to get some more information about this issue?

AndreaCCIE, 2022-07-24:
I have the same issue as well, telemetry lost and sensor lost at 50Hz, all good at 150. Same TX module, Matek receiver

AndreaCCIE, 2022-08-08:
@ctausendfreund it looks like we are the only 2 using this mode on this hardware

ctausendfreund, 2022-09-04:
Yes, looks like that... I'm wondering if it has something to do with this issue: https://github.com/ExpressLRS/ExpressLRS/wiki/Crystal-Oscillator-(XO)-Frequency-Error

But it's strange that it only happens on 50 Hz. I'm not sure, but I think the higher frequencies would be more sensitive to this issue.

ozkancikikci, 2022-10-24:
I have the same issue as well, telemetry lost and recovered betafpv 1w nano receiver ver 3.0

closed 2022-12-27 by pkendall64