ExpressLRS/ExpressLRS #1686: Link loss and no re-acquisition on 900Mhz hardware

chris1seto, 2022-07-08:
## Current Behavior

While flying on two occasions, the link was lost at close range, and even once the aircraft crashed/landed and the TX was within 10ft, the link was not re-established. Power cycling the RX resulted in an immediate lock. 

Note that while the link wasn't "solidly" re-established, it seemed like one frame would come through every ~second. This is seen in the video below where the airplane violently rolls. This is because the FC was getting one frame which it would act on, until timing out immediately after. This was identical to the second instance.

## Steps to Reproduce

I haven't yet been able to make a replication plan for this. Out of ~250+ flights, I've seen this happen twice, each time with nearly identical behavior. 

Video of the first occurrence: https://www.youtube.com/watch?v=20TslQjyBbw&ab_channel=ChrisSeto

Note, at the start of this video, the RX had been freshly booted. In both occurrences, the link was lost less than ~3 minutes into the flight. The TX had been previously on for ~a half hour.

## Details

Both instances were at separate fields. At least one other user has reported this, his report below:

"
Marcopter — Today at 10:06 AM
@CapnBry I took a flight today with the RC1, R9 + R9 MINI, and an unpleasant thing happened to me.
[10:11 AM]
7" long range racer with iNav, set up to perform an appropriate RTL (gps + mag + baro).
After about 10 minutes of flying without any problems it went into failsafe, then performed the RTL correctly, landed and disarmed.
During the RTL obviously no telemetry on the tx, the link is completely dead, as if the rx were turned off.
I tried to turn off and on the tx during RTL, the module turned on correctly with the startup sound but always dead link with the rx.
Once landed  the rx it had the red and green LEDs both on, as when it works correctly linked but it no longer communicated with the tx and also with the flight control.
I removed the main power and put it back on and everything started working again, so i guess the firmware on the rx has stopped working during flight.
868 MHz, telem ratio 1:2, 12 ch mode.
What do you think about it? The problem is definitely in the rx firmware and not in the module side.
In the meantime I'm going back to 2.5.0, where this problem has never occurred with my tx/rx, but I remain available for other tests, as RTL works well.
PS: at the bench, obviously not armed, I left it on for two hours powered by USB, always stable link. 
"

## Your Environment

- TX hardware: ES900TX 2.5.1
- RX hardware: BetaFPV 915 (2.x, not sure exactly the version)
- Handset model: Horus x10s, EdgeTX Black Pearl
- Packet Rate: 150hz
- Telemetry Ratio: default
- user_defines: FCC_915, LOCK_ON_FIRST_CONNECTION,BINDING_PHRASE, HOME_WIFI_SSID, HOME_WIFI_PASSWORD

robustini, 2022-07-08:
I'm Marcopter.
My enviroment:

- Firmware: V3.0.0 RC1
- TX hardware: FrSky R9M
- RX hardware: FrSky R9 Mini 
- Handset model: Horus X10s Espress with the latest official OpenTX firmware
- Full res 12 ch mode
- Packet Rate: 100hz
- Telemetry Ratio: 1:2
- TX power: dynamic (max 250 mW)
- Frequency: 868 MHz EU

commanderguy3001, 2022-07-08:
thinkimg about it, have we thought about the possibility of this being a problem with betaflight? (just a possibility I wanted to get out there cause it might be worth a thought, not saying it has to be betaflight or anything)

CapnBry, 2022-07-08:
I can't reproduce this no matter what I try, another few hours with around -85dBm to -95dBm from across the house with no antenna
* 3.x.x-maint
* ES900TX + R9MM RX
* 100Hz Fullres 1:2 dynamic 25mw
* EU868
* BF 4.3

I've also done several hours of testing on 200Hz Std TLM ratio and FCC915 on RC1 last week. I don't really have anything to go on so I don't have any plans for further testing. I've also tried power cycling the RX at 30 minutes and the TX at 60 minutes and it just picked right back up both times. I'm afraid there's not much I can do without any sort of repeatable test case.

robustini, 2022-07-09:
I already had another block of the rx on the bench, almost immediate, after 20/30 seconds that I power the FC with the USB plug.
I still had to turn on the TX module, so the rx had the green led blinking (no link, ELRS V3 RC1), and now it's off, check if I can put the rx in the bootloader in this state, to check if he is actually no longer communicating with the FC. 
And this is confirmation that the rx in that state is dead, exactly what happened to me in flight yesterday, or the firmware crashing or perhaps a hardware problem.
It is interesting to have noticed that the problem occurred even without being linked to the module and with the system disarmed, imho this eliminates many possible things from the investigation.

[image]

Removed and put back the USB and now it is there that blink in green.
Tried a reflash and yes, now responds.

[image]

Restarted for 60 minutes, the rx still has the green LED that blinks, the tx module is always off.
So the thing is too random, it is not easy to replicate but it came first immediately.

robustini, 2022-07-09:
Another rx lock now, so i can replicate the problem.

commanderguy3001, 2022-07-09:
seems like you found a way to reproduce it more often, but it still is random which is kinda meh, but definitely a better starting point for searching for the error now

robustini, 2022-07-11:
As for me problem solved, faulty receiver.

robustini, 2022-07-18:
> I will try the 3.0 RC1 today and do a few low level close range tests. To see if it repeats it self. Out of frustration I also ordered a BetaFPV micro TX to check if it is the R9 TX fault.

I also bought BetaFPV Micro 2.4 GHz 1W, with R9M I have several random problems with ELRS, on 2.4 modules it seems more stable.

corben9, 2022-07-18:
> > I will try the 3.0 RC1 today and do a few low level close range tests. To see if it repeats it self. Out of frustration I also ordered a BetaFPV micro TX to check if it is the R9 TX fault.
> 
> I also bought BetaFPV Micro 2.4 GHz 1W, with R9M I have several random problems with ELRS, on 2.4 modules it seems more stable. And I am heartbroken because I also have five R9 Mini receivers here, I will reinstall the FrSky firmware.

I did a considerable bit of more testing over the weekend. Might have found the issues. I ran through 5 pack at 12 minute flights on ranges up to 1.5 km in a pretty noisy environment. Basically until my DJI FPV system cut out and did a bit of more analysis on the telemetry. 

It seems that the fault laid with my TBS Diamond Antenna I payed 30 bucks for :(. I switched it to the stock 868 antenna that comes with the BetaFPV Micro TX 900. Ran test at 100 mw 250mw and one flight at 500. RQly only dipped once or twice to the lowest rating of 93. That happend when I approached 100 db. Regularly it stayed at around 60-80 range. The drop occurred when I turned back towards me and the antanna that I have mounted at horizontal (I know its not the best position but should be ok for anything up to 4 km). I also did a test and flew stright up for around 300 m and lost DJI FPV singal at 1.1 km. Control link was stable at that point hovering at around 80 db. I switched the quad to GPS rescue and it returned normaly while the DJI system rebootet so i cound land saftely.

As for 3.0 RC1. Still have not been able to flash the TX. For some reason when I compile the firmware first there is a name mismatch and then when I maualy try to uplaod via USB i get a size error that the FW is to big.

Ill keep u posted on more testing.

robustini, 2022-07-20:
@corben9 it goes in support and clear dependencies and firmware, I tried here to compile for your rx, no problem.
The antenna is essential, and in your case from what you write sure it could be that.

[image]

corben9, 2022-07-20:
> @corben9 it goes in support and clear dependencies and firmware, I tried here to compile for your rx, no problem. The antenna is essential, and in your case from what you write sure it could be that.
> 
> [image]

Hej :) Thxn for the try. I have no problems with the compiling of the firmware's. I cleared the dependencies etc.. Problem is the Firmware update on the BetaFPV Micro TX 900. I tried flashing over USB and Wifi but first get the error that its the wrong firmware target and then that the firmware size is to big to be flashed. 

All other versions for the BetaFPV Micro TX 900 compile well. I am back at 2.5.1 now.

robustini, 2022-07-22:
Just installed the V3 on BetaFPV Micro TX 900, no problem via USB.
Too bad it has a fan that vibrates like a washing machine.
These hardware are cheap but too spartan, CRSF is much better even if it obviously costs much much much more.

robustini, 2022-07-26:
Look at this!

https://youtu.be/Cp5V1GTqNBI

nightflasher, 2022-07-31:
Yesterday I've tried to fly my 8" LR, which is equipped with the 20x20 DIY 868MHZ receiver with the tbs crossfire antenna (the big one) and my transmitter (jumper t-lite) has the ttgo_v2 with the R9M super 8 antenna, but I couldn't get a stable connection within 2m distance.
Settings I've tested:
1) 200 Hz / Std (128:1) / 50mW dyn
2) 100 Hz / Std (128:1) / 50mW fest

Thanks to this system, I was able to flash back to 2.5.1 which worked reasonably well; except for the many RX LOSS messages in betaflight OSD, which I have to measure the distance and have a look at the geographic environment.

commanderguy3001, 2022-07-31:
> Yesterday I've tried to fly my 8" LR, which is equipped with the 20x20 DIY 868MHZ receiver with the tbs crossfire antenna (the big one) and my transmitter (jumper t-lite) has the ttgo_v2 with the [...]

the R9M super 8 antenna is notorious for breaking internally without you noticing, I'd recommend you swap that out to something else and try again.

nightflasher, 2022-08-03:
What I've done so far:
- measured the super8 tx antenna: it's fine
- measured the tbs rx antenna: it's fine 
- changed the tx antenna: same problem
- changed the rx antenna: same problem
- changed the tx to ttgo_v1: same problem 
- changed the SX1276 in the rx, even the rx is build a few weeks ago
- changed firmware back to 2.5.1 - working fine

So, I've done everything I could imagine.

c3r63ru5, 2022-08-04:
Just for the record, I experienced the same (or similar) behaviour after trying the 3.x.x maintenance build. RXLOSS, quad failsafed and returned home, no link for about 10 seconds until crash. Link was only re-acquired after power cycle.

TX hardware: R9 Lite Pro on 3.x.x maintenance
RX hardware: R9 MM on 3.x.x maintenance
Handset model: FrSky T-Lite Pro, EdgeTX 2.7.1
Packet Rate: 200Hz
Telemetry Ratio: default
Transmit Power: 100mW
Frequency: 868 MHz EU

I went back to 2.5.1, no RXLOSS since then. I'm afraid to flash 3.x.x again, as most of my quads will try to return to home when failsafing and this can be a bit dangerous when the link is not established when the quad is coming in. ;)

robustini, 2022-08-04:
> Just for the record, I experienced the same (or similar) behaviour after trying the 3.x.x maintenance build. RXLOSS, quad failsafed and returned home, no link for about 10 seconds until crash. Link [...]

Precisely for that reason I am doing tests with 3.x.x maintenance and various PR on a quad with Ardupilot, the return to home works perfectly, land, disarm, not like iNav and Betaflight, so let the rx freeze! :P

robustini, 2022-08-04:
Here is the crash of the rx! After one hour of bench feeding.
Switched off and on again the module and no link, switched off and on the FC and immediate link.
R9 Mini + R9M, 3.x.x maintenance.
The problem is, it is not easy to get it out, and maybe if it happens in flight for many it will not be nice.

[image]

robustini, 2022-08-04:
About 3 hours without ever a link loss.
But there is a difference: the rx was flashed with the debug activated and connected in passthrough to the FC, so the telemetry part was not used.
Do this test: check if leaving the rx on without talking to the FC if it loses the link with the module.

robustini, 2022-08-09:
Today after 5 minutes in the first flight rx locked, lost link on the tx, the racer successfully performed the RTH, landed and disarmed.
Having the FC that also feeds in USB I connected it and I removed the Li-Ion in order to keep the rx in that state to do some analysis.
Activated the passthrough, launched the monitor inside the configurator on the serial port of the rx, no communication.
So the problem is there, sure on R9 Mini with latest firmware installed from branch 3.x.
Bad story this, I await our feedback.

robustini, 2022-08-10:
Today after 5 minutes in the first flight rx locked, lost link on the tx, the racer successfully performed the RTH, landed and disarmed.
Having the FC that also feeds in USB I connected it and I removed the Li-Ion in order to keep the rx in that state to do some analysis.
Rx with red and green LEDs on, activated the passthrough, launched the monitor inside the configurator on the serial port of the rx, no communication.
Removed and restored power and the rx is back alive.
So the problem is there, sure on R9 Mini with V3 RC2 firmware installed, but @chris1seto had the same problem with BetaFPV 900 rx, so I don't think it's a problem of STM32 or ESP8285 which use different MCU, at this point by exclusion I believe it is something wrong in the code which causes this lockup, my two cents.
Bad story this, I await our feedback.

robustini, 2022-08-10:
> thinking about it, have we thought about the possibility of this being a problem with betaflight/inav/ardu/whatever you are using on your flight computer? (just a possibility I wanted to get out there cause it might be worth a thought, not saying it has to be betaflight or anything)

I don't think there can be any correlation to the type of flight control and software.
When this happens the rx no longer sends any packet to the FC, ascertained with the serial monitor, and obviously does not send it to the module either, so the FC and TX Module continue to run.
If we were talking about Windows imho it would be equivalent to a BSoD.
To ascertain this it would be enough to have an activity of the LEDs during the link instead of making them both stay on all the time.
And just because the serial monitor does not return anything strange during this lockup, it simply stops the serial flow, I think it will be quite complex to understand how it turns out, even with debug enabled.
But we have an important fact: on ELRS 2.4 GHz systems I have not seen anyone complain about problems like these.

robustini, 2022-08-14:
Other bench tests yesterday and R9 MINI locked after about 90 minutes at the test bench.
Another test without the module connected and the rx never stopped working for over 9 hours, when the module was switched on it was immediately connected.
So apparently it crashes when working in transceiver, a MCU that definitely heats up.
Same tests with BetaFPV 900 MHz rx, no lock either with connected or disconnected module.

nightflasher, 2022-08-14:
I have now switched back to my nirvana and set up a telemetry screen:
Transmit power is locked to 10mW and dynamic switching is off, but the module and the screen showes permanent switching through the power levels - even the connection is established. Receivers are r9mx/r9mm/diy 868MHZ

robustini, 2022-08-14:
> I have now switched back to my nirvana and set up a telemetry screen: Transmit power is locked to 10mW and dynamic switching is off, but the module and the screen showes permanent switching through the power levels - even the connection is established. Receivers are r9mx/r9mm/diy 868MHZ

No, I absolutely don't have this problem with R9M and R9 Mini, if septum 10 mW remains constant.

c3r63ru5, 2022-08-15:
> I have now switched back to my nirvana and set up a telemetry screen:
> Transmit power is locked to 10mW and dynamic switching is off, but the module and the screen showes permanent switching through the power levels - even the connection is established. Receivers are r9mx/r9mm/diy 868MHZ

I would need to flash 3.x.x again to check, I remember the TX storing the power setting and in my tests I was using 100mW fixed but I never checked the telemetry to verify.

robustini, 2022-08-17:
Now I have confirmation that the rx crashes from the check I made on the temperature.
In normal operation it warms up enough, when it freezes it goes to room temperature.

nightflasher, 2022-08-23:
> >
> > I would need to flash 3.x.x again to check, I remember the TX storing the power setting and in my tests I was using 100mW fixed but I never checked the telemetry to verify.

Did a few times, but still switching around, will take a video later.

robustini, 2022-08-25:
> > I would need to flash 3.x.x again to check, I remember the TX storing the power setting and in my tests I was using 100mW fixed but I never checked the telemetry to verify.
> 
> Did a few times, but still switching around, will take a video later.
> 
> Edit: https://vimeo.com/742406527

Seen the video, very strange, he absolutely does not do it with the RC2, stable at 10 mW.

robustini, 2022-08-28:
I continue to replicate the problem with two different R9M modules (2018 version modified with the resistor and 2019), two different TX and many R9 Mini, but not with other 900 Mhz receivers.
I hope the developers find the problem because it is really debilitating, or just remove the R9 Mini receivers from those supported by ELRS.
If there is no stability it is better not to use them.

robustini, 2022-08-28:
I continue to replicate the problem with two different R9M modules (2018 version modified with the resistor and 2019), two different TX and many R9 Mini, but not with other 900 Mhz receivers.
Time is random, sometimes it stops working after ten minutes, sometimes it stops working after two hours, there is no rule.
From what people say nobody has ever had these problems with ELRS versions equal to or lower than 2.5.1, personally I have always tried with the V3 dev.
I hope the developers find the problem because it is really debilitating, or just remove the R9 Mini receivers from those supported by ELRS V3.x.
If there is no stability it is better not to use them.
Tested R9M 2019 module with FrSky Access firmware (also on the R9 Mini), left on for a whole day, always connected, so it's not a hardware problem.

Seba112PL, 2022-09-01:
Thats why I started selling my R9mm/mini's and im currently buying other esp based receivers. No problems with R9M as transmitter. Only those R9mm receivers

ewileycoy, 2022-09-05:
I can replicate the OP's issue with two different TX and at least 2 different RX on 915mhz: after a few minutes, the RX appears to lock up and lose connection to the transmitter. Model failsafes with RXLOSS and does not re-establish connection to the TX until it's power-cycled. 
It's not 100% consistent but seems to happen more often than not lately which has started turning me into #team2.4 which is stable as rocks. 
I ran through the discord and found the OP also abandoned 900 for 2.4 which is sad to me :(

Environment:
Radio Taranis Q x7s, EdgeTX 2.5.0
TX R9M (2019) and Betafpv 900 mini
RX R9mini and Betafpv 915
ExpressLRS 2.5.1 on both TX and RX
REGULATORY_DOMAIN_FCC_915 AUTO_WIFI_ON_INTERVAL
This has happened on a flight controller as well as a Matek CSRF to PWM converter, so it doesn't seem like it's specific to a FC firmware.

I have telemetry setup on my radio, but unfortunately no blackbox on the model. Attached telemetry is from a flight using the Betafpv RX (with a 'full size' 915 antenna) and TX, so it's not anything frsky (except my Radio, I guess). Seems to completely lock up around 2:30 in. Distance was a clear 100 meters or so.

On a possibly-related note, I bought an Immersion RF tester to see if the TX hardware is just bad: Directly connected to the TX I'm seeing 12dbi at 100mW on the Betafpv 900 TX, on the r9m I see 39db at 100mW, and 37dbi at 100mW on a 2.4 (happymodel ex24pro). So I think my Betafpv 900 TX mini might be bad too?

[FIXED GYRO-2022-09-02-175348.csv](https://github.com/ExpressLRS/ExpressLRS/files/9490443/FIXED.GYRO-2022-09-02-175348.csv)

robustini, 2022-09-06:
@ewileycoy if you want to use the R9 Mini without ever having a problem use the original FrSky ACCESS firmware which is solid as a rock and the range is good, you will have 16 channels at full res and never a lock.
With ELRS that receiver has that problem, systematically but randomly.

robustini, 2022-09-12:
Just out of curiosity I'm testing an R9 MM and it doesn't give any problems, no firmware lockup.
Now I would like to try to understand if the MCU is different in the R9 Mini.

nightflasher, 2022-09-12:
My latest wild guess: it's the lora chip, not the MCU.

robustini, 2022-09-12:
> My latest wild guess: it's the lora chip, not the MCU.

Yes, it is possible that they are different in quality.

robustini, 2022-09-12:
In the first image the LoRa of the R9 Mini, quite anonymous, with that large number that doesn't seem to me to mean anything, it will be a low quality super clone rf chip.
In the second the LoRa of the R9 MM, correctly labeled.
The MCU is of course the same.

[image]

[image]

EricFPV, 2022-10-02:
I think it’s the R9mm tx chip’s problem, not compatible with new firmware code.telemetry power setting too high maybe .overheat the Mcu/ Lora chip ? I got few bigger R9 slim working great with v3 rc2. But R9mm after 2-3 fligth lost connection link. Power cycle and link come back.  Flash back to 2.5.1 work flawlessly. 

Firmware: V3.0.0 RC2
TX hardware: FrSky R9M lite pro
RX hardware: FrSky R9 Mini
Handset model: taranis x lite pro Edgetx 2.7
Packet Rate: 200hz
Telemetry Ratio: 1:64
TX power: 100mw
Frequency:915 MHz

robustini, 2022-10-02:
> I think it’s the R9mm tx chip’s problem, not compatible with new firmware code.telemetry power setting too high maybe .overheat the Mcu/ Lora chip ? I got few bigger R9 slim working great with v3 [...]

Thanks for your feedback.
Unfortunately it is a known problem, which developers apparently cannot solve via software.
Probably the V3, indeed it definitely leads more to MCU or RF chip exasperation and something goes wrong sooner or later.

robustini, 2022-10-02:
> I think it’s the R9mm tx chip’s problem, not compatible with new firmware code.telemetry power setting too high maybe .overheat the Mcu/ Lora chip ? I got few bigger R9 slim working great with v3 [...]

Thanks for your feedback.
Unfortunately it is a known problem which developers cannot find, a bug somewhere leading to this condition is now sure to exist, when an event is repeated to several users it is no longer a single case.
The hardware has no problems or with the original FrSky firmware would come out, but instead no problem with that.
Probably the ELRS V3, indeed it definitely leads more to MCU or RF chip exasperation and something goes wrong sooner or later.
At least the R9 MINIs supported by the V3 should be removed from the wiki.

alteman, 2022-10-12:
[image]

Everyone affected by this issue and on STM32 based receiver hardware (FrSky R9mm/R9mini/R9mx) are welcome to try flashing the firmware from PR #1874 using ELRS Configurator (see screenshot) and see if it resolves the issue.  
Note that the defining characteristic of this bug is that both red and green LEDs are continuously lit up and rxloss doesn't resolve itself until receiver power is cycled (a lock-up).

RealTadango, 2022-10-17:
I am new to ELRS but have R9M with R9 Mini also, and a few connection losses in 1 hour on the bench. I will try the suggested firmware and test again a few hours.

Seba112PL, 2022-10-18:
PR https://github.com/ExpressLRS/ExpressLRS/pull/1874 Fixed it for me. No more lost connection with no reconnecting until replug of the drone. Im testing it for 24h. no issue. I will keep on testing for another 24h tho just to be sure

robustini, 2022-10-19:
I have carried out further tests and I confirm the total resolution of the problem, finally!
This issue can be closed because the PR has already been inserted in the code.

Seba112PL, 2022-10-19:
It is in 3.0.0 main release already ? or still in ExpressLRS:3.x.x-maintenance

RealTadango, 2022-10-19:
> It is in 3.0.0 main release already ? or still in ExpressLRS:3.x.x-maintenance

It is merged in the 3.x.x-maintenance branch. I think the 3.0.0 main release should be removed to prevent issues.

Seba112PL, 2022-10-19:
So we will get 3.0.1 or 3.1.0 soon. There is no point for removing 3.0.0. Problem only affects stm32 based receivers. It should be noted in configurator somewhere when selecting 3.0.0 and r9 hardware that it might case issues

robustini, 2022-10-19:
Yes, something will have to be done for 3.0.0 as it is affected by this problem.
This is why I asked for 3.0.1 to be released and the other removed as soon as possible.

RealTadango, 2022-10-19:
Not fixed sadly. My plane is currently loitering above without any control :( RC link suddenly stopped....

Seba112PL, 2022-10-19:
> Not fixed sadly. My plane is currently loitering above without any control :( RC link suddenly stopped....

What version is on your gear ?
And when did you flashed ELRS to that version ?

RealTadango, 2022-10-19:
I flashed Sunday with the PR mentioned. I tested for 3 hours straight without a hickup. Looked fine. Now the same effect, complete lockout.

tchleb, 2022-10-19:
I had the same issue 14 days ago. I was flying in a lost quarry, after 2-3 Minutes flying i got a failsafe ( the Copter was ~10m away from me at ~15m hight).
Luckily nothing broke and I was able to salvage the copter. After the Failsafe both LEDs (red & green) on the R9mm was permanently on. Only with a power cycle on the copter i was able to reconnect. 

- RX: R9MM with 3.0.0
- TX: R9M lite pro 3.0.0 (250mw dynamic power)
- Betaflight 4.3.0 on a F405 FC

I flashed today the 3.x.x-maintenanc branch on the R9MM. 
Do I also have to update the FW of the R9M lite pro? The module also uses an STM32

robustini, 2022-10-19:
> I had the same issue 14 days ago. I was flying in a lost quarry, after 2-3 Minutes flying i got a failsafe ( the Copter was ~10m away from me at ~15m hight). Luckily nothing broke and I was able to [...]

Only the rx.

tchleb, 2022-10-19:
> Only the rx.

Thank you I'll try it if this solve the problem.

I have also a R9M slim plus in use. I guess this RX is affected too (it has an stm32).

RealTadango, 2022-10-19:
> > I flashed Sunday with the PR mentioned. I tested for 3 hours straight without a hickup. Looked fine. Now the same effect, complete lockout.
> 
> Which receiver? I have tested it for whole days on a R9 Mini and never a problem.

I have a R9 Slim (not +) with latest bootloader flashed from OpenTX

alteman, 2022-10-20:
> Btw, is there anyway to get any version / build information from the RX?
Yeah, in the bind section of LUA script.
 
[image]

alteman, 2022-10-20:
ExpressLRS LUA, Other devices -> receiver name -> *version and git ID*
[image]

[image]

[image]

@RealTadango Can you post your git revision on the receiver that is hanging?

RealTadango, 2022-10-20:
Mine says HEAD 72e2dc. Firmware file that was generated is called Frsky_RX_R9SLIM-PR_1874.bin

context: {
    userDefineFilters: e {
      target: 'Frsky_RX_R9SLIM_via_BetaflightPassthrough',
      source: 'GitPullRequest',
      gitTag: '3.0.0',
      gitBranch: '',
      gitCommit: '',
      localPath: '',
      gitPullRequest: e {
        title: 'Move uart writes from ISR on STM32',
        id: 1082650166,
        number: 1874,
        headCommitHash: '72e2dcb2d2c5c3da577c51b48480ae42ffb4f8e6'
      }
    }
  },

[49 more replies]

closed 2022-11-21 by pkendall64