ExpressLRS/ExpressLRS #1691: ELRS 3.0 RC1 - AUX switch for dynamic power makes lose the link (repeatable here)

robustini, 2022-07-14:
## Current Behavior

I report this problem with the 3.0 RC1: I have assigned the enabling of dynamic power to a switch (R9 + R9 MINI).
After a few switches, the link is lost and then hung up.
If obviously it happens in flight it is not nice.
I have a video documenting it, and here I can repeat the problem, it constantly arises.

https://youtu.be/Vl-ygKBbi-k

In flight I have never used such switching, but I believe it could happen and it would not be pleasant.

## Steps to Reproduce

1. Enable dynamic power with a switch

2. Activates and deactivates dynamic power, here it appears after no more than 10 commutations, constantly

## Possible Solution (Not obligatory)

## Details

If the bug is not found the only solution is to not use this switch

Hardware and settings used:

## Your Environment

- TX hardware: FrSky R9M 2019
- RX hardware: FrSky R9 MINI
- Handset model: FrSky Horus X10S Express
- OpenTX version: 2.3.15
- ExpressLRS version (TX & RX MUST MATCH): V3 RC1
- Packet Rate: 100 Hz
- Telemetry Ratio: 1:2
- user_defines: 868 Mhz

CapnBry, 2022-07-14:
I can't reproduce this but I don't have the same hardware. However, it doesn't seem to me that the hardware makes any difference. Tested 100 switches (50x on 50x off) for each:

* ES900TX + R9MM RX - With betaflight 4.3. Not quite the same because the ES900TX uses an ESP32 vs the STM32 in the R9M so I tried...
* FM30 TX + EP2 - With betaflight 4.3. Uses the same STM32 the R9M uses, but is Team2.4
* Jumper T-Lite + EP2 - No FC. ESP32 and Team2.4

There's very little going on behind the scenes when you flip the switch, it isn't actually changing any config, it just adjusts locks or unlocks the power. I don't believe there is anything wrong with the code but seems to be your hardware again?

robustini, 2022-07-14:
@CapnBry tested now the maintenance version and the problem is the same.
When the link is restored the module emits the classic two tones of when it hooks the link with the rx, not the music of the reboot.
Tomorrow I try with another R9M and another rx, always R9 MINI.

CapnBry, 2022-07-14:
Ah gotcha. It has been a long time since I've heard the R9M boops and beeps. It definitely looks like it is setting the power though based on the RSSI changing, which it should not on 3.0.0-RC1 (only on 3.x.x-maintenance). If there is a problem it's gotta be just with R9M because I can't reproduce it on any other setup.

robustini, 2022-07-15:
@CapnBry at the moment I have only tried with another rx connected to a Pixhawk with Ardupilot, so we exclude the type of FC and the software, it does exactly the same problem, later I try with another R9M and I tell you what happens.

robustini, 2022-07-15:
@CapnBry i'm sorry to give you this feedback but it does exactly the same thing with another R9M (this was originally an ACCESS while the other one was 2019) and another receiver R9 MINI, so different hardware and same firmware.
I reproduce the issue when I want, no more than 10 commutations and it does.

robustini, 2022-07-15:
More video with that ACCESS module, look at 0:40, I did the problem three times.

https://youtu.be/mMUL0lO51UM?t=42

robustini, 2022-07-16:
@CapnBry I have identified the problem, or at least I was able to understand in what condition just by changing one thing it does not generate any issue.
The problem is related to the telemetry ratio.
With 1:4 (1881bps) i did hundreds of commutations, never lost the link.
With 1:2 (3888bps) the problem presents itself immediately as I have shown you.

robustini, 2022-07-17:
@CapnBry another feedback which I hope can help you find the problem if there is.
She left the rx at home and I went out with the tx to a point where the dymanic power often worked in switching a lot, never lost the link.
As you can see from the video there are a lot of automatic switches and the link remains stable.
So the problem occurs only when switching from dynamics to fixed with AUX channel.

https://youtu.be/Bg_G1QPmlHo

CapnBry, 2022-07-17:
Well that's good information, but it still doesn't make any sense, since the same code is making those changes too. I'm not really following this issue any more because it is not reproducible on any hardware so it would be a waste for me to spend more time on it.

lorenzing, 2022-07-17:
Hi all, I've checked and it also makes me exactly the same problem with R9M and R9 MINI with ELRS V3 RC1.

robustini, 2022-07-17:
> Hi all, I've checked and it also makes me exactly the same problem with R9M and R9 MINI with ELRS V3 RC1.

Thanks @lorenzing for your feedback, I am comforted to know that I'm not the only loser! ;-)

Seba112PL, 2022-07-17:
Tested R9M with R9MM and GepRC Nano 900mhz. 
GepRC works fine with R9M on 3.0.0 but r9mm loses connection after some random time. I belive that it has nothing to do with dynamic power. I've a video of that happening and will upload it tomorrow.. Sometimes happens after 2 min sometimes after 30. once happend after over an hour of bench testing.

brandonrc, 2022-07-17:
@CapnBry want me to send  you a R9M again 🤣 

Maybe this is related to the r9 mini? 

Sounds like @Seba112PL did not have an issue using gepRC but had some issues with r9mm.

robustini, 2022-07-18:
> Tested R9M with R9MM and GepRC Nano 900mhz. GepRC works fine with R9M on 3.0.0 (after 2h of bench testing and some time in air flying) but r9mm loses connection after some random time. I belive that it has nothing to do with dynamic power. I've a video of that happening and will upload it tomorrow.. Sometimes happens after 2 min sometimes after 30. once happend after over an hour of bench testing.

Thanks for your feedback, much appreciated.
But have you tried to replicate what I show in the video, with my same ELRS settings?

Packet Rate: 100 Hz Full
Switch Mode: 12ch Mixed
Telemetry Ratio: 1:2 (3888 bps)
Max Dynamic power: 250 mW (irrilevant, with 100 the issue is the same)

robustini, 2022-07-18:
> Tested R9M with R9MM and GepRC Nano 900mhz. GepRC works fine with R9M on 3.0.0 (after 2h of bench testing and some time in air flying) but r9mm loses connection after some random time. I belive that [...]

Thanks for your feedback, much appreciated.
But have you tried to replicate what I show in the video, with my same ELRS settings?
I did the test with five R9 MINI and two R9M, same problem, if it is replicable on more identical hardware something is there.
Dynamic power has no problems with internal switching, but when using AUX to switch between dynamic to fixed.
In my bench tests I never lost the link with the V3 RC1, only some reports of "lost sensor" from the tx but stable link for hours.
As I wrote above with 1:4 telemetry the problem does not arise, so there is something that bothers him when he has too much stuff to handle at 100Hz Full, in fact at 100Hz (Wide) I did a myriad of switching and the link is always stable.

Packet Rate: 100Hz Full
Switch Mode: 12ch Mixed
Telemetry Ratio: 1:2 (3888 bps)
Max Dynamic power: 250 mW (irrilevant, with 100 the issue is the same)

robustini, 2022-07-18:
@Seba112PL how much do you have the "Max Bauds" in your tx in SYS?
Setting it to 400Kbps I lose the link after a few seconds (why? boh!), at 1.87Kps it does the same thing, the only one that works decently for me is 921Kbps.

Seba112PL, 2022-07-18:
> @Seba112PL how much do you have the "Max Bauds" in your tx inside SYS menu? Setting it to 400Kbps I lose the link after a few seconds (why? boh!), at 1.87Kps it does the same thing, the only one that works decently with my FrSky X10S Express and Radiomaster TX16S is 921Kbps on both. Over 1.87Kbps, which is however unstable here, the module does not communicate with the tx.

I have a R9M (pre 2019) with resistor mod. So 400k is max for me and thats what I use.

robustini, 2022-07-18:
> > @Seba112PL how much do you have the "Max Bauds" in your tx inside SYS menu? Setting it to 400Kbps I lose the link after a few seconds (why? boh!), at 1.87Kps it does the same thing, the only one that works decently with my FrSky X10S Express and Radiomaster TX16S is 921Kbps on both. Over 1.87Kbps, which is however unstable here, the module does not communicate with the tx.
> 
> I have a R9M (pre 2019) with resistor mod. So 400k is max for me and thats what I use.
> 
> ps. Testing r9mm with 2.5.1 since yesterday. No issues. For some reason 3.0.0 on r9mm just hangs up after some time (maybe it affects stm32 based receivers ? since my geprc nano 900 is esp based and had no issues on 3.0.0)

It is possible, but in the issue that I have exposed I believe that it was the module that crashed, if you read carefully what I wrote at the beginning.
Now you made me want to try another rx at 900 MHz, not FrSky.

Seba112PL, 2022-07-18:
[https://youtu.be/B95T_T3RTjE](url)

Thats the problem i'm facing. RXLoss happens around 0.30min. RXLoss is sudden and without any warning Forced me to look for my quad 2km away in neck high vegetation. I am lucky that I have found my drone 😆

robustini, 2022-07-18:
> [https://youtu.be/B95T_T3RTjE](url)
> 
> Thats the problem i'm facing. RXLoss happens around 0:30 (about 8-9 min realtime). RXLoss is sudden and without any warning Forced me to look for my quad 2km away in neck high vegetation. I am lucky that I have found my drone 😆
> 
> I will try this evening to recreate your issue with mine r9mm and geprc receiver

You had to turn the rx off and on again to re-establish the link, which I did too but it remained disconnected.
I turned the module off and on again and the link was re-established.
Well, I don't understand anything anymore.

robustini, 2022-07-18:
> > @CapnBry want me to send you a R9M again 🤣
> > Maybe this is related to the r9 mini?
> 
> I don't ever want to see that thing again! 😂 The R9MM and the R9MINI are exactly the same apart from an external inverter that we don't use and I tested with the R9MM and saw no issue. I believe it has to be on the R9M TX side.
> 
> The code the boost switch uses and the code that is used if the LQ drops or disconnects is the same function so I can't see how the switch would do anything. The code is just too simple for there to be a problem with the switch. If there's a problem I'm guessing it is in controlling the power output DAC, but that is pure conjecture on my part because I can't reproduce it.

If it can help you I'm willing to debug the module, just know how I should proceed.
If the ST-Link is enough, I have that too.
Even just to understand out of curiosity what happens at that moment, since it is not accidental, it is almost systematic.

robustini, 2022-07-18:
> Did yooooouuuuu think I was lying about it? 😂

But absolutely not! :P
That's why I can't explain why he does that thing as much as you do, but you can't say he doesn't do it eh! ;-)

SunjunKim, 2022-07-18:
> > Could you test what the power level which starts the problem is?
> 
> I wrote it above, the power level is not the problem. Even if I put 25 mW as max power in switching from 10 (minimum) to 25 it gives the same problem, it loses the link.

If you change the power manually on the Lua does it happen also?

robustini, 2022-07-18:
Absolutely not, as it doesn't lose the link even during the dynamics power in action, I posted a video above that shows it.
It only happens from dynamic to fixed with the switch, damn.

CapnBry, 2022-07-18:
And just to verify, you're on 3.x.x-maintenance branch, not 3.0.0-RC1? All testing must be done on 3.x.x-maint since 3.0.0-RC1 has known issues on that hardware.

robustini, 2022-07-18:
> And just to verify, you're on 3.x.x-maintenance branch, not 3.0.0-RC1 (look on the bottom of the Lua)? All testing must be done on 3.x.x-maint since 3.0.0-RC1 has known issues on that hardware.

Yes Brian, i use LOCAL in the configurator, and the src is the latest in that branch.

Seba112PL, 2022-07-19:
Tried brand new r9mm. Same problem. Disconnects after a while.

robustini, 2022-07-19:
> Tried brand new r9mm. Same problem. Disconnects after a while.

Not good, there is certainly something in the software that bothers the hardware, or vice versa.
Have you tried to replicate the problem of this issue?
I got the BetaFPV 2.4 1W, absolutely another world.
Let's say that ELRS works on the R9M, but certainly with big limitations and apparently also instability.
I have tried to replicate this problem on the BetaFPV and there is absolutely none.
So the code (which is the same) is fine, it's the R9M that goes haywire for some reason.

Seba112PL, 2022-07-19:
I tried to do the same as you, but it did not cause the problem for me. With mine R9M there is no problem. The problem is with all R9MM receivers that i've tried (after rebooting RX it is connecting again). R9M when combined with other companies' receivers there is no problem at all.

robustini, 2022-07-20:
> I tried to do the same as you, but it did not cause the problem for me. With mine R9M there is no problem. The problem is with all R9MM receivers that i've tried (after rebooting RX it is connecting again). R9M when combined with other companies' receivers there is no problem at all.

Did you use exactly my same settings?
How many sensors does OpenTX detect?

Packet Rate: 100 Hz Full
Switch Mode: 12ch Mixed
Telemetry Ratio: 1:2 (3888 bps)
Max power: 250 mW

robustini, 2022-07-20:
> I tried to do the same as you, but it did not cause the problem for me. With mine R9M there is no problem. The problem is with all R9MM receivers that i've tried (after rebooting RX it is connecting again). R9M when combined with other companies' receivers there is no problem at all.

Did you use exactly my same settings?
It's strange, I also have a TX16S and it does it like on the X10S Express.
It is important to use the settings below or not to give a problem, for example with telemetry ratio 1:4 it does not happen here, only at 1:2.
For that reason I believe that the module at the time of the switch is unable to manage all the channels streams plus the entire telemetry flow and the link drops.
How many sensors does EdgeTX detect?

Packet Rate: 100 Hz Full
Switch Mode: 12ch Mixed
Telemetry Ratio: 1:2 (3888 bps)
Max power: 250 mW

Seba112PL, 2022-07-20:
> Packet Rate: 100 Hz Full Switch Mode: 12ch Mixed Telemetry Ratio: 1:2 (3888 bps) Max power: 250 mW

Tried these settings. Could not replicate your issue. Link stopped working after some time anyway and I couldn't see any correlation with power switching with your settings.

I can see 22 sensors.

robustini, 2022-07-20:
Interesting, you make me want to buy another rx other than a R9 MINI to see what happens.

robustini, 2022-07-21:
@CapnBry now i am sure it is not the fault of the receiver hardware or the type.
Constantly replicating the problem now with BetaFPV 900 MHz arx nd it is exactly the same.

CapnBry, 2022-07-21:
I am pretty sure it has to do with the R9M, as I have said I used an R9MM + ES900TX and flipped that dang switch 100 times and there were zero issues. No change in LQ, no loss of link, no nothing. Just a power change.

robustini, 2022-07-21:
> ES900TX

You convinced me, rather than throwing 6 receivers in the trash now I buy a BetaFPV Micro 900 and try with this one.

robustini, 2022-07-22:
@CapnBry if you can use it as an indication for a possible debug I have ascertained that the problem is exponential to the telemetry ratio.
With 1:2 it shows up almost immediately here, at 1:32 I think I have compromised a switch on the tx by making hundreds of switches and the link has always remained stable.
All other settings do not affect the problem.
The BetaFPV 900 MHz module arrives today and I'm sure it won't happen with that, i keep all informed.

robustini, 2022-07-22:
@CapnBry confirmed, with BetaFPV Micro 868 TX no unwanted behavior with the switch from dynamic to fixed power, so it is the R9M that ELRS has some anomaly.

CapnBry, 2022-07-22:
Can you stop tagging me in your multiple posts a day? The stuff you're telling me I've already said. I do not have R9M hardware, it doesn't happen on other hardware for me. I've tested with the settings you've provided. I don't need a 3x a day check in with you telling me it keeps doing it for you.

robustini, 2022-07-22:
It's called feedback, it can help developers.
But if there is no interest good luck!

robustini, 2022-07-24:
A friend with about the same setup replicated the problem.
So I leave the issue open so that it stays on the list.

CapnBry, 2022-07-25:
I didn't mean to close the issue, just to stop tagging me because that makes GitHub start spamming emails at me. I've bought an R9M TX to attempt to replicate this which should be here this week.

robustini, 2022-08-01:
Fixed with #1734, thanks!

closed 2022-08-01 by robustini

Seba112PL, 2022-08-03:
After like 2-3h hours of testing on newest maintenance the problem for me is fixed too 😄

robustini, 2022-08-03:
> After like 2-3h hours of testing on newest maintenance the problem for me is fixed too 😄 Still testing tho so I will see if it will lock up
> 
> Edit.
> 
> Nope. Same thing. Disconnected. Only way to reconnect is to power cycle rx

We are talking about this problem in the PR that solves this issue.

CapnBry, 2022-08-03:
> After like 2-3h hours of testing on newest maintenance

The PR with the change has not been merged yet, so it is not on maintenance. But also you've probably got a different issue which would not be addressed by the changes, since the fix primarily affects the TX side (but the RX disconnects because of it)