ExpressLRS/ExpressLRS #1697: No support for Betaflight CMS script

CapnBry, 2022-07-16:
The [Betaflight Tx Lua](https://github.com/betaflight/betaflight-tx-lua-scripts) works fine, but its sibling `Betaflight CMS` does not work. The screen is just blank.

There's two issues at play here. First, we do not forward back the DisplayPort telemetry packet. This is easily resolved with the following changeset
```
diff --git a/src/lib/CrsfProtocol/crsf_protocol.h b/src/lib/CrsfProtocol/crsf_protocol.h
index db0bbf9e..9f35c637 100644
--- a/src/lib/CrsfProtocol/crsf_protocol.h
+++ b/src/lib/CrsfProtocol/crsf_protocol.h
@@ -105,6 +105,7 @@ typedef enum
     CRSF_FRAMETYPE_MSP_REQ = 0x7A,   // response request using msp sequence as command
     CRSF_FRAMETYPE_MSP_RESP = 0x7B,  // reply with 58 byte chunked binary
     CRSF_FRAMETYPE_MSP_WRITE = 0x7C, // write with 8 byte chunked binary (OpenTX outbound telemetry buffer limit)
+    CRSF_FRAMETYPE_DISPLAYPORT = 0x7D,
     // Ardupilot frames
     CRSF_FRAMETYPE_ARDUPILOT_RESP = 0x80,
 } crsf_frame_type_e;
diff --git a/src/lib/Telemetry/telemetry.cpp b/src/lib/Telemetry/telemetry.cpp
index 1232bb56..cfaac4e4 100644
--- a/src/lib/Telemetry/telemetry.cpp
+++ b/src/lib/Telemetry/telemetry.cpp
@@ -260,8 +260,8 @@ bool Telemetry::AppendTelemetryPackage(uint8_t *package)
                 else // if no TCP client we just want to forward MSP over the link
             #endif
             {
-                // larger msp resonses are sent in two chunks so special handling is needed so both get sent
-                if (header->type == CRSF_FRAMETYPE_MSP_RESP)
+                // larger MSP and DisplayPort responses are sent in two chunks so special handling is needed so both get sent
+                if (header->type == CRSF_FRAMETYPE_MSP_RESP || header->type == CRSF_FRAMETYPE_DISPLAYPORT)
                 {
                     // there is already another response stored
                     if (payloadTypes[targetIndex].updated)
```

The script then works _sometimes_ but not really. The issue is that our telemetry system doesn't provide a stream of data, it has slots for pending telemetry items. When the CMS does load, you'll see a set of MSP transfers like this
```
<EE 07 7D C8 EA 03 09 20 86                      î.}Èê..  // DisplayPort init, 9 rows, 20 cols
<EE 05 7D C8 EA 05 BD                            î.}Èê.½  // DisplayPort poll
>C8 05 7D EA C8 02 AB                            È.}êÈ.«  // DisplayPort clear
>C8 39 7D EA C8 01 A0 00 A0 22 AD 02 20 4D 41 49 È9}êÈ. . "­. MAI  // CMS display part 1
4E 20 AD 02 A0 16 3E 50 52 4F 46 49 4C 45 A0 13 N ­. .>PROFILE .
3E A0 05 46 45 41 54 55 52 45 53 A0 12 3E A0 05 > .FEATURES .> .
4F 53 44 A0 17 3E A0 05 46 43 F0                OSD .> .FCð
>C8 2C 7D EA C8 01 60 01 26 46 49 52 4D 57 41 52 È,}êÈ.`.&FIRMWAR  // CMS display part 2
45 A0 F 3E A0 05 4D 49 53 43 A0 16 3E A0 05 53 E .> .MISC .> .S
41 56 45 2F 45 58 49 54 A0 11 3E A0 22 75       AVE/EXIT .> "u
```

Note the problem here is that there are 3 packets and we only have 2 telemetry slots to store them in. Sometimes the first one goes out before the second two come in, there are enough slots available, and the CMS loads on the handset. The sticks may work as they would in CMS, assuming the sequence above of all 3 packets getting successfully queued and transmitted from the RX to TX. It can be seen here again on the next screen
``
>C8 05 7D EA C8 02 AB                            È.}êÈ.«
QUEUED slot 7
>C8 38 7D EA C8 01 81 00 A0 22 AD 03 20 46 45 41 È8}êÈ.. "­. FEA
54 55 52 45 53 20 AD 03 A0 10 3E 42 4C 41 43 4B TURES ­. .>BLACK
42 4F 58 A0 12 3E A0 05 56 54 58 A0 17 3E A0 05 BOX .> .VTX .> .
4C 45 44 20 53 54 52 49 50 9F                   LED STRIP
QUEUED slot 7
>C8 38 7D EA C8 01 01 01 A0 11 3E A0 05 50 4F 57 È8}êÈ... .> .POW
45 52 A0 15 3E A0 05 46 41 49 4C 53 41 46 45 A0 ER .> .FAILSAFE 
12 3E A0 03 56 20 50 45 52 53 49 53 54 45 4E 54 .> .V PERSISTENT
20 53 54 41 54 53 A0 0A 3E 37                    STATS .>7
QUEUED slot 8
```
There's also another way it can fail, which is that the packets are sent out of order. We don't guarantee telemetry packets go out in the order they come in, so sometimes the second packet goes first then the first packet and that's not valid either.

To get this to work, we'd have to change our telemetry system to stream properly, perhaps using a ring buffer instead of the dedicated slot system we have now. Just thought I'd make some notes about why this doesn't work current and won't work for 3.0 in case others are looking for it.

0crap, 2022-07-17:
I noticed that as well, thx for the confirmation.
No biggy at all, normal BF Lua works fine.

schugabe, 2022-08-05:
I also had a look some time ago and got some output displayed but decided that the CMS lua script does not provide any real benefit compared to the bf lua script and it's simply not worth the effort to change the telemetry system for that niche usecase.

klutvott123, 2023-09-05:
We did some changes in BF to make the CMS script work with ELRS too. Should work in BF master.

pkendall64, 2024-04-06:
@CapnBry shall we close this now that BF has made changes that allow the script to work with ExpressLRS?

CapnBry, 2024-04-06:
Absolutely. The delay added to BF definitely seems to make it work almost all of the time now.

I'd still like to consider the hybrid queue system on our RX side to make it more robust at some point, but Betaflight definitely proves you can work around it in our current setup if you just show a little restraint in your sending 😂

closed 2024-04-06 by CapnBry

0crap, 2024-04-08:
Tested this as well on BF 4.5 latest dev. build.
The CMS Lua script now works, but only on high Packet rates.
333Hz and higher.
(ELRS bumps the telem. ratio to get the highest possible bandwidth for the packet rate used.)

Still a niche use case, fully understand to leave things as they are on the ELRS part IMHO.
The normal BF Lua script works fine, also on lower Packet/Telem. rates. 👍