ExpressLRS/ExpressLRS #1711: New LUA script got stuck on "Loading" with ExpressLRS3.0.0-RC1 flashed on HGLRC Hermes TX module

woodoo81, 2022-07-22:
## Current Behavior

When running LUA script (the version for ExpressLRS v3.0.0-RC1) on my TX16S radio with EdgeTX v2.7.1 and HGLRC Hermes TX module with ExpressLRS v3.0.0-RC1 flashed on it, the screen is stuck on "Loading...".

Note all works fine when I flash ExpressLRS v2.5.1 and copy corresponding LUA script on my HW.

## Steps to Reproduce

1. Flash HGLR Hermes TX module to ExpressLRS v3.0.0-RC1 using the ExpressLRS Configurator v1.4.2 (I tried both UART and WIFI, but with the same result)
2. Download the LUA script using the "Download LUA script" button in ExpressLRS Configurator.
3. Turn on the TX16S radio, select the model with ExpressLRS external module configured (one the I know it works on ExpressLRS v2.5.1).
4. Go to the LUA script in "tools" directory and execute it.

## Possible Solution (Not obligatory)

- Something wrong in the LUA script or the ExpressLRS configurator points to a wrong / not up-to-date version?

## Details

As I wrote all works fine for me when using WxpressLRS v2.5.1 so there is just a very small chance that this is a user error. ;-)
My motivation to upgrade to ExpressLRS v3.0.0 is that I bough a new Skystarsrc 2.4GHz diversity receiver and I don't see a target for it in the ExpressLRS v2.5.1. I found it only in ExpressLRS v3.0.0-RC1...

## Your Environment

- TX hardware: HGLRC Hermes ExpressLRS 2400TX
- RX hardware: N/A (but I'm using HGLRC Hermes ELRS 2400RX-S, Hermes ELRS 2400RX-T and would like to start with Skystars ELRS 2.4GHz Nano Dual Antenna)
- Handset model: TX16S with hall sensors
- EdgeTX version "Black Pearl" v2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.0.0-RC1
- Packet Rate: N/A for this issue (but I usually run 500 for freestyle and 150 for long range)
- Telemetry Ratio: N/A for this issue (but I run 1/64 for 500 and 1/32 for 150)

robustini, 2022-07-22:
It's an EdgeTX problem, if you have a widget in the background which includes the use of CRFS that happens.
If you install a recent compiled version you will see that it works, or try with OpenTX.

woodoo81, 2022-07-22:
I tried with latest nightly build of EdgeTX and also OpenTX v2.3.15 (latest version) and I'm getting still the same problem! :-(
I guess the problem will be somewhere else...
Any other suggestions?
Attached is a screenshot of config. from ExpressLRS Configurator as I have it setup to flash my TX module.

[image]

woodoo81, 2022-07-23:
I "debug" the issue a bit more (by placing `lcd.drawText(...)` all over the place in the LUA script) and it looks like the 3.0.0-RC1 firmware simply bricks the HGLRC Hermes 2400 TX module. I mean the OpenTX / EdgeTX is not able to communicate with the TX module anymore.
It gets stuck in `refreshNext()` function (in LUA script), never getting anything else than nil from `crossfireTelemetryPop()` function.
It keeps getting to 
`elseif time > devicesRefreshTimeout and fields_count < 1  then`
and 
`elseif time > linkstatTimeout then`
branches in the `refreshNext()` function, sending commands via `crossfireTelemetryPush()`, but that's all. Never gets out of there...

CapnBry, 2022-07-25:
Can you try flashing with Configurator's "Pull Request" feature using this PR?
https://github.com/ExpressLRS/ExpressLRS/pull/1716

Be sure to do a "Clear Firmware Files" from the Configurator Support -> Troubleshooting section, as the build process is not smart enough to know the required file has changed and will just keep bootlooping if the firmware image isn't actually rebuilt again.

CapnBry, 2022-07-26:
Awesome! Thanks for reporting back. Yeah the PR got merged before you could test it I guess, which was a little premature considering it takes it out of the list. Sounds like you're a real clever person though and figured out how to test it, so I appreciate that.

I'm not sure that PR will connect to 3.0.0-RC1 because we're made some changes to further break compatibility with the 2.x software, which also breaks compatibility with 3.0.0-RC1. The RX will need to be on 3.x.x-maintenance branch until RC2 comes out (this week probably) I believe.

woodoo81, 2022-07-26:
OK, clear. I'm on the commit for now and will update to RC2 when available.
Thank you again! People around ExpressLRS are awesome! :-)

woodoo81, 2022-07-26:
Please close the issue or do whatever should be done with it. I'm not familiar with ExpressLRS dev. procedures and I don't want to mess it up. ;-)

closed 2022-07-26 by pkendall64