ExpressLRS/ExpressLRS #1719: Matek R24-P RxBat displays approx correct voltage and drops and drops

miggy1952, 2022-07-26:
Running RC1 3.0
Matek R24-P 
Namimno OLED
TX16s

Getting some form of Rx battery voltage. It starts off at something close to the correct Rx voltage but then falls gradually to 1.5V and then holds 1.5V

CapnBry, 2022-08-05:
Aer you using the proper external voltage divider to lower VBAT down to below 1V? Are you the person who I talked to on Discord about this? I've tested and this works fine for me, for am hour it just holds the correct VBAT.

miggy1952, 2022-08-05:
Thanks and yes it was me. Please can you explain to me the voltage divider, is that in EdgeTX or a physical wired soldered thing?

CapnBry, 2022-08-07:
> Please can you explain to me the voltage divider, is that in EdgeTX or a physical wired soldered thing?

It is physical. The max input voltage for the VBAT "pad" is 1.1V, but should be below 1.0V. Hooking VBAT directly to this pad does terrible things to the receiver that you'd expect from putting, say, 12V on a 1V input. A voltage divider is used to bring the voltage down to the appropriate level, and was not included in the original designs because nobody bothered with it (apart from crteensy's reference design)

Information about what is hardware needed is in the original PR
https://github.com/ExpressLRS/ExpressLRS/pull/1466#issuecomment-1123992909

miggy1952, 2022-08-14:
CapnBry is a hero!

closed 2022-08-14 by miggy1952