ExpressLRS/ExpressLRS #1726: 3.0.0_RC1 Flash to HAPPYMODEL_ES24TX_2400_TX Fails

DocTrucker, 2022-07-28:
## Current Behaviour

Flash update to a Happymodel 2.4ghz JR module bay Tx module fails. Web update suggests a 'Typemismatch'. Force update via web also fails.

## Steps to Reproduce

Using ExpressLRS configurator on Ubuntu options:
Release: 3.0.0-RC1

Target:
Device category: Happymodel 2.4 GHz
Device: HappyModel ES24TX 2400 TX

Flashing Method:
WIFI selected

Device options:
Standard Mode selected

Regulatory domains:
REGULATORY_DOMAIN_EU_CE_2400 enabled

Binding phrase setup:
BINDING_PHRASE enabled
Custom binding phrase specified.

Compatibility options
UART_INVERTED enabled

Extra data:
TLM_REPORT_INTERVAL_MS disabled

Network:
AUTO_WIFI_ON_INTERVAL enabled
Value 60
HOME_WIFI_SSID enabled
Value specified
HOME_WIFI_PASSWORD enabled
Value specified

Other options:
WS2812_IS_GRB disabled
## Possible Solution (Not obligatory)

Actions:
Tx Module auto appears on "WiFi Device Selection" after selecting "Enable WiFi" option on the LUA script (express 2.5.1) page under "WiFi Connectivity".

Tried both "BUILD" followed by using the webpage to load the firmware and "BUILD & FLASH"

## Details

## Your Environment

Hardware: Ubuntu
ExpressLRS Configurator: expresslrs-configurator_1.4.2_amd64.deb
Tx: Jumper T16 Pro
Tx module: Happymodel ES24TX 2.4G ExpressLRS Micro TX Transmitter ( https://www.ebay.co.uk/itm/164983737086 )
Existing ExpressLRS version on Tx Module 2.5.1

## Attached files

[ExpressLRSBuildLog_20220728083828689_FailWifiUpdate.txt](https://github.com/ExpressLRS/ExpressLRS/files/9207535/ExpressLRSBuildLog_20220728083828689_FailWifiUpdate.txt)
[ExpressLRSBuildLog_20220728084249717_BuildOnly.txt](https://github.com/ExpressLRS/ExpressLRS/files/9207537/ExpressLRSBuildLog_20220728084249717_BuildOnly.txt)
[image]

pkendall64, 2022-07-28:
Yes, thats correct, because the targets are now Unified targets, you will need to select Flash anyway.

DocTrucker, 2022-07-29:
Please share a link to something discussing the "Unified targets".

I pressed force anyway and it failed to update.

DocTrucker, 2022-07-29:
I will run the force anyway and capture screenshots of the results. Is it perhaps something to do with the fixes applies to force wifi flash needing to be apploed to the website flash?

https://github.com/ExpressLRS/ExpressLRS/pull/1600

DocTrucker, 2022-07-29:
These are the screen shots of the webpage update process and it's ultimate failure even after force flash - bad size error.

I tried with the 3.0.0-rc1 lua script and had the same result.

[image]

[image]

[image]

[image]

pkendall64, 2022-07-29:
You will probably be better opening it up and flashing using the the USB connector and choosing UART flash in configurator.

DocTrucker, 2022-07-29:
This is the resulting logs  from the "Build & Flash" and "Force Flash" options on the ELRS Configurator.

[image]

[ExpressLRSBuildLog_20220729083336281_BuildAndFlashFail.txt](https://github.com/ExpressLRS/ExpressLRS/files/9218254/ExpressLRSBuildLog_20220729083336281_BuildAndFlashFail.txt)

[ExpressLRSBuildLog_20220729083145272-ForceFlashFail.txt](https://github.com/ExpressLRS/ExpressLRS/files/9218258/ExpressLRSBuildLog_20220729083145272-ForceFlashFail.txt)

DocTrucker, 2022-07-29:
The UART and WiFi flashing from configurator isn't working for me at all. Only way I have had successful updates is via the webpage. Webpage update fails with bad size.

I suspect this is two issues; the configurator isn't happy on my Ubuntu machine, & the bad size issue on the webpage based update.

pkendall64, 2022-07-29:
The "Bad Size" issue is because the upload page doesn't seem to be supplying the X-FileSize header with size of the binary file.
You could try from the command line using curl. 
Find the bin file, ls -al firmware.bin, to get the file size and put then in the place of xxx below.
```
curl -Fupload=@firmware.bin -H "X-FileSize=xxx" http://elrs_tx.local/upload?force
```

DocTrucker, 2022-07-29:
Will have a look at that later. Why would that be an issue post 3.0.0-rc1 with force flash, but not 2.5.1 with normal flash?

pkendall64, 2022-07-29:
It shouldn't, Ive been over that code so many times! You could even try opening the web-page on the module with ?force at the end of the URL so it will just do the force and not check for the mismatch.
i.e. http://elrs_tx.local/?force
Also try with the module really close to the wifi router as well.

pkendall64, 2022-07-29:
Also, you should join the discord server and ask in the help-and-support channel, there will be plenty of helpful folk there who can help too.

DocTrucker, 2022-07-29:
Yeah, trying to get on discord but they decided I was a spam bot on log in and blocked me until late last night!

DocTrucker, 2022-07-29:
Curl not behaving properly either...

[image]

DocTrucker, 2022-07-29:
http://elrs_tx.local/?force still results in a Targets mismatch warning. Ultimately fails anyway.

DocTrucker, 2022-07-29:
UART flashing issue solved. Power only USB lead! :-D

Think the wifi build and flash issues relate to curl, as when I manually run curl it fails to find the file, despite ls having justbreported its size fine.

Type mismatch and bad size issues remain.

pkendall64, 2022-07-29:
I'm glad you got it sorted, but it seems odd that curl is broken on your system!

tomhorst2004, 2022-07-29:
Currently having the same problem with the slim pro series, will try to do it via uart  but flashing via wifi does not work (exact same problem)

SplinterII, 2022-07-30:
Exactly same story here!

fraro, 2022-08-29:
same problem here (2.5.1 -> 3.0.0rc2):
curl -F upload=@firmware.bin -H "X-FileSize: 1455061" -X POST   http://192.168.1.228/forceupdate
results in
{"status": "ok", "msg": "Update cancelled"}
!?

pkendall64, 2022-08-29:
forceupdate is the command to complete an update after an upload.
You need upload?force instead which will do the upload and assume you want to force flash the device.
I that does not work, then a UART flash is required.

fraro, 2022-08-29:
Since UART seems to be the only way, I would like to drop https://www.youtube.com/watch?v=yJ8Gu9XVjV0 for the next BetaFPV micro 1W users spending hours to get the UART (USB-C) update working :-/
Now my TX is on 3.0.0RC2, which hopefully solves an issue on the RX (PWM) side with my ESC...

pkendall64, 2022-09-02:
We've figured out the issue and provided a solution in the pinned comments in the #help-and-supprt channel on discord.

closed 2022-09-02 by pkendall64

mazzus, 2022-09-21:
This solved the issue for my HappyModel ES24TX Pro :smile: 

For anyone who doesn't want to search for the pinned message on the discord:

> deadbyte — 09/01/2022
>
> Regarding "Bad Size Given" on some TX modules being flashed to 3.x, here's your fix:
> https://github.com/pkendall64/repartitioner#what-is-this
> use the Releases page: https://github.com/pkendall64/repartitioner/releases/tag/1.0
> and download the 
> repartitioner.bin
> and flash it first to your tx module,
> once flashed, exit the lua script and relaunch it, then reactivate the WiFi on the module and upload the 3.x firmware
> tagging @V-22 @commanderguy3001