ExpressLRS/ExpressLRS #1733: Mixing tough and scroll support in Lua breaks input

CapnBry, 2022-08-01:
ExpressLRS 3.x.x-maint elrsv3.lua (latest)

## Steps to Reproduce

1. Enter Lua on a touchscreen device (TX16S used), click any `luaItem_selection` item to start changing the value
2. Swipe right to try to change the value, the loading bar comes up?
3. Note the srollwheel no longer works
4. Use Exit/Rtn physical button to exit edit mode
5. Scrollwheel doesn't work at all any more

CapnBry, 2022-08-08:
I've debugged this, and after any touch event, EdgeTX stops sending us `EVT_VIRTUAL_NEXT / EVT_VIRTUAL_PREV` events. I still get `EVT_VIRTUAL_EXIT / EVT_VIRTUAL_ENTER` but the scroll wheel no longer generates an event.
```lua
local function run(event, ts)
  lcd.clear()
  if ts then
    lcd.drawText(0, 0, string.format("%ux%u", ts.x, ts.y))
  elseif event ~= 0 then
    lcd.drawText(0, 0, tostring(event))
  end
  
  return 0
end

return { run=run }
```

With this code, when I use the scroll wheel, I see the event for the scroll appearing briefly on the screen, but then I tap, and I never get another scroll event. EdgeTX 2.7.0, 2.7.1 and the PR#2100, but works fine in the simulator

pkendall64, 2022-08-16:
@CapnBry should this be closed now?

CapnBry, 2022-08-16:
Oh yes sorry, I need to open a EdgeTX issue for it I suppose.

PR #1753 (2022-08-14): Fix popup touch events, remove from B&W
This refactors the touch events slightly and fixes the issue of no way to exit a popup with touch events. It also reduces the Lua compiled code size on all platforms, and 904 bytes on B&W.

### Details
I started this cleanup to fix #1733, but once I cleaned it up, I found that the bug is on the EdgeTX side. Considering this reduces Lua memory usage on all platforms, and fixes the bug of not being able to exit a popup, I'mma put it on ya, gurl.

* There was no way to exit a popup message with touch, as all the events were translated inline for every insta [... cut]
closed 2022-08-16 by CapnBry