ExpressLRS/ExpressLRS #1736: RM TX16s /w Namimno Flash TX OLED (V1 board) fan no spin on 3.0.0RC1

ClockwiseCounterclockwise, 2022-08-03:
## Current Behavior

- On ELRS FW 2.5.1 the Namimno Flash TX OLED (V1 PCB board) fan spins.
- When ELRS FW is updated to 3.0.0RC1 fan does not spin

## Steps to Reproduce

1. Started ELRS Configurator v1.4.2
2. Enable 'Show pre-releases'
3. Selected 3.0.0-RC1
4. Selected NamimnoRC FLASH 2.4 GHz target
5. Selected NamimnoRC FLASH 2400 OLED TX device
6. Flashing Method = UART

7. Device options:
    - Only change was BINDING_PHRASE set to a custom value, no other changes
8. Connect Namimno Flash TX OLED module to PC using USBC
9.  Select BUILD & FLASH
10. Flash completes, disconnect from USBC connector and insert module to TX16s
11. Turn on TX16s, wait for TX module to complete the startup process
12. Fan does not spin when start up completed

13.  Flash the module back to 2.5.1 and the fan works again.
14. Flash back to 3.0.0RC1 fan does not work, flash back to 2.5.1 fan works.

## Possible Solution (Not obligatory)

## Details

Deadbyte assisted with the issue on ELRS Discord help group.  He had tried on his Namimno FLASH TX module and it works on 3.0.0-RC1.  The only difference between our modules is that I have version 1, and he has version 2 of the PCB.

Fan not working and unit heats up too much.  Would like the fan working on 3.0.0 for my version 1 unit.

## Your Environment

- TX hardware: Namimno Flash 2.4 ghz OLED TX
- RX hardware: Namimno Flash 2.4 ghz  ESP RX
- Handset model: RadioMaster TX16s
- OpenTX version (including nightly number)  EDGETX 2.7.1
- ExpressLRS version (TX & RX MUST MATCH):  3.0.0-RC1 for both TX & RX
- Packet Rate: 250
- Telemetry Ratio: 64
- user_defines: (not sure what this means)

CapnBry, 2022-08-05:
I have a Namimno Flash OLED v1 as well and just tested 3.0.0-RC1 and the fan comes on as expected. Maybe check the Fan Threshold in the Lua to verify it is set properly and verify with the OLED that power is at or above that threshold? Note that it also takes like 10 seconds before the fan kicks on as well, to prevent turning it on and off for just a few seconds.

ClockwiseCounterclockwise, 2022-08-08:
Everything you've mentioned I've tried with support and advice from deadbyte on my initial encounter to no avail.  

I did further investigation yesterday and have noticed something weird.  The RMTX16s is on EdgeTX 2.7.1 and the weird thing is that the fan _does_ come on BUT only for one model. When I switch to the other models the fan doesn't work...(weird?) I have not made any changes to EdgeTX prior to this issue, so I'm at a loss as to why it is not working for the other models. I will look further into this. 

Thanks for assisting/checking Cap'in!

ClockwiseCounterclockwise, 2022-08-29:
Hello, seems this issue came back.  This time it happened after I updated to 3.0.0RC2.  Difference this time is that the SD Card is OK, no corruption.

Hardware:
Namimno Flash OLED TX (V1 board) & Flash RX 2.4

Issue: Fan does not spin on all models except 1.

Tried:
- switching models and waiting around 20sec for fan to kick in > fan does not spin except for 1 model
- Tried setting the [Fan On] @25mw / 50mw > no fan spinning

CapnBry, 2022-08-29:
Worked for me on 3.0.0-RC1, still works for me now on 3.0.0-RC2 with Namimno OLED (not sure of version). 
* Set DynPower = off, Fan Thresh = 250mW, Max Power = 250mW. Turns on after a few seconds, stays running 30 seconds after power lowered
* Set DynPower = off, Fan Thresh = 10mW, Max Power = 25mW. Same, except stays running until thresh raised
* Set DynPower = on, Fan Thresh = 250mW, Max Power = 250mW. Fan does not come on. Arm, and fan comes on after a few seconds and stays on until power down. (No receiver)

Maybe your fan is bad? I dunno I got nothing for you, it is not a bug in the software. It is either bad configuration or bad hardware.

ClockwiseCounterclockwise, 2022-08-29:
I deleted all models (again) and recreated the models.  The fan comes on again after recreating all models...not sure why this is happening.  I'm going to reload EdgeTX 2.7.1 again fresh and reload all files from the SDCard zip file. We'll see if this solves the issue.

Thanks for checking again Cap'in.

CapnBry, 2022-08-30:
There's gotta be something we're missing with why one model would work and the rest wouldn't and then no models worked. The code isn't complicated so there's not much that can go wrong.

Just to be sure we're on the same page 100%, you said you set [Fan On] @ 25mW, but there is no Fan On setting so I assumed you meant [Fan Thresh] in the Lua. Was that assumption correct? I'd hate if we just weren't talking about the same thing.

ClockwiseCounterclockwise, 2022-09-10:
@CapnBry My apologies, yes  you're correct.  I was referring to the [Fan Threshold].   

Unfortunately the issue is still there..

I've since gone through quite a bit of testing.  Both on 2.5.1 and 3.0.0RC2.

My findings for **FW 2.5.1**:
---------------
ELRS FW: 2.5.1
Packet Rate: 250Hz
Telem Ratio: 1:32
Switch Mode = Wide
Model Match = Depending on selected quad will be from 1-4
TX Power:
Max Power tired following: 250mW or 500mW
Dynamic = Dyn
Fan Thresh 25mW
---------------
Hardware:
RM TX16S /w external Namimno module
---------------
Namimno 2.4 OLED TX (V1 PCB)
Namimno FLASH RX 2.4
---------------

To start clean, I re-flashed the TX with EdgeTX 2.7.1 and cleared the SD Card and loaded with the download SDCard content.

4 quads created on the TX.  All 4 created using [Blank Model] entry in EdgeTX2.7.1

All models setup with mixes / logical switches and special functions.  They are all consistent with 1 model that has a GPS which has been setup with an extra switch.

Steps used in testing:

Current selected model = fan starts 2-4 sec after the last switch position call.
Switch to model 2 = fan starts 2-4 sec after the last switch position call.
Switch to model 3 = fan starts 2-4 sec after the last switch position call.
Switch to model 4 = fan starts 2-4 sec after the last switch position call.

Power off TX > Power on TX after 15min = same results as above.
Next day power on TX and same results as above.

I've also tried random model selection combinations. (e.g. 1-3-2-4 / 2-4-1-3 / etc..)

=================================================

My findings for FW **3.0.0RC2**:
---------------
ELRS FW: 3.0.0RC2
Packet Rate: 250Hz
Telem Ratio: 1:32
Switch Mode = Wide
Model Match = Depending on selected quad will be from 1-4
TX Power:
Max Power tired following: 250mW or 500mW
Dynamic = Dyn
Fan Thresh 25mW

For 3.0.0RC2 I also re-flashed the TX with EdgeTX 2.7.1 and cleared the SD Card and loaded with the download SDCard content.

4 quads created on the TX.  All 4 created using [Blank Model] entry in EdgeTX2.7.1

All models setup with mixes / logical switches and special functions.  They are all consistent with 1 model that has a GPS which has been setup with an extra switch.

Steps used in testing:

Current selected model = fan started 11 sec after the last switch position call.
Switch to model 2 = fan started 14 sec after the last switch position call.
Switch to model 3 = fan started 17 sec after the last switch position call.
Switch to model 4 = fan started 22 sec after the last switch position call.

Power off TX > Power on TX after 15min =

Model1 X (waited over 1min no fan spin)
Model2 O (fan started after 35sec)
Model3 X (same as model 1)
Model4 X (same as model 1)

Next day power on TX = fan does not start for all models. Tried switching models in random order and still no fan start up. (Did not change any settings parameters)

I understand that I can use 2.5.1, but 3.0.0RC2 seems to have better signal in certain areas where I fly.

Please let me know what logs/info on my end that I can provide to assist with resolving this issue I'm experiencing.

Thanks again for your time and assistance in this matter.

Cheers

ClockwiseCounterclockwise, 2022-09-11:
@CapnBry - Further investigation results:

I found that by enabling Wifi in the ELRS > Wifi connectivity menu the fan starts up.

I switch to a model and if the fan will not start (after x min), I navigated to the ELRS > Wifi Connectivity menu > Wifi submenu enable Wifi and the fan will start spinning...

After finding that turning on the wifi will start the fan I switched to the other models where the fan will not start and tried the same steps.  Fan started for all models.!

Hope this information helps in resolving the issue.

ClockwiseCounterclockwise, 2022-09-12:
@CapnBry - I just updated to 3.0.0 Final release.  Unfortunately it is showing the same symptoms.

No fan spin (for all models) until I go into the ELRS > Wifi Connectivity > Wifi > enable  = fan starts spinning.

pkendall64, 2022-09-12:
Can you visit the /hardware.html page in the web UI and see if a warning message is at the top of the screen.
If there is then there is a reset link which you should click to reset all settings back to the provided defaults.

ClockwiseCounterclockwise, 2022-09-13:
Hello Paul, there's no warning. (on both Namimno TX modules I have)
[image]

Another setting that Bryan asked to validate on Discord just now:
[image]

closed 2023-04-14 by github-actions[bot]