ExpressLRS/ExpressLRS #1747: NamimnoRC Flash TX 2400 OLED no 10mW mode in v3.0.0-RC1

HazzaHFPV, 2022-08-05:
It's basically just a quick question:
With v2.5.1 I can select 10mW output power on my NamimnoRC Flash 2400 OLED TX. On v3.0.0-RC1 the minimum I can select is 25mW, the 10mW Option is gone.
I only noticed this because I wanted to do a bench test.
I am not sure if this is a bug, feature or if the problem is sitting in front of the radio (i.e. me being an idiot).
Thank you for your help :)

deadbytefpv, 2022-08-05:
It's always been a minimum of 25mW since 2.x
https://github.com/ExpressLRS/ExpressLRS/blob/36493b88c59cebfbc3ba7fdf24027ff2f9514234/src/include/target/NamimnoRC_FLASH_2400_OLED_TX.h#L44

HazzaHFPV, 2022-08-05:
Hm. Thank confuses me. I just flashed back to 2.5.1 and 10mW is selectable again. Strange..

[image]
[image]

HazzaHFPV, 2022-08-05:
Yeah the problem was in front of the radio. The 10mW EU regulatory option was selected. Then it shows 10mW.
Question then is: dies it only show 10mW but is outputting 25mW in this mode as the hardware cant go lower?

deadbytefpv, 2022-08-05:
> Yeah the problem was in front of the radio. The 10mW EU regulatory option was selected. Then it shows 10mW. Question then is: does it only show 10mW but is outputting 25mW in this mode as the hardware cant go lower or is it really doing 10mW in this case?

It shouldn't even possible to get it flashed with EU_CE domain on 2.x since minimum for 2.x EU_CE is 10mW.
So I think this is on the Configurator side of things.

CapnBry, 2022-08-05:
It is outputting 25mW, it can't go any lower, even on 2.x. Everything displays 10mW because that's the only value actually available to the backend, but the output power is actually 25mW.

In 3.x with LBT the limit is 100mW so now the backend has the actual power available and displays the device's true minimum power of 25mW.

closed 2022-08-18 by github-actions[bot]