ExpressLRS/ExpressLRS #1751: -DUSE_OLED_I2C not working in 3.0 firmware?

DennyRoadman, 2022-08-06:
## Current Behavior

ExpressLRS Configurator v1.4.2
Firmware 3.0.0-RC1 
Try to update my DIY TX, update successful but OLED display is not working.

In compilation log there are a lot of messages like:

{
  context: {
    data: 'In file included from <command-line>:0:0:\r\n' +
      'include/target\\Unified_ESP32_TX.h:131:0: warning: "USE_OLED_I2C" redefined\r\n' +
      ' #define USE_OLED_I2C\r\n'
  },
  level: 'info',
  message: 'logs stream output'
}

Re build and flash it to 2.4.0 and 2.5.1 and Display working fine!

How to set up it in 3.0???

## Details

## Your Environment

- TX hardware: DIY 2400 TX ESP32 SX1280 E28
- Handset model: TX12
- OpenTX version (including nightly number)
- ExpressLRS version (3.0.0-RC1):
- Packet Rate: 150hz
- Telemetry Ratio: 1-32
- user_defines:

-DMY_BINDING_PHRASE="bindingphrase"
-DRegulatory_Domain_ISM_2400
-DFEATURE_OPENTX_SYNC
-DUART_INVERTED
-DAUTO_WIFI_ON_INTERVAL=60
-DUSE_OLED_I2C

pkendall64, 2022-08-06:
Remove the USE_OLED_I2C from the settings.
Once you flash the device, connect to the wifi `http://elrs_tx.local/hardware.html` and you can add the pin definitions and select the screen type on the page.

DennyRoadman, 2022-08-07:
Thanks it's working perfect! Great job! there are a lot of options on this page ;=) appreciate=))))

pkendall64, 2022-08-07:
Yes, that page is a UI for the hardware.json file which v1 will use for all ESP based TXes and RXes.

closed 2022-08-07 by pkendall64