ExpressLRS/ExpressLRS #1754: Lua script says my module is flashed for ISM but it is flashed for CE

AtthosTheGreat, 2022-08-09:
## Current Behavior

Go to ExpressLRS lua script and read the last row before the EXIT row. It says ISM2G4 instead of CE.

## Steps to Reproduce

1. Flash ExpressLRS 3.0.0-RC1

2. Get the latest lua script (as of 06.08.2022)

3. Go to ExpressLRS lua script

4. Read the last row before the EXIT row. It says ISM2G4 instead of CE

5. NOTE: If I go inside TX Power I can see that it has the 100mW CE LIMIT, so the module is flashed for CE, not ISM. The bug probably comes from the lua script

## Your Environment

- TX hardware: Internal Radiomaster TX16S ELRS module
- RX hardware: not important, this bug is about the TX module only
- Handset model: Radiomaster TX16S with internal ELRS module
- EdgeTX version: 2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 3.0.0-RC1

deadbytefpv, 2022-08-09:
Should now be resolved on RC2
https://github.com/ExpressLRS/ExpressLRS/pull/1740

AtthosTheGreat, 2022-08-09:
I can confirm it's resolved on RC2. I'll close this as completed

closed 2022-08-09 by AtthosTheGreat