ExpressLRS/ExpressLRS #1756: 3.0 RC2 Target mismatch for HM 2.4 Pro

fordtom, 2022-08-09:
As the title says, getting a target mismatch when trying to flash my HM Pro 2.4 Tx to 3.0 RC2. I was aware of the size issue that affected this module with RC1, is this the same issue showing itself in a different way? I've also seen people saying flash via USB works; while I will use this to test the RC am I going to have to flash my module via USB for every future firmware version?

deadbytefpv, 2022-08-09:
I wasn't able to reproduce this issue on my own HM ES24TX Pro, flashing it to 3.0RC1 once via USB, going back to 2.5.0, then flashing back to 3.0RC1, and then going back to 2.4.0, then flashing back to 3.0RC1 (all these back and forths done via WiFi after that initial via USB flash)

fordtom, 2022-08-09:
so just the very first time to 3.0 needs to be USB and after that it's fine? Seems strange but I'll give it a go, anyone know the reason why it's like this?

jawz101, 2022-08-14:
I got the same Target mismatch on a Radiomaster TX16S Mark II updating from 2.5.1 to 3.0 RC 2 via wifi

closed 2022-09-02 by pkendall64

MartinWa, 2022-09-10:
Can confirm that the "repartitioner" solved the problem for me. Here is the repo:
[](https://github.com/pkendall64/repartitioner)