ExpressLRS/ExpressLRS #1759: ExpressLRS documentation confusion (Bardwell style)

jawz101, 2022-08-10:
## Details

I've been following a few documents and videos to get my new Radiomaster tx16s Mark II (ELRS model) setup with my new quad and I ran into confusion on how to reconcile the "[all modes on one channel](https://www.youtube.com/watch?v=Ut7Hpmmy_Us)" setup and the ExpressLRS documentation.

[https://www.expresslrs.org/2.0/software/switch-config/#aux-8-hybrid)

The document repeatedly says arm should be on AUX1

they then say AUX8 (Hybrid) is where you'd set up a "Bardwell style" stacked mode setup.

So do they want a total of 2 AUX's used with arm on AUX1 and everything else on AUX8  if you want to do the Bardwell-style setup?

Furthermore, they say for the Dynamic Transmit Power feature

[OSD Power Readout](https://www.expresslrs.org/2.0/software/dynamic-transmit-power/#osd-power-readout)

... under the OSD Power Readout heading, it says to display your Tx uplink power in your Betaflight/iNav OSD, you must have your switch mode in wide (as opposed to the Aux8 hybrid bardwell thing above).

So my question is "how would I Bardwell my switches with everything on AUX8 with the radio set up in AUX8 (Hybrid Switch Mode) if arm has to be on AUX1 and to see my TX power on my OSD must be in Switch Mode Wide instead of Hybrid?
## Your Environment

- TX hardware: [ELRS EL24P 2.4G](https://flywoo.net/products/0-4g-flywoo-2-4g-expresslrs-elrs-el24e-el24p?variant=40438891708611)
- RX hardware: Radiomaster tx16s Mark II (internal ELRS model)
- Handset model: Radiomaster tx16s Mark II (internal ELRS model)
- EdgeTX 2.7.1
- ExpressLRS version (TX & RX MUST MATCH): 2.5.1
- Packet Rate: 150Hz
- Telemetry Ratio: 1:32
- user_defines:

commanderguy3001, 2022-08-10:
simply said: ALWAYS put your arm switch on AUX1, independent of what you do with all the other channels.
bardwell style uses channel 8 for everything, but aux1 is still needed by a couple features of the link itself, and I would strongly encourage you to just set betaflight to use AUX1 for arming too, and do bardwell style on AUX8 if you really want to, but leave arming out of there, as it makes things simpler

jawz101, 2022-08-10:
The thing is, Bardwell style in his videos **puts all the things on AUX1** including Arm.  So if Arm has to be on AUX1 but stacking can only be done on AUX8, it's not really the same thing as a "Bardwell style."

commanderguy3001, 2022-08-11:
just have arming on AUX1, and only on that. And have everything else on whatever other aux channel you want. Also, to (partly) answer your second question, why do you need to use hybrid mode in the first place, there really is no good point to it, just use wide mode, it doesn't have any real disadvantages.

jawz101, 2022-08-11:
I think it is just that the documentation page says AUX8 (Hybrid).  It throws me off.

commanderguy3001, 2022-08-11:
yeah the documentation definitely isn't the most up to date anymore, but usually you just want to use AUX1 for arming as per design, and then use wide switch mode (128 positions per channel, up to AUX 6 or 7 I think) for everything else.

closed 2022-08-11 by jawz101