ExpressLRS/ExpressLRS #1768: Matek PWM RX  - WiFi functionality lost upon update

malinhiles, 2022-08-13:
## Current Behavior

I recently bought two Matek pwm receivers. One is the ELRS R24 P6, and the other is the ELRS R24 P. I recently updated the R24 P6 to  v3.0.0 RC2 and I updated the R24 P to v2.5.1. Both of them exhibit the same Wifi Issue.

Both receivers, right out of the bag, seemed to work as expected. When I powered each one on for the first time,  the "ExpressLRS_RX" WiFi network showed up and I could update the firmware for both receivers. After updating the firmware to the versions that I mentioned above, both receivers no longer worked at all with Wifi. Both receivers wouldn't join my home network  with the credentials I gave them, and they wouldn't create the ExpressLRS_RX WiFi network.

I have used 4 other expresslrs receivers; 3 from Axis Flying, and 1 from iFlight.  All 4 of these receivers have wifi functions that work as expected, so I feel somewhat familiar with how the wifi should work with these ELRS receivers. 

## Steps to Reproduce

1. Open new Matek R24 P6 or R24 P receiver

2. Power on receiver

3. Open ExpressLRS Configurator and compile appropriate firmware

4. Join the ExpressLRS_RX wifi network

5. Load firmware file into Web interface at 10.0.0.1

6. Flash receiver and wait for blinking light / rx reboot

7. Endure wifi issues

## Details

I was able to flash these receivers via FTDI adapter, but its a bit of a PITA to do considering that I wont be able to manage the level of physical access required to do that on more integrated fixed wing builds. 

With flashing new firmware being such an integral part of ELRS ecosystem, It's too much of a hassle to pull the rx out of all of my airplanes and wire it up to an FTDI adapter each time there's a pertinent update.

When flashing via FTDI adapter, the R24-P6 (and I think the R24-P too) didn't appear to reboot into its normal mode after successful flash. Booting into bootloader mode makes the status LED on the RX solid, and after the flash, the receiver still showed a solid LED, even minutes after the flash. Not sure if this is normal behavior or not, but I though it might be worth mentioning that it didn't reboot and start blinking like it does in normal operation.

AtthosTheGreat, 2022-08-13:
I've also had some problems with the wifi on RC2. After I flashed my AxisFlying module to RC2, the wifi page loaded correctly on desktop (microsoft edge, firefox, chrome, brave) but it didn't load completely on mobile (firefox, chrome, brave). On mobile it was stuck loading and the buttons were unusable. It worked fine on RC1 though. I thought I might add this here, but I'm not sure it's related
[image]
.

pkendall64, 2022-08-13:
If it doesn't display anything in the UID field or the other fields, then it has not finished loading the data from device.
Some of the devices you have to be very close to get it to work, especially when the device is connecting to your home network, some of them need to be quite close to your wifi router/access point.

AtthosTheGreat, 2022-08-13:
In my case (AxisFlying not loading the page on mobile on RC2) it never loads completely if the RX makes its own network. However, on RC1 it loads the page instantly even if I'm pretty far away. The problem is just on RC2. It's not a big problem, as when it connects to my home network (from far away even on RC2), I can use it that way. I just wanted to let you know.

malinhiles, 2022-08-13:
> Some of the devices you have to be very close to get it to work, especially when the device is connecting to your home network, some of them need to be quite close to your wifi router/access point.

@pkendall64  Thought this could be the case, but left it right next to my router for a few minutes, still no luck

pkendall64, 2022-08-16:
The thing with 3.0 is that all the firmware for ESP based RXes is the same! i.e. the binary is the same and the only difference is a hardware layout file attached to end of the firmware file.

malinhiles, 2022-08-16:
> The thing with 3.0 is that all the firmware for ESP based RXes is the same! i.e. the binary is the same and the only difference is a hardware layout file attached to end of the firmware file.

@pkendall64  Ahh, would this suggest that this is some sort of hardware issue on Matek's side of the court?

malinhiles, 2022-08-18:
Any thoughts on this? I've got Matek on the other line and they don't think this is an issue with their receivers. I've got three of these Matek PWM receivers now and they all have this issue, its pretty lame

pkendall64, 2022-08-18:
If you are able to flash via FTDI, then perhaps try a full erase flash before flashing the latest firmware. We have seen issues before, and a full flash erase has fixed it before.
If you would like you could try a new we-based flasher I'm working on https://pkendall64.github.io/elrs-web-flasher/
It has the option to select "Erase flash first" at the final step.

CapnBry, 2022-08-18:
I only flash my PWM RX through wifi so I know this works with RC2. I just tried a R24-P6 both with and without a home network defined and both came up just fine.
[image]

If you're using Chrome maybe hit F12 to open dev tools and see if there's an error in the Console tab? Maybe there's some broken JSON we're putting out from your configuration and the javascript is stopping before it fills the data?

jkramarz, 2022-09-02:
Not sure if it is the exact same issue, but I ran into a problem flashing EPW5 (target DIY 2400 RX PWMP) receivers today with 3.0.0-RC2. Flashing via WiFi (using the ExpressLRS Configurator directly or uploading the produced binary via web interface) results in a receiver giving short LED blinks and not binding to TX nor connecting to WiFi. Flashed via UART it works correctly.

malinhiles, 2022-09-08:
> If you are able to flash via FTDI, then perhaps try a full erase flash before flashing the latest firmware. We have seen issues before, and a full flash erase has fixed it before. If you would like [...]

@pkendall64 I had some time today to try this and probe around at this issue, and we are getting closer to tackling the issue, but we aren't completely there.  

I flashed via FTDI using your browser flasher with a full erase, and now the receiver joins my home wifi, and is detected by the ExpressLRS Configurator! Now that the rx joins the home wifi and shows up in the configurator, I wanted to try and flash it over wifi to see if it really works. Unfortunately, the receiver still acts funny, here's what I see.
[image]

I can't select the receiver as the WIFI device for flashing using the blue "SELECT" button in the bottom right. It gives the above shown error when I try to. If I then manually paste the receiver's IP, i can proceed through to the flashing page where it attempts to upload and flash the firmware to the receiver, but the flash fails due to a target mis-match, even though I am using what I believe to be the correct target, according to the receivers documentation. 
[image]

Any thoughts on this? I feel I'm starting to see light at the end of the tunnel lol

pkendall64, 2022-09-08:
Yeah, if you are flashing again with configurator, then the the "Build & Flash" has a chevron which you can press instead and select "Force Flash". That will ignore the mismatch and flash anyway.

malinhiles, 2022-09-08:
@pkendall64 just tried the force flash, no luck; it still fails the flash. 

Any ideas as to why the configurator cant auto select the wifi device? (in reference to the first photo in my last comment) It might also be worth noting that I can't connect to the RX to flash via wifi using your online flasher either. 
[image]

pkendall64, 2022-09-08:
It is expected that "no device found" will be seen.

The web-flasher won't be able to connect to devices over wifi unless you have the proxy running on your computer.
It's a security feature of browsers that they don't allow making connections to different domains and they certainly can't open random ports for doing mDNS etc. So the proxy forwards HTTP requests to mDNS and also to the actual RX/TX device. It does this by returning proper CORS headers to the browser (running the web-flasher).

malinhiles, 2022-09-08:
> It is expected that "no device found" will be seen.
>
> The web-flasher won't be able to connect to devices over wifi unless you have the proxy running on your computer. It's a security feature of [...]

Copy that 👍

Any thoughts about the other findings?
1. Can't select the receiver as the WIFI device for flashing using the blue "SELECT" button in the bottom right of the configruator
2. Flash via wifi doesn't work, Force flash via wifi doesn't work either

pkendall64, 2022-09-08:
No clue about the "select" problem. Thats something the Configurator guys will have to look at.

What was the error with Force Flash?

pkendall64, 2022-09-08:
What if you connect to the RX wifi with a browser and try updating that way rather than using configurator.

malinhiles, 2022-09-08:
> What if you connect to the RX wifi with a browser and try updating that way rather than using configurator.

By this do you mean a flash via the "Expresslrs_RX" hotspot network that the rx makes when it doesn't have home network credentials?

pkendall64, 2022-09-08:
Yes, or if you have your wifi credentials you can also connect to the IP address that Configurator is telling you. 192.168.86.25
Next to the "SELECT" button.

malinhiles, 2022-09-08:
> Yes, or if you have your wifi credentials you can also connect to the IP address that Configurator is telling you. 192.168.86.25 Next to the "SELECT" button.

Ahh i see. 

Just tried a browser flash, and a couple other things. and it didn't work right off of the bat.  It requires some jumping through hoops, but with **many** round-about steps, ive managed to get these Matek receivers work as intended now. Ive replicated this process on two new receivers as well with the same results. Here's what happened:

1. Power up brand-new receiver 
2. Compile a 3.0.0RC2 firmware using the configurator
3. Join ExpressLRS_RX receiver hotspot. 
4. Flash the receiver with the compiled firmware via 10.0.0.1 in browser
5. Receiver now connects to transmitter and receives as intended, **_but wifi functionality does not work at all_**
6. Connect receiver to FTDI adapter
7. Flash the receiver over Serial UART with 3.0.0RC2 firmware again, but this time via https://pkendall64.github.io/elrs-web-flasher/ in order utilize and enable  "Full Chip Erase" option.
8. Now the receiver will join home wifi network and will be visible in configurator as a WiFi device, **_but the receiver will not accept a firmware flash over wifi via configurator, even if using "Force Flash"_**
9. Compile another 3.0.0RC2  firmware for the receiver using the configurator
10. Connect to the IP address of the receiver shown in the configurator.
11. Upload newly compiled firmware.bin to the receiver over home wifi using the Receiver IP address in browser.
12. Flash receiver
13. Receiver now works as expected, **_including full wifi functionality_**

Glad to have had your prompt help with this @pkendall64! 
@pkendall64 all of this in mind, any ideas as to what might need to be changed to make it so that these receivers don't require this whole ritual in order to work properly?

pkendall64, 2022-10-30:
It's just sometimes you have to erase flash before, we have no idea why. It just seems to be on of those things.

closed 2022-10-30 by pkendall64