ExpressLRS/ExpressLRS #1769: HGLRC Hermes - failed to update to 3.0.0-RC2 with error [Not enough space]

offer-shmuely, 2022-08-13:
Failed to update TX module with error "not enough space"
this failure happen only on 3.0.0-RC2

The TX module work ok on other versions.
I check the update couple of times, some times repeated again and again, and some with full restart.
I also verify that no "double upload page" exist

update from 2.4.0 to 3.0.0-RC2 failed
update from 2.4.0 to 2.5.1 succ
update from 2.5.1 to 3.0,0-RC2 failed

## Environment
- TX hardware: Radio Master TX16s
- TX module: HGLRC Hermes
- OS EdgeTx 2.7.1

[image]

pkendall64, 2022-08-16:
Generally this just means you need to power it off and on and try again.

offer-shmuely, 2022-08-16:
Hi
I did restart it again and again, no help.

does it means that with ver 3 we will never be able to use the wifi feature? (At least with the current available modules)

pkendall64, 2022-08-16:
> Hi I did restart it again and again, no help.
> 
> does it means that with ver 3 we will never be able to use the wifi feature? (At least with the current available modules)

No, it just means that you will have to do the first upgrade to 3.0 via the USB/UART. Then the partition sizes on the flash will be big enough.

JyeSmith, 2022-08-16:
> Hi I did restart it again and again, no help.
> 
> does it means that with ver 3 we will never be able to use the wifi feature? (At least with the current available modules)

WiFi may never work well with any HGLRC gear.  They never added the wifi antenna to the ESP MCUs on the tx and rx.

offer-shmuely, 2022-08-17:
I know, i put the transmitter on top of the laptop for reception, but is not the issue here, since
I can burn version 2.4, 
i can burn version 2.5.1, 
i can NOT burn version 3.0.0-RC2

commanderguy3001, 2022-08-17:
please try just [flashing it via USB](https://www.expresslrs.org/2.0/quick-start/transmitters/hglrc-hermes/#flashing-via-usbuart), the "not enough space" error is a pretty common one, that only gets fixed by flashing a recent enough version via USB.

offer-shmuely, 2022-08-17:
The USB update worked without space issues.

pkendall64, 2022-08-23:
Can we close please

offer-shmuely, 2022-08-23:
Your choice, the bug say:
“can not upgrade to 3.0 through wifi” - is that solved?

And not:
“Me, personally can not upgrade to 3.0” - that is indeed solved

offer-shmuely, 2022-08-25:
Is that problem a one time?
When I will upgrade from 3.x to 3.y , will it be ok?

pkendall64, 2022-08-25:
Yes, it should be a one-time thing

commanderguy3001, 2022-08-25:
> Your choice, the bug say:
> “can not upgrade to 3.0 through wifi” - is that solved?
> 
> And not:
> “Me, personally can not upgrade to 3.0” - that is indeed solved

well it is a known issue which is likely not gonna get fixed, as the devs focus on the new versions and not the old 2.x versions.
also, USB flashing is a known workaround the problem, which is the best you will get, so I would say that this can be closed.

offer-shmuely, 2022-08-26:
Ok, so just close it.

I do suggest to add a small description of that issue in the website, (so there will be a need to send others to discord to find a solution)

closed 2022-08-26 by pkendall64