ExpressLRS/ExpressLRS #1773: F500 mode causes Failsafes in Arducopter

AaryanL, 2022-08-14:
## Current Behaviour

Setting F500 mode causes Arducopter randomly go to FS mode. It can happen couple of times in 5 mins. Sometimes more often, sometimes very rare. 

It happens even before Arm on the ground also displaying messages in MP:
Radio Failsafe Cleared
Radio Failsafe - Disarming

Changing serial port baud rate to 400 in Arducopter doesn't solve the problem.
Switching to 250Hz mode, completely stops any failsafes.

## Steps to Reproduce

Just set F500 in lua script. 
LQ is 100 all the time for both Rx/Tx.

This quad has already many hours flying with ELRS 2.X and never had such a problem before. In my opinion this might be some F500 mode implementation bug.

## Possible Solution (Not obligatory)
no solution found

## Your Environment

- TX hardware: HM ES24TX PRO
- RX hardware: Matek R24D
- Handset model:
- OpenTX version 2.3.15
- ExpressLRS version (TX & RX MUST MATCH): 3.0 RC2
- Packet Rate: F500
- Telemetry Ratio: 1:2
- user_defines: TX power 25mW, dyn, max 250
- 
Arducopter V4.3.0-dev, F405, Serial1 DMA

[image]

pkendall64, 2022-08-16:
When you switched from F500 to 250, do you mean D250 or just 250?
Because the difference here is that F & D modes use FLRC and the others are LoRa modulation.
And, FLRC is a lot more picky about the XTAL frequencies of the TX and RX and if they don't match closely you can get a failsafe after they warm up as you get thermal drift from the XTAL oscillator.

SunjunKim, 2022-08-16:
ES24TX Pro and Matek R24D pair is a known problematic combination.
If XO unmatch&drift is an issue, you can check it by yourself. 

Could you test the following procedure here and report the value appearing on the RSNR field?

https://github.com/ExpressLRS/ExpressLRS/wiki/Crystal-Oscillator-(XO)-Frequency-Error#measuring-relative-xo-error-between-a-tx-rx-pair

AaryanL, 2022-08-16:
> When you switched from F500 to 250, do you mean D250 or just 250? Because the difference here is that F & D modes use FLRC and the others are LoRa modulation. And, FLRC is a lot more picky about the [...]

I switched to Lora 250

But I've removed oscillator capacitors in R24D as recommended, and due to freq test FW the frq match now is much better.

AaryanL, 2022-08-16:
> ES24TX Pro and Matek R24D pair is a known problematic combination. If XO unmatch&drift is an issue, you can check it by yourself.
>
> Could you test the following procedure here and report the value [...]

I will do asap and report.
Thank you. As I mentioned caps already removed.

AaryanL, 2022-08-16:
Here is the result.
Same in 10 mins, when heated I guess.

BTW why is it initially TPW2 instead of TPWR? (just changed it manually)

[image]

SunjunKim, 2022-08-16:
Would like to test it on any of Lora mode?

AaryanL, 2022-08-16:
> Would like to test it on any of Lora mode?

Well, it's -59Db in 250 Lora )))
Even without caps. What the hell! )

SunjunKim, 2022-08-16:
> Well, it's -59Db in 250 Lora )))

It's not ideal but it should work in all modes. Given that the freq offset is not your problem, let's diagnose the original error further. I think few devs will try to reproduce the problem from your procedure, and move on.

AaryanL, 2022-10-28:
UPDATE!!!!

More tests just revealed that FS problem persist even at every Lora frequency except of 150! It happens with Ardupilot on f405 controllers. Most likely this is an ardupilot problem.

Will report any news on that.

AaryanL, 2022-11-02:
UPDATE!!!!

https://github.com/ExpressLRS/ExpressLRS/issues/1927

jimsynz, 2022-11-14:
I have the same issue with lots of short failsafes in Ardupilot. Pretty sure it’s not ELRS though because I reflashed INAV and it works perfectly and the link is rock solid.

AaryanL, 2022-11-14:
Same here with inav. It might be fs is just too short for inav and it doesn't detect it.

Why I'm sure it's elrs - same settings work good on 2.5.1 ears fw. How could that be?

AaryanL, 2022-11-14:
If it's confirmed by one more user - it looks like quite a critical issue. Will someone investigate?

pkendall64, 2022-11-14:
Can you try the master branch as we've done quite a lot of work on optimising the radio SPI communications

AaryanL, 2022-11-14:
Sure. Will report asap.

jimsynz, 2022-11-14:
I was chatting with @andyp1per over on the Ardupilot discord. He's suggested he may try and replicate it on the bench. I'll leave it up to him to follow up here if he wants.

pkendall64, 2022-11-14:
It's a bit difficult for us devs as I don't think any of us actually have any arducopter capable hardware!

AaryanL, 2022-11-14:
> Can you try the master branch as we've done quite a lot of work on optimising the radio SPI communications

Updated all to master - no luck, all the same. Downloaded also latest LUA script, now in 500 1:2 script takes ages to start. The top bar is moving really slow. It was not like that before.

jimsynz, 2022-11-18:
> Updated all to master - no luck, all the same. Downloaded also latest LUA script, now in 500 1:2 script takes ages to start. The top bar is moving really slow. It was not like that before.

I've experienced the same thing. I _think_ it might be related to Ardupilot CSRF pass through telemetry/Yaapu because if I switch to another model in EdgeTX that doesn't have Yaapu setup (even while the rx is still on and bound) then the script comes straight to life.

andyp1per, 2022-11-18:
> I've experienced the same thing. I _think_ it might be related to Ardupilot CSRF pass through telemetry/Yaapu because if I switch to another model in EdgeTX that doesn't have Yaapu setup (even while the rx is still on and bound) then the script comes straight to life.

This sounds likely to me

andyp1per, 2022-11-19:
I stupidly updated to 3.1.0 instead of 3.0.1 on my BetaFPV 2.4Ghz micro - now the lua menu (v3) does not work and the 5-way OLED doesn't work - any ideas how to get out of this mess? Not going to be able to debug this issue without a working setup, and ELRS makes this really hard - upgrading my 915Mhz TX/RX was equally frustrating...

andyp1per, 2022-11-20:
Ok fixed. Why does ELRS do this (915Mhz is the same)?:
[image]
Note the RC packets come in groups of 3 - 600us apart and then the gap between groups of 3 is 1.5 - 3.5ms. Is this by design? Can't think of a reason you would want to do this, and its a problem for flight controllers because we are expecting a roughly even packet rate.

pkendall64, 2022-11-20:
Probably because you have telemetry set to 1:4 and the one where theres only 2 is because there was probably a sync packet as well.

andyp1per, 2022-11-20:
Ok, so 1:4 means "skip every 4th frame to send telemetry"? That's not how regular CRSF works.

pkendall64, 2022-11-21:
> Ok, so 1:4 means "skip every 4th frame to send telemetry"? That's not how regular CRSF works.

ELRS uses LoRa and the window of time that we have is almost fully utilised sending the RC data so we can keep the spreading factor as high as possible so we can get the best range possible. That means that for telemetry we have to sacrifice a time slot for it.

If you want regular packet intervals, set the telemetry rate to 1:2 then it will alternate RX/telemetry. This is effectively what Crossfire and other protocols with telemetry do, they just don't tell you that because you can't adjust the ratio like you can in ELRS.

andyp1per, 2022-11-22:
I've had a bit more of a dig into this. I suspect its just a data rate thing and even with good DMA it easy to lose too many packets. I've played around with different things to try and improve, but with little joy so far. I suspect that these rates on F4 will be unattainable even with DMA - you are generally going to need H7 (which I have and works quite well).

jimsynz, 2022-11-22:
> I've had a bit more of a dig into this. I suspect its just a data rate thing and even with good DMA it easy to lose too many packets. I've played around with different things to try and improve, but [...]

I guess we can at least put in the [ExpressLRS Ardupilot page](https://www.expresslrs.org/2.0/quick-start/ardupilot-setup/) that for anything other than an H7 you need to use a lower data rate.

@andyp1per I'm using an H7 and seeing these radio failsafes though - is that to do with the timing stuff you discussed earlier?

andyp1per, 2022-11-22:
> @andyp1per I'm using an H7 and seeing these radio failsafes though - is that to do with the timing stuff you discussed earlier?

Yeah I didn't say I was done 😄 I also occasionally get these so am still investigating - but I think high data rates are mostly to blame

andyp1per, 2022-12-08:
Anyone know if ELRS sends the telemetry rate in a link statistics frame at all? We seem to be guessing this in a way that seems dubious to me.

andyp1per, 2022-12-08:
When I get the RC failsafes, ELRS is sending a garbage frame - the reported frame length is 200. Clearly nonsensical since the max frame size is 64 bytes

andyp1per, 2022-12-08:
Also what is the actual baudrate that CRSF uses on the wire?

andyp1per, 2022-12-08:
Ok, solved. The issue is the baudrate that ELRS uses - 420kbaud, this is not the CRSF baud rate which is 416kbaud. At high rates this can be quite important. @AaryanL this is an option when you upgrade the ELRS firmware - can you try upgrading your firmware but set the RX UART baud to 416666 and see if this solves the problem for you?

andyp1per, 2022-12-23:
I have added an option to arducopter in RC_OPTIONS that allows you to force the bitrate to 420k

AaryanL, 2022-12-23:
> Ok, solved. The issue is the baudrate that ELRS uses - 420kbaud, this is not the CRSF baud rate which is 416kbaud. At high rates this can be quite important. @AaryanL this is an option when you [...]

Thank you! I'm waiting for a new RX, so I can't check it right now, will test and reply asap.

AaryanL, 2023-03-01:
I can confirm that setting RCVR_UART_BAUD to 116666 solves the problem.
Not tested with original Matek but with Beta dual RX, but should be no difference.

Can be closed?

closed 2023-03-01 by pkendall64

NikosSprocket, 2023-06-22:
btw I got the same issue with F500 and in my case when the VTx (Walksnail Avatar v2) was powered on, using PIO to switch power on/off to the VTx.
After chatting with other Ardupilot and ELRS I moved to 250Hz and issues disappeared.