ExpressLRS/ExpressLRS #1777: Blinking RXBt Telemetry, Low Battery warning

caspervanbeek, 2022-08-18:
## Current Behavior

INAV 5.0, EDGE TX
From my Telemetry an RXBt flashes 16.1v and 4.7v.
Bat% flashes 76% and 0%

## Steps to Reproduce

1.

2.

3.

4.

## Possible Solution (Not obligatory)

## Details
Three questions:
I think this alarm can't be there because the battery gives 16.1v.
Why is the RXBt and the Bat% blinking?
How can I cancel this alarm?

## Your Environment

- TX hardware: Radiomaster TX16S
- RX hardware: INAV 5.0,  TBS Nano receiver
- Handset model: ??
- OpenTX version (including nightly number)
- ExpressLRS version (TX & RX MUST MATCH): edgetx-tx 16s - version 2.71 (a55aff02)
DATE 2022-05-09
TIME 04:02:03
CFGV: 221

- Packet Rate:
- Telemetry Ratio:

https://user-images.githubusercontent.com/81965645/185372190-2dc2963e-c58f-43a9-bfe1-f58a806bfd30.mp4

- user_defines:

caspervanbeek, 2022-08-18:
[image]
[image]
[image]
[image]
[image]
[image]

Greetings casper

qubic, 2022-08-18:
Ask TBS.

jurgelenas, 2022-08-18:
ExpressLRS is not affiliated with TBS products in any way. Try contacting TBS customer support or post this question to Inav community page. Best of luck!

closed 2022-08-18 by jurgelenas