ExpressLRS/ExpressLRS #1778: Backpack options in Lua script missing

nightflasher, 2022-08-18:
## Current Behavior

Lua Script -> Wifi Connectivity -> Enable Backpack Wifi (missing)
Lua script -> Backpack Settings

## Possible Solution (Not obligatory)

Options missing in script?

## Details

none

## Your Environment

Jumper t-lite & flysky nirvana nv14 + edgetx v2.7.1
DIY 2400MHz ESP32 + E28 + ESP8285 backpack (ESP-M3)

pkendall64, 2022-08-18:
This means you have compiled without the backpack enabled.
You have not said what version of ELRS you are using either!
If on 2.x firmware you need to enable the USE_TX_BACKPACK option when compiling the firmware, if on 3.x then you go to the `http://elrs_tx.local/hardware.html` page and there is an option to enable it, assuming all the pins have been defined for the backpack.

nightflasher, 2022-08-19:
Yeah...I'm on 3.0.0-RC2 and I've set the backpack option, but it's not there. After  enabling the backpack on hardware.html the options are available, but I can't enable the wifi on the backpack.
I have connected my esp backpack to the serial Pins used for programming the wroom, but it seems to not work.

pkendall64, 2022-08-20:
You need to set these options:
You should find them on the hardware.html page.
You will not be able to have anything plugged into the USB connector of a DIY module. You could of course wire it to some other free pins and configure it to use them instead.
    "use_backpack": true,
    "debug_backpack_baud": 460800,
    "debug_backpack_rx": 3,
    "debug_backpack_tx": 1,

nightflasher, 2022-08-20:
here is a screenshot of my backpack settings
[image]

nightflasher, 2022-08-20:
I've flashed the tx and the backpack esp again and now it works.

closed 2022-08-20 by nightflasher

khikmatovfarkhod, 2024-08-26:
even when i activate the lua, i cant see the wifi connectivity and espnow. please let me know what might be causing this issue. the hardware is betafpv 1w tx and x10s hours