ExpressLRS/ExpressLRS #1779: BLE Joystick mode auto exits after few seconds.

tonvfx, 2022-08-18:
## Current Behavior

I tried using the BLE Joystick on my Radiomaster TX12 (mk1) (edgetx 2.7.1) with HM es24tx pro (2.5.1) module and it looks like it's exiting ble mode after a few seconds. I can connect for a moment. I'll get input in the sim (velocidrone) but withing a few seconds, the connection is lost and the lua (v2) script has also exited the ble joystick mode menu.
I've been using the ble joystick on my older es24tx (non-pro) before (with my t8sgv3) without problems (elrs 2.0) so not sure why it doesn't work with my new setup. It was suggested on the discord to file a bug report.

## Steps to Reproduce

1. Launch ble joystick and start it.

2. Wait. (verify it connected)

3. It exits.

4.

## Possible Solution (Not obligatory)

Couldn't find any.

https://user-images.githubusercontent.com/83398761/185495329-60f4f4bf-dd97-43ed-8f57-b3a27ef46f7c.mp4

## Details

Using OSX (but that wasn't a problem with my older elrs module..)

## Your Environment

- TX hardware: HM ES24TX PRO
- RX hardware: N/A
- Handset model: Radiomaster TX12 mk1
- OpenTX version (including nightly number): edgetx 2.7.1 (a55aff02)
- ExpressLRS version (TX & RX MUST MATCH): 2.5.1 (a7e2e4)
- Packet Rate: N/A
- Telemetry Ratio: N/A
- user_defines: 
[image]

kvark85, 2022-11-09:
the same issue on Radiomaster TX12 on ExpressLRS release 3.0.0 and 3.0.1.
[image]

pkendall64, 2023-04-01:
Tested with HM ES24TX Pro Module in TX12 MKII radio with 3.2 branch and is working fine connected to Mac using Tryp Sim.
If you have issues with the latest version you can re-open the issue.

closed 2023-04-01 by pkendall64

dang-itt, 2023-08-20:
How did you install the bluetooth module? Does it work to send telemetry over bluetooth?