ExpressLRS/ExpressLRS #1789: Antennae diversity on Matek ELRS-R24-D

miggy1952, 2022-08-23:
Using 3.0.0-RC2. target MATEK 2400 RX R24D.

When I flashed via wifi the diversity wasn't working.

But when I flashed via UART antenna diversity is working fine.

pkendall64, 2022-08-23:
After flashing via WiFi you perhaps needed to clear the config, as the RX probably had an invalid config which selected only one of the antennas. You could have also used the LUA to set it to diversity mode as well.

miggy1952, 2022-08-23:
Thanks, using configurator 1.4.4. Diversity is not an option when I select target Matek 2.4 and RX R24D for 3.0.0-RC2 . However "Use diversity" is an option for 2.5.1.

pkendall64, 2022-08-23:
Diversity as a compile option has been removed and now the firmware detects and makes available via the RX LUA.

miggy1952, 2022-08-23:
Thanks again. Now I see "Other Devices" down at the bottom of the menu in the Lua. Didn't realise that there was more info to scroll down too after the version number. Yes this is the answer, thank you very much.

closed 2022-08-23 by miggy1952

SunjunKim, 2022-08-23:
I see you closed the issue, but there was a typo in the Matek Diversity target define. (#1776)
So try testing `3.x.x-maintenance` branch.