ExpressLRS/ExpressLRS #1791: TX16s with NamimnoRC FLASH 2400 does not bind With V3 RC2.   V2.5.1 works fine

RandMental, 2022-08-23:
## Current Behavior
ELRS V2.5.1 works find and bind immediately with a Binding Phrase

## Steps to Reproduce
Flash ELRS V3 RC2 
System does not bind 

## Possible Solution (Not obligatory)
Flash V2.5.1 back on the TX module, then binding happens again, but with underfined operations.
go back to V2.5.1 on both devices.   

## Your Environment
Win 11
ExpressLRS Configurator v1.4.3
- TX hardware: TX16s with NamimnoRC FLASH 2400 OLED TX
- RX hardware: NamimnoRC FLASH 2400 ESP RX
- Handset model:  TX16s
- OpenTX version (including nightly number) V2.3.15 with ELRS V3
- ExpressLRS version (TX & RX MUST MATCH): V3 RC2
- Packet Rate: 500
- Telemetry Ratio:
- user_defines:

SunjunKim, 2022-08-23:
Would you check with the following process?

- Turn on your radio with the TX module mounted
- Check the Lua script to make sure the TX is running fine
- Enter TX WiFi mode
- On the WebUI, check the version number (3.0-RC2), and write down the UID that appeared on the page (six numbers)
- Turn off the radio
- Power the RX on, and wait for 60s. Check the RX is getting into WiFi mode (fast blinking)
- On the WebUI, check the version number (3.0-RC2), and compare the UID with the TX's.
- Go to the MODEL tab, and uncheck if anything is checked (model match and telemetry off)
- Turn off the RX
- Turn on the radio, make sure your TX is running
- Turn on the RX, and check whether it's connecting. If not, please report what's your RX LED state
https://www.expresslrs.org/2.0/quick-start/led-status/

RandMental, 2022-08-25:
**On the WebUI, check the version number (3.0-RC2), and write down the UID that appeared on the page (six numbers**)

### From WEBUI 
* ExpressLRS
* NamimnoRC Flash OLED 2.4GHz TX
* Firmware Rev. 3.0.0-RC2 (7855d1) ISM2G4

[image]

**Power the RX on, and wait for 60s. Check the RX is getting into WiFi mode (fast blinking)**
    I have got the Nano RX V3.0.0 RC2 in some funny mode having downgraded to V2.5.1 on the TX side.
    It now slowly blinks the green led, and does not switch to WIFI after 20/60 seconds.
    However the LUA script previously recorded the version: 

[image]

**Turn on the RX, and check whether it's connecting. If not, please report what's your RX LED state https://www.expresslrs.org/2.0/quick-start/led-status/ **
      It does not connect and now only blinks the green led slowly, it doesn not switch to WIFI after 20-60 seconds.

SunjunKim, 2022-08-25:
Does your Rx blink exactly same to the “slow blinking”? Otherwise your RX might be in boot looping. In this case you need make a recovery flash the RX through UART.

RandMental, 2022-08-26:
Could be, will confirm today whether I can get it out of this mode

RandMental, 2022-08-26:
Hi Kim
Took a new ELRS RX and upgrade it from V2 to 

[image]

The Radio (external Module on TX16s) and has also been reflashed with the lastes V3.0.0 RC2.  TX screen as above with ID on both sides disabled.  (RX ID = 255)

With same binding phrase as used both side as for V2.5.1 it does not bind.

RandMental, 2022-08-26:
**Hi Kim - found the problem**

I have had a capital letter error in my bind-phrase for the RX 
After seeing the error entering the binding phrase on the Wifi Run time options screen it binds immediately 

Aplogise for wahsting your time 

regards

closed 2022-08-26 by RandMental